esphome/issues #5816: Devices do rolling reboots since version 2024.5.0

MisterX-RC, 2024-05-18:
### The problem

Since updating 4 off my devices start to do rolling reboots, going unavailable vor e fee seconds, then reconnecting for lets say 1 minute.

not all devices have this problem, Only my airquality meters. They run multiple sensors like the BME680 on i2c and PMS 5003S and Senseair CO2 meter on uart. 

Updating back to 2024.4.2 was difficult because the upload fails because the rolling reboots. 

I first compiled a smaller software by removing the sensor config and this would go thru after some tries. 
then I could reupload the full software without problems on 2024.4.2

### Which version of ESPHome has the issue?

2024.5.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

jsuanet, 2024-05-18:
I have the same problem with a D1 mini ESP8266 with a BME680, a SSD1306 and a PIR connected. No problems with 2024.4.2 and after upgrade the device started rebooting. When compiled with debug flag it crashes wit a OOM exception after 5 minutes, when I compile it with verbose flag it crashes within a minute after reboot with following stackdump

[10:39:21]User exception (panic/abort/assert)
[10:39:21]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[10:39:21]
[10:39:21]Unhandled C++ exception: OOM
[10:39:21]
[10:39:21]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[10:39:21]
[10:39:21]ctx: cont
[10:39:21]sp: 3ffffcf0 end: 3fffffd0 offset: 0010
[10:39:21]3ffffd00:  3fffb45c 00000000 3fffb608 4023b694  
[10:39:21]3ffffd10:  00000020 00000031 3fffba1c 40214520  
[10:39:21]3ffffd20:  00000031 00000000 00000034 4010116e  
[10:39:21]3ffffd30:  3fffb5fd 00000000 3fffbe24 3fff2bec  
[10:39:21]3ffffd40:  3ffffdb0 00000031 3fffb9f4 40214786  
[10:39:21]3ffffd50:  3fffb5fd 00000000 3fffb9f4 40214dbc  
[10:39:21]3ffffd60:  3fffb5ec 3fffb5fd 3fffb5fd 40101066  
[10:39:21]3ffffd70:  3ffffde0 3ffe8620 3fffb470 40101066  
[10:39:21]3ffffd80:  3ffeb40f 3ffe8620 00000e73 401012bd  
[10:39:21]3ffffd90:  00000000 3f313534 3fff3590 3ffe9806  
[10:39:21]3ffffda0:  00000005 00000001 00000000 00000000  
[10:39:21]3ffffdb0:  00000000 00000000 00000000 00000001  
[10:39:21]3ffffdc0:  3fff2bfc 00000011 0000000c 3fff6adc  
[10:39:21]3ffffdd0:  3fff9f5c 40226a5c 3ffffe50 3fff6adc  
[10:39:21]3ffffde0:  3fffaba0 3fffaba0 3fffb9f4 402150d8  
[10:39:21]3ffffdf0:  00000000 00000000 3fffb934 40212e30  
[10:39:21]3ffffe00:  00000000 00000000 00000000 00000000  
[10:39:21]3ffffe10:  3ffe0000 3ffe9730 0000000c 00514862  
[10:39:21]3ffffe20:  0000000e 3fffb950 3ffffe50 3fff6adc  
[10:39:21]3ffffe30:  40213f44 3fffb950 3fffb9f4 40212ba4  
[10:39:21]3ffffe40:  00000000 402181a6 000000b0 40101660  
[10:39:21]3ffffe50:  3ffffee4 00000000 2e323900 2e383631  
[10:39:21]3ffffe60:  2e383731 2e003435 00000001 3fff6adc  
[10:39:21]3ffffe70:  3fffaba0 3fffaba0 3fff6adc 40218311  
[10:39:21]3ffffe80:  17a50210 38b2a8c0 3fffb3a8 40101066  
[10:39:21]3ffffe90:  40242568 00000000 00000e5a 00000000  
[10:39:21]3ffffea0:  0000000e 2e323931 2e383631 2e383731  
[10:39:21]3ffffeb0:  00003435 3fffb934 3fff33d8 3fff33d8  
[10:39:21]3ffffec0:  00000000 0000464b 46666666 00000000  
[10:39:21]3ffffed0:  40106501 00000000 3ffffef0 3fff752c  
[10:39:21]3ffffee0:  00000000 0000001c 3fffb3c0 3fff752c  
[10:39:21]3ffffef0:  00000000 0000464b 683126e9 00514730  
[10:39:21]3fffff00:  3fffdad0 3fff2b74 3fff2b74 3fffa8a4  
[10:39:21]3fffff10:  3fff33d8 00000001 0000001c 00000000  
[10:39:21]3fffff20:  4027844e 3fff677c 00000001 40234570  
[10:39:21]3fffff30:  40279637 00000010 0000463a 4023bb9d  
[10:39:21]3fffff40:  00000000 0000464b 7cac0831 00514748  
[10:39:21]3fffff50:  3fffb3c0 00000002 3fff752c 3fff6adc  
[10:39:21]3fffff60:  3fffb3d0 3fff6adc 3fffff80 3fff6adc  
[10:39:21]3fffff70:  3fffb3d0 00000002 3fff2b74 40227cce  
[10:39:21]3fffff80:  0000464b 3fff6adc 3fff9ba4 3fff37a0  
[10:39:21]3fffff90:  3fffb3d4 3fff7308 3fff2ba8 3fff37a0  
[10:39:21]3fffffa0:  3fffdad0 00000000 3fff3774 3fff37a0  
[10:39:21]3fffffb0:  3fffdad0 00000000 3fff3774 4023bc48  
[10:39:21]3fffffc0:  feefeffe feefeffe 3fffdab0 4010048d  
[10:39:21]<<<stack<<<
[10:39:21]
[10:39:21]last failed alloc call: 40214520(49)
WARNING Memory allocation of 49 bytes failed at 40214520
[10:39:21]
[10:39:21]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[10:39:21]
[10:39:21]last failed alloc caller: 0x40214520

If needed I can also post the yaml file, but I use a lot of packages so I need to do some work to make the posted yaml useful for debugging by another user

jsuanet, 2024-05-18:
I looked a little further to other stackdumps, some of them give another reason. So it looks for me definitely as memory manipulation error (using the pointer itself instead of the content, or vice versa or some kind of thing) introduced in some changes in this version of ESPHome or used libraries.

[10:58:04]atal exception 29(StoreProhibitedCause):
[10:58:04]epc1=0x4000df64, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
[10:58:04]
[10:58:04]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[10:58:04]
[10:58:04]Exception (29):
WARNING Exception type: Access to invalid address: STORE (wild pointer?)
[10:58:04]epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[10:58:04]
[10:58:04]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[10:58:04]
[10:58:04]ctx: sys
[10:58:04]sp: 3fffec70 end: 3fffffb0 offset: 0150
[10:58:04]3fffedc0:  40261d3b 3fff11e0 00000002 40261cd0  
[10:58:04]3fffedd0:  00000002 40261c77 00000002 40260dcc  
[10:58:04]3fffede0:  40260df5 3fffee90 3fff11e0 00000016  
[10:58:04]3fffedf0:  4025e858 3fffee90 3fff1088 3fff0a50  
[10:58:04]3fffee00:  3ffedc54 3fffee90 3fffee90 3fffee50  
[10:58:04]3fffee10:  6f4a6f4a 5f343273 40004246 00000100  
[10:58:04]3fffee20:  3ffecf7c 7fffffff 00000000 00000001  
[10:58:04]3fffee30:  4025abb1 00004288 3ffecfcc 40101066  
[10:58:04]3fffee40:  ffffffd9 3fff0b9c 3ffedc64 3fff11e0  
[10:58:04]3fffee50:  3fff0180 00000039 00000000 4025f553  
[10:58:04]3fffee60:  00000000 3fff9f5c ffffffd9 00000000  
[10:58:04]3fffee70:  00000000 3fff11e0 00000010 ffffffff  
[10:58:04]3fffee80:  40000f58 00000000 00000020 00000000  
[10:58:04]3fffee90:  00000000 10310606 00640100 00000049  
[10:58:04]3fffeea0:  3ffedc78 00000116 3ffedc91 3ffedc6c  
[10:58:04]3fffeeb0:  3ffedc97 3ffedc78 3ffedc84 3ffedca2  
[10:58:04]3fffeec0:  00000000 00000000 3ffedd78 3ffedd04  
[10:58:04]3fffeed0:  3ffedcb6 3ffedcd2 00000000 3ffedd59  
[10:58:04]3fffeee0:  00000000 00000000 00000039 00000000  
[10:58:04]3fffeef0:  3fffa334 4025efc2 3fff0180 3fff9f5c  
[10:58:04]3fffef00:  00000000 3fff11e0 3fff0180 3ffedc54  
[10:58:04]3fffef10:  3ffedc54 0000013e 00000000 00000039  
[10:58:04]3fffef20:  00000000 3ffedc5e 4027268f 3fff0180  
[10:58:04]3fffef30:  3ffedc48 3fffdcc0 3ffec800 3ffec800  
[10:58:04]3fffef40:  00000080 3fff0180 00000000 3fffdab0  
[10:58:04]3fffef50:  40271f53 3fffdab0 00000000 4023b99e  
[10:58:04]3fffef60:  3ffec800 40000f49 3fffdab0 40000f49  
[10:58:04]3fffef70:  40000e19 000945a5 bff00000 0000bfff  
[10:58:04]3fffef80:  00000005 aa55aa55 000000ed 40105ea9  
[10:58:04]3fffef90:  40105eaf bff00000 0000bfff ffffffff  
[10:58:04]3fffefa0:  4010000d bff00000 000945a5 401000ab  
[10:58:04]3fffefb0:  40266edc 3fffef4c 40266e8d 3ffffcd8  
[10:58:04]3fffefc0:  3fffffd0 00000000 00000000 feefeffe  
[10:58:04]3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff000:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff010:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff020:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff030:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff040:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff050:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff060:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff070:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff080:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff090:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff0c0:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff0d0:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff0e0:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff0f0:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff100:  feefeffe feefeffe feefeffe feefeffe  
[10:58:04]3ffff110:  feefeffe feefeffe feefeffe feefeffe  
[... 231 lines cut ...]
[10:58:05]3fffff90:  3fffb3c4 3fff7308 3fff2ba8 3fff37a0  
[10:58:05]3fffffa0:  3fffdad0 00000000 3fff3774 3fff37a0  
[10:58:05]<<<stack<<<
[10:58:05]
[10:58:05]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[10:58:05]
 
If decode this stackdump with an online decoder (https://maximeborges.github.io/esp-stacktrace-decoder/), I get the following output:

0x4024937c: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:232
0x40244981: _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c:196
0x4024937c: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:232
0x402492b8: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:179
0x40244ab0: _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c:246
0x402492b8: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:179
0x4024eff9: glue2esp_linkoutput at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:301
0x402497cc: _svfprintf_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:662
0x4024f227: new_linkoutput at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:272
0x4024f67a: ethernet_output at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c:312
[... cut ...]
0x402569ac: ip4_output_if_opt at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:1577
0x402537c6: tcp_output at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_out.c:1362
0x40250ad5: tcp_close_shutdown_fin at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp.c:448
0x402516cf: tcp_close at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp.c:498

oli737, 2024-05-19:
Same here with esp8266 and SSD1306 128x64 Display.

apeeters, 2024-05-30:
The cause is most likely https://github.com/esphome/issues/issues/5857 which was fixed in ESPHome 2024.5.4 with change https://github.com/esphome/esphome/pull/6819.

MisterX-RC, 2024-05-31:
I updated 2 of my devices to 2024.5.4 to verify. First few times they didn't go beyond 2 hours of uptime, Normaly this would not be a problem, but because the bme680_bsec sensor the calibration resets, this gives incorrect readings.

After around 8 hours of still sometime rebooting they both look more stable where 1 is currently at 800 minutes uptime and the other around 600. I'll monitor for now and report back! 

Thank you all for the effort so far!