esphome/issues #5817: Template switch causes non-booting devices

ksya, 2024-05-18:
### The problem

One of my devices does not boot (or is in bootloop) after uploading the following code:
Unfortunately I also could not get serial log working.

The problem is on the addition of line 121 and the interlock for this switch. The goal was to use this as a "block" to not lower the screen when the window is open. This switch was supposed to be enabled using home assistant.

### Which version of ESPHome has the issue?

2024.5.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

Sonoff Dual-R3

### Example YAML snippet

```yaml
substitutions:
  devicename: screen-keuken
  long_devicename: Screen Keukenraam
  
esphome:
  name: $devicename
  platform: ESP32
  board: esp32dev

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Screen-Keuken Fallback Hotspot"
    password: "OHxPoT0UpDIp"
  
# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  safe_mode: True
  password: "629eae8216e1e88bbc11476be3abb1a3"

web_server:
  port: 80

logger:
  baud_rate: 0
  
captive_portal:

# remove # below to enable ble tracking
#esp32_ble_tracker:
uart:
  tx_pin: GPIO25
  rx_pin: GPIO26
  baud_rate: 4800
  parity: NONE
  stop_bits: 2
  
sensor:
  - platform: wifi_signal
    name: "$long_devicename WiFi Signal"
    update_interval: 60s
  - platform: bl0939
    update_interval: 1s
#    voltage:
#      name: "$long_devicename Voltage"
    current_1:
      name: "$long_devicename Current Open"
      id: open_current_sensor
      internal: True
    current_2:
      name: "$long_devicename Current Close"
      id: close_current_sensor
      internal: True
    active_power_1:
      name: "$long_devicename Power Open"
    active_power_2:
      name: "$long_devicename Power Close"
#    energy_1:
#      name: '$long_devicename Energy Open'
#    energy_2:
#      name: '$long_devicename Energy Close'
    energy_total:
      name: '$long_devicename Energy Total'

cover:
  - platform: current_based
    device_class: shutter
    id: shutter
    name: "Screen Keuken"

    open_sensor: open_current_sensor
    open_moving_current_threshold: 0.35
    open_obstacle_current_threshold: 0.8
    open_duration: 32s
    open_action:
      - switch.turn_on: open_cover_switch

    close_sensor: close_current_sensor
    close_moving_current_threshold: 0.35
    close_obstacle_current_threshold: 0.8
    close_duration: 31s
    close_action:
      - switch.turn_on: close_cover_switch

    stop_action:
      - switch.turn_off: close_cover_switch
      - switch.turn_off: open_cover_switch

    obstacle_rollback: 30%
    start_sensing_delay: 2s

    malfunction_detection: true
    malfunction_action:
      then:
        - logger.log: "Malfunction detected. Relay welded."
        
switch:
  - platform: gpio
    pin: GPIO27
    name: "$long_devicename - Open"
    id: open_cover_switch
    restore_mode: RESTORE_DEFAULT_OFF
    interlock: [close_cover_switch] #ensure only one relay active at a time
    interlock_wait_time: 200ms
    internal: True
  - platform: gpio
    pin: GPIO14
    name: "$long_devicename - Close"
    id: close_cover_switch
    restore_mode: RESTORE_DEFAULT_OFF
    interlock: [open_cover_switch, window_unsafe] #ensure only one relay active at a time
    interlock_wait_time: 200ms
    internal: True
  - platform: template
    id: window_unsafe
    name: "Unsafe to close"
    turn_on_action:
      - switch.turn_on: window_unsafe
    turn_off_action:
      - switch.turn_on: window_unsafe

status_led:
  pin:
    number: GPIO13
    inverted: yes

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO32
      mode: INPUT_PULLUP
      inverted: True
    name: $long_devicename Open Button
    on_press:
      then:
        cover.open: shutter

  - platform: gpio
    pin:
      number: GPIO33
      mode: INPUT_PULLUP
      inverted: True
    name: $long_devicename Close Button
    on_press:
      then:
        cover.close: shutter

# binary_sensor:
  # - platform: gpio
  #   pin:
  #     number: GPIO0
  #     mode: INPUT_PULLUP
  #     inverted: True
  #   name: "$long_devicename button"
  #   on_press:
  #     - logger.log: "$long_devicename button"
  #     - switch.toggle: sw1
```
```

ChrisOboe, 2024-12-27:
i have the same issue with a template switch and turn_on_action / turn_off_action. In my case it's not recursive so i'm not sure if thats the reason. 

here is my minimal example where this problem happens: 

```
esp32:
  variant: ESP32C3
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

esp32_ble:
  id: ble

switch:
  - platform: template
    turn_on_action:
     - ble.enable:
    turn_off_action:
     - ble.disable:
```

ssieb, 2024-12-27:
I don't know why that would cause a boot loop, but it won't work.  You need to enable optimistic.