esphome/issues #5818: AP fallback not working after upgrading ESPHome to 2024.5.0

rp2019433, 2024-05-18:
### The problem

After upgrading ESPHome to version 2024.5.0, WiFi AP fallback (captive portal) is not working anymore. AP network is created and visible to any device that wants to connect, but it's not possible to join the ESPHome AP network.

This function is working properly in version 2024.3. Below affected configuration section:
```
wifi:
  ap:
    ssid: "device fallback"
    password: ""

captive_portal:
```

https://esphome.io/components/captive_portal.html

### Which version of ESPHome has the issue?

2024.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.4

### What platform are you using?

ESP32-IDF

### Board

Sonoff NSPanel

### Example YAML snippet

```yaml
wifi:
  ap:
    ssid: "device fallback"
    password: ""

captive_portal:
```
```

bkaufx, 2024-05-20:
Pretty sure this is the same problem from https://github.com/esphome/issues/issues/5793.  The something went wrong with the web_server_base component in 2024.5.0.

ssieb, 2024-05-26:
Can you try this with the PR mentioned in #5793 or else with the next point release 5.4, whichever you get to first?

rp2019433, 2024-05-26:
> Can you try this with the PR mentioned in #5793 or else with the next point release 5.4, whichever you get to first?

Yes. As soon as release 5.4 is rolled out, I will test again and report the result on this issue. Thanks.

rp2019433, 2024-05-28:
> Can you try this with the PR mentioned in #5793 or else with the next point release 5.4, whichever you get to first?

Hi @ssieb. AP fallback (captive portal) still does not work even using release 2024.5.4. When flashing ESPHome with version 2024.3, it works properly.

GelidusResearch, 2024-06-07:
The issue started after 2024.4.2, the captive portal on esp32's fails to open the WEB server and also errors with "wrong password" when none is set. If the password is set it will spin waiting on the web server.

rp2019433, 2024-06-23:
> The issue started after 2024.4.2, the captive portal on esp32's fails to open the WEB server and also errors with "wrong password" when none is set. If the password is set it will spin waiting on the web server. #5793 does not resolve it.

Hi. Yes, still running into this captive portal problem.

rp2019433, 2024-08-21:
It seem this problem is already fixed on 2024.7.3 in my tests. Can someone else confirm it as well?

coxtor, 2024-09-13:
I also have this issue on version 2024.8.3 using a d1 mini

GelidusResearch, 2024-09-13:
It works on 2024.8.3 open a browser on the connecting client.

coxtor, 2024-09-13:
Ok sorry, indeed - it just takes >3 minutes after booting to become responsive. Is that normal ? Prior I was basically connecting and more or less instantly accessing the ip - patientce I gues..

rp2019433, 2024-09-13:
> It works on 2024.8.3 open a browser on the connecting client.

Thank you for corming. I tested on 2024.7.3 and it works.