esphome/issues #5819: datetime has not been declared with MQTT component

CAREVN, 2024-05-18:
### The problem

I have included time and datetime component in my esphome yaml, however, the compilation stopped at this mqtt_datetime.cpp file. Following is the log file of the compiling process.

### Which version of ESPHome has the issue?

Current version: 2024.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.4

### What platform are you using?

ESP32

### Board

ESP32-S2-Lolin Mini

### Component causing the issue

Mqtt

### Example YAML snippet

```yaml
time:
  - platform: sntp

datetime:
  - platform: template
    entity_category: config
    id: ${name}_start_discharge_1
    type: time
    name: Start discharging time 1
    optimistic: yes
    initial_value: "12:34:56"
    #restore_value: true
  - platform: template
    entity_category: config
    id: ${name}_end_discharge_1
    type: time
    name: End discharging time 1
    optimistic: yes
    initial_value: "12:34:56"
    #restore_value: true
  - platform: template
    entity_category: config
    id: ${name}_start_discharge_2
    type: time
    name: Start discharging time 2
    optimistic: yes
    initial_value: "12:34:56"
    #restore_value: true
  - platform: template
    entity_category: config
    id: ${name}_end_discharge_2
    type: time
    name: End discharging time 2
    optimistic: yes
    initial_value: "12:34:56"
    #restore_value: true
```

### Anything in the logs that might be useful for us?

```txt
In file included from src/esphome/components/mqtt/mqtt_datetime.cpp:1:
src/esphome/components/mqtt/mqtt_datetime.h:20:34: error: 'datetime' has not been declared
   explicit MQTTDateTimeComponent(datetime::DateTimeEntity *time);
                                  ^~~~~~~~
src/esphome/components/mqtt/mqtt_datetime.h:20:58: error: expected ')' before '*' token
   explicit MQTTDateTimeComponent(datetime::DateTimeEntity *time);
                                 ~                        ^~
                                                          )
src/esphome/components/mqtt/mqtt_datetime.h:38:3: error: 'datetime' does not name a type; did you mean 'adjtime'?
   datetime::DateTimeEntity *datetime_;
   ^~~~~~~~
   adjtime
src/esphome/components/mqtt/mqtt_datetime.cpp:16:26: error: 'datetime' is not a namespace-name
 using namespace esphome::datetime;
                          ^~~~~~~~
src/esphome/components/mqtt/mqtt_datetime.cpp:16:34: error: expected namespace-name before ';' token
 using namespace esphome::datetime;
                                  ^
src/esphome/components/mqtt/mqtt_datetime.cpp:18:45: error: expected constructor, destructor, or type conversion before '(' token
 MQTTDateTimeComponent::MQTTDateTimeComponent(DateTimeEntity *datetime) : datetime_(datetime) {}
                                             ^
src/esphome/components/mqtt/mqtt_datetime.cpp: In lambda function:
src/esphome/components/mqtt/mqtt_datetime.cpp:22:23: error: 'class esphome::mqtt::MQTTDateTimeComponent' has no member named 'datetime_'
     auto call = this->datetime_->make_call();
                       ^~~~~~~~~
src/esphome/components/mqtt/mqtt_datetime.cpp: In member function 'virtual void esphome::mqtt::MQTTDateTimeComponent::setup()':
src/esphome/components/mqtt/mqtt_datetime.cpp:43:9: error: 'class esphome::mqtt::MQTTDateTimeComponent' has no member named 'datetime_'
   this->datetime_->add_on_state_callback([this]() {
         ^~~~~~~~~
src/esphome/components/mqtt/mqtt_datetime.cpp: In lambda function:
src/esphome/components/mqtt/mqtt_datetime.cpp:44:31: error: 'class esphome::mqtt::MQTTDateTimeComponent' has no member named 'datetime_'
     this->publish_state(this->datetime_->year, this->datetime_->month, this->datetime_->day, this->datetime_->hour,
                               ^~~~~~~~~
src/esphome/components/mqtt/mqtt_datetime.cpp:44:54: error: 'class esphome::mqtt::MQTTDateTimeComponent' has no member named 'datetime_'
     this->publish_state(this->datetime_->year, this->datetime_->month, this->datetime_->day, this->datetime_->hour,
                                                      ^~~~~~~~~
src/esphome/components/mqtt/mqtt_datetime.cpp:44:78: error: 'class esphome::mqtt::MQTTDateTimeComponent' has no member named 'datetime_'
     this->publish_state(this->datetime_->year, this->datetime_->month, this->datetime_->day, this->datetime_->hour,
                                                                              ^~~~~~~~~
src/esphome/components/mqtt/mqtt_datetime.cpp:44:100: error: 'class esphome::mqtt::MQTTDateTimeComponent' has no member named 'datetime_'
     this->publish_state(this->datetime_->year, this->datetime_->month, this->datetime_->day, this->datetime_->hour,
                                                                                                    ^~~~~~~~~
src/esphome/components/mqtt/mqtt_datetime.cpp:45:31: error: 'class esphome::mqtt::MQTTDateTimeComponent' has no member named 'datetime_'
                         this->datetime_->minute, this->datetime_->second);
                               ^~~~~~~~~
src/esphome/components/mqtt/mqtt_datetime.cpp:45:56: error: 'class esphome::mqtt::MQTTDateTimeComponent' has no member named 'datetime_'
                         this->datetime_->minute, this->datetime_->second);
                                                        ^~~~~~~~~
src/esphome/components/mqtt/mqtt_datetime.cpp: In member function 'virtual const esphome::EntityBase* esphome::mqtt::MQTTDateTimeComponent::get_entity() const':
src/esphome/components/mqtt/mqtt_datetime.cpp:55:76: error: 'const class esphome::mqtt::MQTTDateTimeComponent' has no member named 'datetime_'
 const EntityBase *MQTTDateTimeComponent::get_entity() const { return this->datetime_; }
                                                                            ^~~~~~~~~
src/esphome/components/mqtt/mqtt_datetime.cpp: In member function 'virtual bool esphome::mqtt::MQTTDateTimeComponent::send_initial_state()':
src/esphome/components/mqtt/mqtt_datetime.cpp:61:13: error: 'class esphome::mqtt::MQTTDateTimeComponent' has no member named 'datetime_'
   if (this->datetime_->has_state()) {
             ^~~~~~~~~
src/esphome/components/mqtt/mqtt_datetime.cpp:62:38: error: 'class esphome::mqtt::MQTTDateTimeComponent' has no member named 'datetime_'
     return this->publish_state(this->datetime_->year, this->datetime_->month, this->datetime_->day,
                                      ^~~~~~~~~
src/esphome/components/mqtt/mqtt_datetime.cpp:62:61: error: 'class esphome::mqtt::MQTTDateTimeComponent' has no member named 'datetime_'
     return this->publish_state(this->datetime_->year, this->datetime_->month, this->datetime_->day,
                                                             ^~~~~~~~~
src/esphome/components/mqtt/mqtt_datetime.cpp:62:85: error: 'class esphome::mqtt::MQTTDateTimeComponent' has no member named 'datetime_'
     return this->publish_state(this->datetime_->year, this->datetime_->month, this->datetime_->day,
                                                                                     ^~~~~~~~~
src/esphome/components/mqtt/mqtt_datetime.cpp:63:38: error: 'class esphome::mqtt::MQTTDateTimeComponent' has no member named 'datetime_'
                                this->datetime_->hour, this->datetime_->minute, this->datetime_->second);
                                      ^~~~~~~~~
src/esphome/components/mqtt/mqtt_datetime.cpp:63:61: error: 'class esphome::mqtt::MQTTDateTimeComponent' has no member named 'datetime_'
                                this->datetime_->hour, this->datetime_->minute, this->datetime_->second);
                                                             ^~~~~~~~~
src/esphome/components/mqtt/mqtt_datetime.cpp:63:86: error: 'class esphome::mqtt::MQTTDateTimeComponent' has no member named 'datetime_'
                                this->datetime_->hour, this->datetime_->minute, this->datetime_->second);
                                                                                      ^~~~~~~~~
*** [.pioenvs/homehna/src/esphome/components/mqtt/mqtt_datetime.cpp.o] Error 1
```

### Additional information

I tried to add time-entry into esphome but it seems there is no way to do so!
[image]
Many thanks for your supports!

jkl1337, 2024-07-12:
I ran into this problem a few weeks ago but was traveling and unable to post a fix at the time. This seems like it should be fixed by esphome/esphome PR 6914.

jesserockz, 2024-07-14:
Fixed by https://github.com/esphome/esphome/pull/6914

closed 2024-07-14 by jesserockz