esphome/issues #5823: Firmware update failure due to binary size error

z3liff, 2024-05-19:
### The problem

I upgraded to 2024.5.0 and now both 8266 devices I use are failing intermittently to update over the network.  I get an "unknown error from ESP" when the firmware should upload.  If I reboot the ESP8266, it will update.  It appears to be a random failure, as sometimes the update works without a reboot.  I never noticed any such issue before upgrading to 2024.5.0.

After this issue, and all the other issues detailed in #5821 I decided to revert to an earlier version of ESPHome (2024.4.2).  Initial flash attempt for 1 of the 2 ESP 8266 devices failed to go from 2024.5.0 to 2024.4.2.  Multiple firmware flashes once running 2024.4.2 have succeeded without issue, which leads me to believe this is an issue explicitly present in 2024.5.0.

### Which version of ESPHome has the issue?

2024.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.4

### What platform are you using?

ESP8266

### Board

D1 Mini

### Example YAML snippet

```yaml
substitutions:
  name: haks-kitchen
  friendly_name: Kitchen HAKS

packages:
  haks:
    url: https://PigLab.ReaperLegion.net/home-automation/hass/esphome/packages/haks.git
    files:
      - core.yaml
      - WiFi.yaml
      - BME280.yaml
      - PMS7003.yaml
      - OLED_US_kPa.yaml

api:
  encryption:
    key: !secret haks_key

ota:
  password: !secret haks_ota
  version: 2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret domain
  ap:
    password: !secret haks_ota

# Sunrise/sunset for OLED day/night mode (reduce burn-in)
sun:
  latitude: !secret latitude
  longitude: !secret longitude
```

### Anything in the logs that might be useful for us?

```txt
esp8266_copy_factory_bin([".pioenvs/haks-kitchen/firmware.bin"], [".pioenvs/haks-kitchen/firmware.elf"])
======================== [SUCCESS] Took 116.45 seconds ========================
INFO Successfully compiled program.
INFO Connecting to xxxxxxx
INFO Uploading /data/build/haks-kitchen/.pioenvs/haks-kitchen/firmware.bin (601856 bytes)
INFO Compressed to 412367 bytes
ERROR Error binary size: Unknown error from ESP
```

sweetpants, 2024-05-21:
I'm having the same error(s) after upgrade from 2024.4.2 -> 2024.5.2 preventing to update through OTA.

vgomenyuk, 2024-05-21:
I have the same issue with OTA update since 2024.5.0. Updating by USB to 2024.5.1 and 2024.5.2 didn't fix an issue. Previously (before 2024.5.0) I didn't have issues with the OTA updates.
Rebooting didn't help. 

Got two kind of errors:
- ERROR Error binary size: Unknown error from ESP
- ERROR Error receiving acknowledge binary size: timed out

Logs example:
```
INFO ESPHome 2024.5.2
INFO Reading configuration /config/esphome/my-device.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing my-device (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/my-device/src/main.cpp.o
Linking .pioenvs/my-device/firmware.elf
RAM:   [==        ]  15.8% (used 51684 bytes from 327680 bytes)
Flash: [======    ]  64.7% (used 1186729 bytes from 1835008 bytes)
Building .pioenvs/my-device/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/my-device/firmware.bin"], [".pioenvs/my-device/firmware.elf"])
Wrote 0x1332a0 bytes to file /data/build/my-device/.pioenvs/my-device/firmware-factory.bin, ready to flash to offset 0x0
======================== [SUCCESS] Took 124.54 seconds ========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.64
INFO Uploading /data/build/my-device/.pioenvs/my-device/firmware.bin (1192608 bytes)
ERROR Error binary size: Unknown error from ESP
```

GroteGehaktBal, 2024-05-21:
Rebooting the ESP fixed this issue for me too

maretodoric, 2024-06-13:
I had the same problem, rebooting the ESP **did not** work for me. Tried multiple times through few days..
But then something weird happened.

I've opened another tab and started log output in it to see if there are any errors logged during the update.
In first tab I've attempted to update it again and it updated it successfully.

This is log output during the update process:

```
[22:23:55][W][mqtt:323]: MQTT Disconnected: TCP disconnected.
[22:23:55][W][component:157]: Component mqtt set Warning flag: unspecified
[22:23:55][I][mqtt:234]: Connecting to MQTT...
[22:23:55][W][component:170]: Component mqtt cleared Warning flag
[22:23:55][I][mqtt:274]: MQTT Connected!
[22:23:55][W][component:237]: Component mqtt took a long time for an operation (110 ms).
[22:23:55][W][component:238]: Components should block for at most 30 ms.
[22:23:57][D][ota:154]: Starting OTA Update from 192.168.0.201...
[22:24:02][D][ota:330]: OTA in progress: 0.3%
[22:24:04][D][ota:330]: OTA in progress: 63.9%
[22:24:04][D][ota:330]: OTA in progress: 94.3%
[22:24:04][I][ota:363]: OTA update finished!
[22:24:04][W][component:170]: Component ota cleared Warning flag
[22:24:05][I][app:132]: Rebooting safely..
```

I'm not sure why viewing the log allowed the update to go through but please try yourself.

z3liff, 2024-07-09:
Updating to the newest esphome version apparently fixed this somehow.  Not sure what got unbroken but I have not been able to replicate on the latest version.

closed 2024-07-09 by z3liff

cociweb, 2024-08-25:
On 2024.8.0 the restart helped... (so the issue is still persist.)

ornolio, 2024-11-23:
Me too have the error:
Compiling .pioenvs/seplos-bms/src/main.cpp.o
Linking .pioenvs/seplos-bms/firmware.elf
RAM:   [====      ]  44.4% (used 36380 bytes from 81920 bytes)
Flash: [====      ]  40.3% (used 420605 bytes from 1044464 bytes)
Building .pioenvs/seplos-bms/firmware.bin
esp8266_copy_factory_bin([".pioenvs/seplos-bms/firmware.bin"], [".pioenvs/seplos-bms/firmware.elf"])
esp8266_copy_ota_bin([".pioenvs/seplos-bms/firmware.bin"], [".pioenvs/seplos-bms/firmware.elf"])
========================= [SUCCESS] Took 19.25 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.113 port 8266...
INFO Connected to 192.168.1.113
INFO Uploading /data/build/seplos-bms/.pioenvs/seplos-bms/firmware.bin (424752 bytes)
INFO Compressed to 291437 bytes
ERROR Error binary size: Unknown error from ESP

ssieb, 2024-11-24:
What do the device logs show?

dbuggz, 2024-12-07:
Just a thought. I had the ERROR binary size error and did a 'Clean Build Files' from the ESPHome Dashbord.

jdranchman, 2025-01-16:
Ditto on the error. Only running Cleaning Build Files wasn't enough. I restarted my esp dev,  cleaned build files and then ran the install - that cleared up the error for me.