esphome/issues #5825: Mitsubishi integration isn't reporting current temperature

thomabx, 2024-05-20:
### The problem

esphome/esphome#3886 doesn't pass on the current temperature from a sensor. Home Assistant shows 32 F or 0 C as the current temperature.

I experienced the same error as a user on the [forums](https://community.home-assistant.io/t/ha-sensor-shows-0-degrees-in-frontend-of-esphome-ir-remote-climate-card/714618).

I was able to hack a fix by adding this to ClimateTraits:

`traits.set_supports_current_temperature(this->sensor_ != nullptr);`

### Which version of ESPHome has the issue?

2024.3.0

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
sensor:
  - platform: homeassistant
    id: room_temperature
    entity_id: sensor.livingroom_temp_temperature
    filters:
      - lambda: return (x - 32) * (5.0/9.0);
climate:
  - platform: mitsubishi
    supports_dry: "true"
    supports_fan_only: "true"
    set_fan_mode: "quiet_4levels"
```

solankimihir, 2024-07-08:
Hi,
I just read your post on the said forum and managed to resolve it. I submitted a PR as per the code given by you. 
If you want, I can take back my PR so that you can send it in under your name. https://github.com/esphome/esphome/pull/7060

RubyBailey, 2024-08-12:
Hi @thomabx , did the above mentioned PR solve your issue?

thomabx, 2024-08-16:
Upgrading to esphome 2024.7.3 with the PR corrected this issue.

closed 2024-08-16 by thomabx