esphome/issues #5826: After update 2024.5.0 - ESP8266 web server randomly don't display the content.

attawutbcc, 2024-05-20:
### The problem

After updating to version 2024.5.0, 
the ESP8266 web server page occasionally fails to display content when opened. 
Users need to repeatedly press [F5] to refresh the page. 
Interestingly, the same .yaml configuration running on an ESP32 doesn’t exhibit this issue.

### Which version of ESPHome has the issue?

2024.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.4

### What platform are you using?

ESP8266

### Board

D1 mini

### Component causing the issue

Web server page

### Example YAML snippet

```yaml
# Enable Web server.
web_server:
  port: 80
```

MisterRadish, 2024-05-20:
I can confirm this issue.

I have 12 ESP8266 Nodemcu devices - a mixture of v1 and v3 all providing different functions, running on different networks and accessed by different machines and browsers. All are experiencing issues with the web server on 2024.5.0 and all worked without issue previously. The web page displays but there are not entities or log messages displayed. If I refresh the web page on average 10 times I may get content. I cannot see any debug errors in the ESP output
All ESPs use digest auth which appear to be still working, All run V2 js using default JS source and run on a custom port.
```
web_server:
  port: !secret web_server_port
  auth:
    username: !secret web_server_user
    password: !secret web_server_password
```
 
To resolve\test  I have tried V1 and V2 js and css and local and non local and explicily embedded V2 JS script but none of the settings appears to fix the issue although changing to V1 appears to slightly increase the chance of the webserver page displaying but this is not conclusive.

bkaufx, 2024-05-20:
Duplicate of https://github.com/esphome/issues/issues/5793 I believe.  Workaround listed here: https://github.com/esphome/issues/issues/5793#issuecomment-2119729825

MisterRadish, 2024-05-20:
> Duplicate of #5793 I believe. Workaround listed here: [#5793 (comment)](https://github.com/esphome/issues/issues/5793#issuecomment-2119729825)

Perfect!

nookbg, 2024-05-21:
With 2024.5.2. Problem still persist.

mad-tunes, 2024-05-21:
I'm still having similar trouble after updating to 2024.5.2 too

ssieb, 2024-05-26:
Duplicate of #5793

closed 2024-05-26 by ssieb