esphome/issues #5828: Ethernet Component compile library error with ESP32-S2 "error: 'esp_eth_mac_new_esp32' was not declared in this scope"

AndySymons, 2024-05-20:
### The problem

Compiling a config that uses the Ethernet Component with an ESP32-S2 board gives "error: 'esp_eth_mac_new_esp32' was not declared in this scope"
Looks like a missing library element?  

### Which version of ESPHome has the issue?

2024.4.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

N/a

### What platform are you using?

ESP32

### Board

WEMOS Lolin S2 Mini

### Component causing the issue

ESPHome compiler 

### Example YAML snippet

```yaml
esphome:
  name: hardwired-alarm-interface-4
  friendly_name: Hardwired Alarm Interface 4

esp32:
  board: lolin_s2_mini
  variant: ESP32S2
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "ZYc0E6QEx+ZapSDwoTtI985nky2YlY9dQFkb8SXEe20="

ota:
  password: "6763657d8379c0587de43ad3141ac26c"

ethernet:
  type: LAN8720
  mdc_pin: GPIO16
  mdio_pin: GPIO17
  clk_mode: GPIO0_IN
  phy_addr: 0
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.4.1
INFO Reading configuration /config/hardwired-alarm-interface-4.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing hardwired-alarm-interface-4 (board: lolin_s2_mini; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- noise-c @ 0.1.4
Compiling .pioenvs/hardwired-alarm-interface-4/src/esphome/components/ethernet/ethernet_component.cpp.o
Compiling .pioenvs/hardwired-alarm-interface-4/src/esphome/components/ota/ota_component.cpp.o
Compiling .pioenvs/hardwired-alarm-interface-4/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling .pioenvs/hardwired-alarm-interface-4/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling .pioenvs/hardwired-alarm-interface-4/src/esphome/components/socket/lwip_sockets_impl.cpp.o
Compiling .pioenvs/hardwired-alarm-interface-4/src/esphome/components/socket/socket.cpp.o
Compiling .pioenvs/hardwired-alarm-interface-4/src/esphome/core/application.cpp.o
Compiling .pioenvs/hardwired-alarm-interface-4/src/esphome/core/color.cpp.o
Compiling .pioenvs/hardwired-alarm-interface-4/src/esphome/core/component.cpp.o
Compiling .pioenvs/hardwired-alarm-interface-4/src/esphome/core/component_iterator.cpp.o
Compiling .pioenvs/hardwired-alarm-interface-4/src/esphome/core/controller.cpp.o
Compiling .pioenvs/hardwired-alarm-interface-4/src/esphome/core/entity_base.cpp.o
src/esphome/components/ethernet/ethernet_component.cpp: In member function 'virtual void esphome::ethernet::EthernetComponent::setup()':
src/esphome/components/ethernet/ethernet_component.cpp:129:24: error: 'esp_eth_mac_new_esp32' was not declared in this scope
   esp_eth_mac_t *mac = esp_eth_mac_new_esp32(&mac_config);
                        ^~~~~~~~~~~~~~~~~~~~~
src/esphome/components/ethernet/ethernet_component.cpp:129:24: note: suggested alternative: 'esp_eth_mac_new_w5500'
   esp_eth_mac_t *mac = esp_eth_mac_new_esp32(&mac_config);
                        ^~~~~~~~~~~~~~~~~~~~~
                        esp_eth_mac_new_w5500
*** [.pioenvs/hardwired-alarm-interface-4/src/esphome/components/ethernet/ethernet_component.cpp.o] Error 1
========================== [FAILED] Took 1.77 seconds ==========================
```

### Additional information

I also tried board types "esp32-s2-saola-1" (the default) and "micros2", with the same result. 
I also tried different pin numbers for the mdc and mdio pins, with the same result.

xttlegendapi, 2024-09-06:
same problem with esp s3 lolin mini. did you ever solve this @AndySymons

AndySymons, 2024-09-23:
I cannot fix it because it is a system problem. As far as I know it is still not possible to use the Ethernet component with an ESP32-S2 processor. I had to change the processor in my system to one that is not an S2. The Ethernet component works OK with a WROOM-32D, for example. 

For details of all my trials see my [project HARVIE Wiki page](https://github.com/AndySymons/HARVIE-Hardwired-Alarm-Replacement-Board/wiki/Introduction).