esphome/issues #5830: Issue with accuracy_decimals

roykrikke, 2024-05-20:
### The problem

https://esphome.io/components/sensor/index.html >> states:
_accuracy_decimals (Optional, int): Manually set the number of decimals to use when reporting values. This does not impact the actual value reported to Home Assistant, it just sets the number of decimals to use when displaying it._

The part: _"This does not impact the actual value reported to Home Assistant" in my view is not correct."_

When I use this this with modbus component:
```
sensor:
- platform: modbus_controller
  modbus_controller_id: ${device_name}
  name: "${device_name}_phase_1_line_to_neutral_volts"
  id: "${device_name}_phase_1_line_to_neutral_volts"
  <<: &volts_sensor
    address: 0
    register_type: read
    value_type: FP32
    register_count: 4
    unit_of_measurement: V
    device_class: voltage
    state_class: measurement
```
    
Output was:
`[20:21:49][D][sensor:094]: 'pv_phase_1_line_to_neutral_volts': Sending state 229.98114 V with 0 decimals of accuracy`

With this config I see 0 decimals in HA History of the entity.

But when is did this (i added _accuracy_decimals: 2_):
sensor:
```
- platform: modbus_controller
  modbus_controller_id: ${device_name}
  name: "${device_name}_phase_1_line_to_neutral_volts"
  id: "${device_name}_phase_1_line_to_neutral_volts"
  <<: &volts_sensor
    address: 0
    register_type: read
    value_type: FP32
    accuracy_decimals: 2
    register_count: 4
    unit_of_measurement: V
    device_class: voltage
    state_class: measurement
```
 
Output was:
`[20:48:36][D][sensor:094]: 'pv_phase_3_line_to_neutral_volts': Sending state 229.44279 V with 2 decimals of accuracy`

With this config I see 2 decimals in HA History of the entity.

Is something wrong or is the documentation outdated or am I missing the documentation somehow?

### Which version of ESPHome has the issue?

Core 2024.5.4 Supervisor 2024.05.1 Operating System 12.3 Frontend 20240501.1

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

roykrikke, 2024-05-20:
See also: https://github.com/esphome/issues/issues/5595

MrFr33man123, 2025-03-06:
i have a similar issue. dalls temp sensor is with 12bit precision and log says 23,34747 but send with 1 decimal. I cannot get Home assistant with 2 decimals of precision. Help?

ssieb, 2025-03-06:
Did you set the config for 2 decimals?

MrFr33man123, 2025-03-06:
yeah, i got it working with accuracy_decimals: 2 in the sensor config. Now HA receives 2 decimals.

gdt, 2025-04-22:
The essence of this issue is that the esphome documentation, while accurate about what goes across the esphome API, ends up misleading HA users.   esphome does send the full resolution, but it (99%) seems that the HA esphome integration rounds to the configured (or default) accuracy_decimals value before using and storing the value.    Thus one can reasonably expect "just set the number of digits in HA" to work, when it doesn't.   (Separately from that issue, I find that the default number of digits seems tuned for efficiency and fewer database writes, rather than extreme nerd data analysis.)