esphome/issues #5834: ESP32 never wakes up from deep sleep

devjklein, 2024-05-21:
### The problem

With a minimal configuration, I have set up a deep sleep component for a 45 second run duration and a 60 second sleep duration. The device logs these values and notes `[I][deep_sleep:119]: Sleeping for 60000000us`, but then never wakes up. I am checking the logs via serial.

### Which version of ESPHome has the issue?

2024.5.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.4

### What platform are you using?

ESP32

### Board

ESP32-WROOM-32

### Component causing the issue

deep_sleep

### Example YAML snippet

```yaml
esphome:
  name: weatherstation-garden
esp32:
  board: esp32dev
  framework:
    type: arduino

substitutions:
  loc: "Garden"
  name: "Garden Weather Station"

wifi:
  ssid: "Cat House Alpha"
  password: !secret wifi_pw
  fast_connect: true
  manual_ip:
    static_ip: xxx.xxx.xxx.xxx
    gateway: xxx.xxx.xxx.xxx
    subnet: xxx.xxx.xxx.xxx
    dns1: xxx.xxx.xxx.xxx
  ap:
    ssid: ${name} Fallback
    password: !secret fallback_pw

captive_portal:

logger:
  level: DEBUG

api:
  encryption:
    key: "xxx"

ota:
  password: !secret ota_pw
  
deep_sleep:
  run_duration: 45s
  sleep_duration: 60s
```

### Anything in the logs that might be useful for us?

```txt
[C][wifi:580]: WiFi:
[C][wifi:408]:   Local MAC: [redacted]
[C][wifi:413]:   SSID: [redacted]
[C][wifi:416]:   IP Address: [redacted]
[C][wifi:420]:   BSSID: [redacted]
[C][wifi:421]:   Hostname: 'weatherstation-garden'
[C][wifi:423]:   Signal strength: -53 dB ▂▄▆█
[C][wifi:427]:   Channel: 11
[C][wifi:428]:   Subnet: [redacted]
[C][wifi:429]:   Gateway: [redacted]
[C][wifi:430]:   DNS1: [redacted]
[C][wifi:431]:   DNS2: 0.0.0.0
[C][logger:185]: Logger:
[C][logger:186]:   Level: DEBUG
[C][logger:188]:   Log Baud Rate: 115200
[C][logger:189]:   Hardware UART: UART0
[C][captive_portal:088]: Captive Portal:
[C][mdns:115]: mDNS:
[C][mdns:116]:   Hostname: weatherstation-garden
[C][ota:096]: Over-The-Air Updates:
[C][ota:097]:   Address: [redacted]:3232
[C][ota:100]:   Using Password.
[C][ota:103]:   OTA version: 2.
[C][api:139]: API Server:
[C][api:140]:   Address: [redacted]:6053
[C][api:142]:   Using noise encryption: YES
[C][deep_sleep:049]: Setting up Deep Sleep...
[C][deep_sleep:052]:   Sleep Duration: 1 ms
[C][deep_sleep:055]:   Run Duration: 45000 ms
[D][api:102]: Accepted [redacted]
[W][component:170]: Component api cleared Warning flag
[D][api.connection:1321]: Home Assistant 2024.5.4 ([redacted]): Connected successfully
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][deep_sleep:117]: Beginning Deep Sleep
[I][deep_sleep:119]: Sleeping for 60000000us
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
```

### Additional information

Debugging I have tried:

- I've tried this code on multiple boards, including a new board.
- I have tried running via USB and the 3.3v pin with no change

Possibly related to #4451 and #2341

ssieb, 2024-05-21:
The logs do not match the config:
```
[C][deep_sleep:049]: Setting up Deep Sleep...
[C][deep_sleep:052]:   Sleep Duration: 1 ms
[C][deep_sleep:055]:   Run Duration: 45000 ms
```

devjklein, 2024-05-21:
> The logs do not match the config:
> 
> ```
> [C][deep_sleep:049]: Setting up Deep Sleep...
> [C][deep_sleep:052]:   Sleep Duration: 1 ms
> [C][deep_sleep:055]:   Run Duration: 45000 ms
> ```

Apologies, I grabbed the log from a test before where I set the duration super low to see if it was interpreting 1000 seconds or some other time somehow. The logs do read 60000ms in that location on my current config, but still no success.

devjklein, 2024-05-30:
Tested on a third esp32 board, and this is finally working. It appears I had two defective boards that for whatever reason could not wake from deep sleep. I do not know what would cause this issue other than a defective hardware component, so I am closing this issue.

closed 2024-05-30 by devjklein