esphome/issues #5836: DHT sensor config on wemos D1 mini keeps boot looping when accessing WiFi

liviupr, 2024-05-22:
### The problem

The bug appears on wemos D1 mini tested and working without any problems with other code, as example dallas.

By inserting the following code in the configuration, the wemos D1 mini keeps boot looping when accessing WiFi

# Example configuration entry
sensor:
  - platform: dht
    pin: D2
    temperature:
      name: "Living Room Temperature"
    humidity:
      name: "Living Room Humidity"
    update_interval: 60s

Here the log:
[10:06:37][C][dht:011]: Setting up DHT...
[10:06:37][C][wifi:038]: Setting up WiFi...
[10:06:37][C][wifi:051]: Starting WiFi...
[10:06:37][C][wifi:052]:   Local MAC: xxxxxxxxxxx
[10:06:37][D][wifi:462]: Starting scan...
[10:06:45]
[10:06:45] ets Jan  8 2013,rst cause:4, boot mode:(3,3)
[10:06:45]
[10:06:45]wdt reset
[10:06:45]load 0x4010f000, len 3424, room 16 
[10:06:45]tail 0
[10:06:45]chksum 0x2e
[10:06:45]load 0x3fff20b8, len 40, room 8 
[10:06:45]tail 0
[10:06:45]chksum 0x2b
[10:06:45]csum 0x2b
[10:06:45]v0007c830
[10:06:45]~ld
[10:06:45]����o�;��o|��l�{$#��|r�l�o��o�$`��r�$�l��[I][logger:034]: Log initialized
[10:06:45][C][ota:483]: There have been 9 suspected unsuccessful boot attempts.
[10:06:45][I][app:029]: Running through setup()...
[10:06:45][C][dallas.sensor:032]: Setting up DallasComponent...
[10:06:45][E][dallas.sensor:198]: Reading scratchpad failed: reset
[10:06:45][E][component:164]: Component dallas set Error flag: unspecified
[10:06:45][E][dallas.sensor:198]: Reading scratchpad failed: reset
[10:06:45][C][dht:011]: Setting up DHT...
[10:06:45][C][wifi:038]: Setting up WiFi...
[10:06:45][C][wifi:051]: Starting WiFi...
[10:06:45][C][wifi:052]:   Local MAC: xxxxxxxxxxx
[10:06:45][D][wifi:462]: Starting scan...
[10:06:46][D][sensor:093]: 'Temperature #1': Sending state nan °C with 1 decimals of accuracy
[10:06:46][D][sensor:093]: 'Temperature #2': Sending state nan °C with 1 decimals of accuracy
[10:06:53]
[10:06:53] ets Jan  8 2013,rst cause:4, boot mode:(3,3)
[10:06:53]
[10:06:53]wdt reset
[10:06:53]load 0x4010f000, len 3424, room 16 
[10:06:53]tail 0
[10:06:53]chksum 0x2e
[10:06:53]load 0x3fff20b8, len 40, room 8 
[10:06:53]tail 0
[10:06:54]chksum 0x2b
[10:06:54]csum 0x2b
[10:06:54]v0007c830
[10:06:54]~ld
[10:06:54]����g�r��g|��l�{$#��|r�l�n��n�d`��{�l�l��[I][logger:034]: Log initialized
[10:06:54][C][ota:483]: There have been 10 suspected unsuccessful boot attempts.
[10:06:54][E][ota:490]: Boot loop detected. Proceeding to safe mode.
[10:06:54][E][component:164]: Component ota set Error flag: unspecified
[10:06:54][I][app:029]: Running through setup()...

### Which version of ESPHome has the issue?

2024.5.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.4

### What platform are you using?

ESP8266

### Board

D1 mini

### Component causing the issue

DHT

### Example YAML snippet

```yaml
sensor:
  - platform: dht
    pin: D2
    temperature:
      name: "Living Room Temperature"
    humidity:
      name: "Living Room Humidity"
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
[10:06:37][C][dht:011]: Setting up DHT...
[10:06:37][C][wifi:038]: Setting up WiFi...
[10:06:37][C][wifi:051]: Starting WiFi...
[10:06:37][C][wifi:052]:   Local MAC: xxxxxxxxxxx
[10:06:37][D][wifi:462]: Starting scan...
[10:06:45]
[10:06:45] ets Jan  8 2013,rst cause:4, boot mode:(3,3)
[10:06:45]
[10:06:45]wdt reset
[10:06:45]load 0x4010f000, len 3424, room 16 
[10:06:45]tail 0
[10:06:45]chksum 0x2e
[10:06:45]load 0x3fff20b8, len 40, room 8 
[10:06:45]tail 0
[10:06:45]chksum 0x2b
[10:06:45]csum 0x2b
[10:06:45]v0007c830
[10:06:45]~ld
[10:06:45]����o�;��o|��l�{$#��|r�l�o��o�$`��r�$�l��[I][logger:034]: Log initialized
[10:06:45][C][ota:483]: There have been 9 suspected unsuccessful boot attempts.
[10:06:45][I][app:029]: Running through setup()...
[10:06:45][C][dallas.sensor:032]: Setting up DallasComponent...
[10:06:45][E][dallas.sensor:198]: Reading scratchpad failed: reset
[10:06:45][E][component:164]: Component dallas set Error flag: unspecified
[10:06:45][E][dallas.sensor:198]: Reading scratchpad failed: reset
[10:06:45][C][dht:011]: Setting up DHT...
[10:06:45][C][wifi:038]: Setting up WiFi...
[10:06:45][C][wifi:051]: Starting WiFi...
[10:06:45][C][wifi:052]:   Local MAC: xxxxxxxxxxx
[10:06:45][D][wifi:462]: Starting scan...
[10:06:46][D][sensor:093]: 'Temperature #1': Sending state nan °C with 1 decimals of accuracy
[10:06:46][D][sensor:093]: 'Temperature #2': Sending state nan °C with 1 decimals of accuracy
[10:06:53]
[10:06:53] ets Jan  8 2013,rst cause:4, boot mode:(3,3)
[10:06:53]
[10:06:53]wdt reset
[10:06:53]load 0x4010f000, len 3424, room 16 
[10:06:53]tail 0
[10:06:53]chksum 0x2e
[10:06:53]load 0x3fff20b8, len 40, room 8 
[10:06:53]tail 0
[10:06:54]chksum 0x2b
[10:06:54]csum 0x2b
[10:06:54]v0007c830
[10:06:54]~ld
[10:06:54]����g�r��g|��l�{$#��|r�l�n��n�d`��{�l�l��[I][logger:034]: Log initialized
[10:06:54][C][ota:483]: There have been 10 suspected unsuccessful boot attempts.
[10:06:54][E][ota:490]: Boot loop detected. Proceeding to safe mode.
[10:06:54][E][component:164]: Component ota set Error flag: unspecified
[10:06:54][I][app:029]: Running through setup()...
```

DENightOne, 2024-05-23:
I had a similar issue a while back, the ESP is "browing out" meaning the voltage falls to low so it reboots, my solution was to attach the DHT temp sensor to the 3.3v pin not the 5v pin.....DHT sensors can run on both voltages..... This fixed my problem.

liviupr, 2024-05-23:
My tests were made without any attached sensor.

liviupr, 2024-06-12:
Problem solved by using pin D4 GPIO02.

closed 2024-06-12 by liviupr