esphome/issues #5842: Error binary size: at Athom presense sensors

MarkuBu, 2024-05-23:
### The problem

I have several Athom presense sensors. Since the update to ESPHome 2024.5.1 or .2 I can't update the firmware. Latest installed version is 2024.5.0

Maybe I missed .1, so I'm not sure if .1 or .2 caused the issue at first

### Which version of ESPHome has the issue?

2024.5.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.4

### What platform are you using?

ESP8266

### Board

Athom presense sensor PS01

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.5.2
INFO Reading configuration /config/esphome/athom-presence-sensor-f9a20d.yaml...
INFO Detected timezone 'Europe/Oslo'
INFO Generating C++ source...
INFO Compiling app...
Processing athom-presence-sensor-f9a20d (board: esp8285; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- Wire @ 1.0
|-- ArduinoJson @ 6.18.5
RAM:   [=====     ]  49.1% (used 40240 bytes from 81920 bytes)
Flash: [======    ]  59.4% (used 608089 bytes from 1023984 bytes)
========================= [SUCCESS] Took 5.94 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.178.68
INFO Uploading /data/build/athom-presence-sensor-f9a20d/.pioenvs/athom-presence-sensor-f9a20d/firmware.bin (612240 bytes)
INFO Compressed to 414758 bytes
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)
```

whateverthatis, 2024-05-24:
I have the same issue...

mshrem, 2024-05-24:
Same issue here

seiferrtalle, 2024-05-24:
Same issue here. Does anyone have a workaround/solution? Thanx! BR, Ralf

bdleung, 2024-05-25:
> Same issue here. Does anyone have a workaround/solution? Thanx! BR, Ralf

A workout around is to flash via direct-connected USB port. A pain, but better than nothing.

whateverthatis, 2024-05-25:
The workaround I did is that I took the original .yaml form Athom's GitHub page and took out the webserver component and it did flash OTA and works as good as ever. I never used the webserver anyway as I'm using it through Home Assistant.

Here is my code:

`esphome:
  name: "presence-sensor"
  friendly_name: "Presence Sensor"

esp8266:
  board: esp8285
  restore_from_flash: true

preferences:
  flash_write_interval: 1min

logger:

api:
  encryption:
    key: "whoknows?"

ota:
  password: "whoknows?"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

light:
  - platform: status_led
    name: "Status LED"
    id: led
    pin: GPIO16
    disabled_by_default: true
    entity_category: config

i2c:
  sda: GPIO4
  scl: GPIO5
  scan: true

uart:
  tx_pin: GPIO15
  rx_pin: GPIO13
  baud_rate: 115200

binary_sensor:
  - platform: status
    name: "Status"

  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: true
    name: "Button"
    disabled_by_default: true
    on_multi_click:
      - timing:
          - ON for at least 4s
        then:
          - button.press: factory_reset_device

  - platform: gpio
    pin:
      number: GPIO14
      mode:
        input: true
[... 231 lines cut ...]
    internal: true
    on_press:
      - uart.write: "resetSystem\r\n"

  - platform: template
    name: Restart device
    entity_category: config
    on_press:
      - button.press: restart_mmwave
      - delay: 1s
      - button.press: restart_esp

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "IP Address"
    ssid:
      name: "Connected SSID"
    mac_address:
      name: "Mac Address"

time:
  - platform: sntp
    id: my_time
`

MarkuBu, 2024-05-25:
Thanks for the work around. I forked the repo, removed the web server config and changed the link in the config yaml to the new repo

https://github.com/MarkuBu/athom-configs/blob/main/athom-presence-sensor.yaml

```
substitutions:
  name: athom-presence-sensor-xxxx
  friendly_name: Presence
packages:
  Athom_Technology.Presence_Sensor: github://MarkuBu/athom-configs/athom-presence-sensor.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

Since this is just a work around and not a solution I leave this issue open

seiferrtalle, 2024-05-25:
Thanx at all for working on workarounds!!! 👍🏻

bdleung, 2024-05-28:
I was able to OTA from 5.3->5.4

closed 2024-05-28 by ssieb

MarkuBu, 2024-05-29:
Great, thanks. But I think I will continue without the Webserver. I just don't need it.