esphome/issues #5843: esphome - Install slimmelezer.yaml - compile fatal error

xvedar, 2024-05-23:
### The problem

Hi, I've got the slimmelezer dsmr device from here: https://github.com/zuidwijk/dsmr
I installed everything as the instruction on the page, everything is working well. But as this device is an esp home one I decided to add some extra logic, like restart button, I've updated the yaml file for that reason and now I'm trying to update the slimmelezer via the ESPHome page of the Home Assistant, this is where I click on 3 vertical dots of the slimmelezer and select 'Install' from the drop-down menu. Unfortunately it brings me no result as the compile step exits with errors as displayed below. The full compile log is attached.

`api_frame_helper.h:2:10: fatal error:`

Please advise. Thanks.

### Which version of ESPHome has the issue?

2023.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.4

### What platform are you using?

ESP8266

### Board

Raspberry 4

### Component causing the issue

slimmelezer

### Example YAML snippet

```yaml
---
substitutions:
  device_name: slimmelezer
    
esphome:
  name: ${device_name}
  platform: ESP8266
  esp8266_restore_from_flash: true
  board: d1_mini
  name_add_mac_suffix: false
  project:
    name: zuidwijk.slimmelezer
    version: "1.2"
  on_boot:
    then:
      - if:
          condition:
            lambda: return id(has_key);
          then:
            - lambda: |-
                std::string key(id(stored_decryption_key), 32);
                id(dsmr_instance).set_decryption_key(key);
          else:
            - logger.log:
                level: info
                format: "Not using decryption key. If you need to set a key use Home Assistant service 'ESPHome:  ${device_name}_set_dsmr_key'"

switch:
  - platform: restart
    name: "Slimmelezer Restart Switch"
  
wifi:
  # remove leading '#' and fill in your wifi details
  ssid: !secret wifi_ssid 
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}

captive_portal:

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  services:
    service: set_dsmr_key
    variables:
      private_key: string
    then:
      - logger.log:
          format: Setting private key %s. Set to empty string to disable
          args: [private_key.c_str()]
      - globals.set:
          id: has_key
          value: !lambda "return private_key.length() == 32;"
      - lambda: |-
          if (private_key.length() == 32)
            private_key.copy(id(stored_decryption_key), 32);
          id(dsmr_instance).set_decryption_key(private_key);

ota:

dashboard_import:
  package_import_url: github://zuidwijk/dsmr/slimmelezer.yaml@main
  import_full_config: true

web_server:
  port: 80

uart:
  baud_rate: 115200
  rx_pin: D7
  rx_buffer_size: 1700

globals:
  - id: has_key
    type: bool
    restore_value: yes
    initial_value: "false"
  - id: stored_decryption_key
    type: char[32]
    restore_value: yes

dsmr:
  id: dsmr_instance
  max_telegram_length: 1700
  # For Luxembourg users set here your decryption key
  #decryption_key: !secret decryption_key // enable this when using decryption for Luxembourg; key like '00112233445566778899AABBCCDDEEFF'

sensor:
  - platform: dsmr
    energy_delivered_lux:
      name: "Energy Consumed Luxembourg"
    energy_delivered_tariff1:
      name: "Energy Consumed Tariff 1"
    energy_delivered_tariff2:
      name: "Energy Consumed Tariff 2"
    energy_returned_lux:
      name: "Energy Produced Luxembourg"
    energy_returned_tariff1:
      name: "Energy Produced Tariff 1"
    energy_returned_tariff2:
      name: "Energy Produced Tariff 2"
    power_delivered:
      name: "Power Consumed"
      accuracy_decimals: 3
    power_returned:
      name: "Power Produced"
      accuracy_decimals: 3
    electricity_failures:
      name: "Electricity Failures"
      icon: mdi:alert
    electricity_long_failures:
      name: "Long Electricity Failures"
      icon: mdi:alert
    voltage_l1:
      name: "Voltage Phase 1"
    voltage_l2:
      name: "Voltage Phase 2"
    voltage_l3:
      name: "Voltage Phase 3"
    current_l1:
      name: "Current Phase 1"
    current_l2:
      name: "Current Phase 2"
    current_l3:
      name: "Current Phase 3"
    power_delivered_l1:
      name: "Power Consumed Phase 1"
      accuracy_decimals: 3
    power_delivered_l2:
      name: "Power Consumed Phase 2"
      accuracy_decimals: 3
    power_delivered_l3:
      name: "Power Consumed Phase 3"
      accuracy_decimals: 3
    power_returned_l1:
      name: "Power Produced Phase 1"
      accuracy_decimals: 3
    power_returned_l2:
      name: "Power Produced Phase 2"
      accuracy_decimals: 3
    power_returned_l3:
      name: "Power Produced Phase 3"
      accuracy_decimals: 3
    gas_delivered:
      name: "Gas Consumed"
    gas_delivered_be:
      name: "Gas Consumed Belgium"
  - platform: uptime
    name: "SlimmeLezer Uptime"
  - platform: wifi_signal
    name: "SlimmeLezer Wi-Fi Signal"
    update_interval: 60s

text_sensor:
  - platform: dsmr
    identification:
      name: "DSMR Identification"
    p1_version:
      name: "DSMR Version"
    p1_version_be:
      name: "DSMR Version Belgium"
  - platform: wifi_info
    ip_address:
      name: "SlimmeLezer IP Address"
    ssid:
      name: "SlimmeLezer Wi-Fi SSID"
    bssid:
      name: "SlimmeLezer Wi-Fi BSSID"
  - platform: version
    name: "ESPHome Version"
    hide_timestamp: true
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.5.2
INFO Reading configuration /config/esphome/slimmelezer.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing slimmelezer (board: d1_mini; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- ArduinoJson @ 6.18.5
|-- Dsmr @ 0.8.0
|-- Crypto @ 0.4.0
Compiling .pioenvs/slimmelezer/src/esphome/components/api/api_connection.cpp.o
In file included from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/api_frame_helper.h:2:10: fatal error: cstdint: No such file or directory
    2 | #include <cstdint>
      |          ^~~~~~~~~
compilation terminated.
Compiling .pioenvs/slimmelezer/src/esphome/components/api/api_frame_helper.cpp.o
*** [.pioenvs/slimmelezer/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from src/esphome/components/api/api_frame_helper.cpp:1:
src/esphome/components/api/api_frame_helper.h:2:10: fatal error: cstdint: No such file or directory
    2 | #include <cstdint>
      |          ^~~~~~~~~
compilation terminated.
*** [.pioenvs/slimmelezer/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
========================== [FAILED] Took 3.64 seconds ==========================
```

ssieb, 2024-05-23:
Why are you using such an old version of esphome?
There's something wrong with the install, so try re-installing the addon or try updating it to the latest version first.

ssieb, 2024-05-23:
No, I just noticed that your logs don't match the version you claimed it to be.
Try re-installing the addon.

xvedar, 2024-05-24:
Thanks @ssieb for the feedback, re-installing esphome addon solved the issue.

closed 2024-05-24 by xvedar