esphome/issues #5845: Latest version of Voice Assistant Installation on M5 Stack Atom Echo is causing a BLE Crash bootloop

MonsterVic, 2024-05-24:
### The problem

Trying to install Voice Assistant from https://esphome.io/projects/index.html  to an M5 Stack Atom Echo, and after the installation it gets stuck in scanning Wifi networks and that's it, if I'm checking the logs it stuck in a bootloop BLE is crashing. 
The device is working well with Empty ESP Home.   There seems to be a problem with the latest version of the Voice Assistant Binary.  I saw others are having the same issue, saw it on facebook, saw it on reddit and saw it discord as well.  

### Which version of ESPHome has the issue?

The latest you have on https://esphome.io/projects/index.html

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.4

### What platform are you using?

ESP32

### Board

M5 Stack Atom Echo

tsbwn, 2024-05-24:
I'm experiencing the same with the same hardware. I tried binary advertised here https://esphome.io/projects/ and this config on a working online device https://github.com/esphome/firmware/blob/main/voice-assistant/m5stack-atom-echo.yaml. 
In both cases the device is stuck in a boot loop with the following errors:
```
[D][esp32_ble_server:077]: BLE server setup successfully[0
assert failed: xQueueGenericSend queue.c:820 (pxQueue)

Backtrace: 0x400835ee:0x3ffd1e00 0x40092981:0x3ffd1e20 0x4009930d:0x3ffd1e40 0x40093392:0x3ffd1f60 0x400dae9f:0x3ffd1fa0 0x400dbf79:0x3ffd1fe0 0x400dbfe2:0x3ffd2050 0x400dc308:0x3ffd20a0 0x400dc8e6:0x3ffd20f0 0x401de8e9:0x3ffd2210 0x401de94d:0x3ffd2230 0x400e8c18:0x3ffd2250 0x400ebe6e:0x3ffd2280 0x400d91a6:0x3ffd22a0
```
or
```
[D][esp32_ble_server:077]: BLE server setup successfullyuru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400dacce  PS      : 0x00060830  A0      : 0x800dc323  A1      : 0x3ffd2050  
A2      : 0x00000000  A3      : 0x3ffd20a0  A4      : 0x00000005  A5      : 0x97422d7c  
A6      : 0x3ffff7c8  A7      : 0x3ffd336c  A8      : 0x00000000  A9      : 0x3ffd2010  
A10     : 0x3ffff954  A11     : 0x00000064  A12     : 0x00000000  A13     : 0x3ffff95c  
A14     : 0x00000000  A15     : 0x800d0000  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400daccb:0x3ffd2050 0x400dc320:0x3ffd20a0 0x400dc8e6:0x3ffd20f0 0x401de8e9:0x3ffd2210 0x401de94d:0x3ffd2230 0x400e8c18:0x3ffd2250 0x400ebe6e:0x3ffd2280 0x400d91a6:0x3ffd22a0
```

wsheppa21, 2024-05-25:
I am having the same issue. Nothing appears to be loading. Tried this with 3 ATOMS and seeing the same issues.

Taomyn, 2024-05-25:
v2024.5.3 also broke my Atom, caused it to loop on buffer initialisation for the microphone - reverted ESPHome to 2024.5.2 to fix it.

davidrpfarinha, 2024-05-25:
> v2024.5.3 also broke my Atom, caused it to loop on buffer initialisation for the microphone - reverted ESPHome to 2024.5.2 to fix it.

How did you revert it back? With the ESPHome plus an older yaml config for the Atom?

Taomyn, 2024-05-25:
> > v2024.5.3 also broke my Atom, caused it to loop on buffer initialisation for the microphone - reverted ESPHome to 2024.5.2 to fix it.
> 
> How did you revert it back? With the ESPHome plus an older yaml config for the Atom?

No, I restored the backup HA took before the update, then restarted HA, and then re-installed onto the Atom.

Ceralor, 2024-10-23:
Not stale, ongoing, but zero response.

Ceralor, 2024-10-23:
This WAS working for me on ESPHome 2024.9, but on 2024.10 this crashes. This is even building myself. I thought it might be the display I was including but I removed it and the issue persists.

```
[11:36:27][D][binary_sensor:036]: 'Button': Sending state ON
[11:36:27][D][binary_sensor:036]: 'Button': Sending state OFF
[11:36:27][D][voice_assistant:510]: State changed from IDLE to START_MICROPHONEGuru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
[11:36:27]
[11:36:27]Core  1 register dump:
[11:36:27]PC      : 0x4000c433  PS      : 0x00060e30  A0      : 0x800e13a1  A1      : 0x3ffc0520
[11:36:27]A2      : 0xbf0322af  A3      : 0x00000000  A4      : 0x00000400  A5      : 0xbf0322af  
[11:36:27]A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800e71d8  A9      : 0x3ffc04a0
[11:36:27]A10     : 0x3ffb9d70  A11     : 0x00000005  A12     : 0x3f405adc  A13     : 0x000000de
[11:36:27]A14     : 0x3f405dd1  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001d
[11:36:27]EXCVADDR: 0xbf0322af  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000
[11:36:27]
[11:36:27]
[11:36:27]Backtrace: 0x4000c430:0x3ffc0520 0x400e139e:0x3ffc0530 0x400e217b:0x3ffc0550 0x40186211:0x3ffc0600 0x40186275:0x3ffc0620 0x400e5680:0x3ffc0640 0x400e857a:0x3ffc0670 0x400d9f42:0x3ffc0690
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400e139e: esphome::voice_assistant::VoiceAssistant::clear_buffers_() at C:\Users\Kay\code\esphome\esphome_configs\.esphome\build\m5stack-atom-echo-8046b4/src/esphome/components/voice_assistant/voice_assistant.cpp:122
WARNING Decoded 0x400e217b: esphome::voice_assistant::VoiceAssistant::loop() at C:\Users\Kay\code\esphome\esphome_configs\.esphome\build\m5stack-atom-echo-8046b4/src/esphome/components/voice_assistant/voice_assistant.cpp:230
WARNING Decoded 0x40186211: esphome::Component::call_loop() at C:\Users\Kay\code\esphome\esphome_configs\.esphome\build\m5stack-atom-echo-8046b4/src/esphome/core/component.cpp:77
WARNING Decoded 0x40186275: esphome::Component::call() at C:\Users\Kay\code\esphome\esphome_configs\.esphome\build\m5stack-atom-echo-8046b4/src/esphome/core/component.cpp:104
WARNING Decoded 0x400e5680: esphome::Application::loop() at C:\Users\Kay\code\esphome\esphome_configs\.esphome\build\m5stack-atom-echo-8046b4/src/esphome/core/application.cpp:74 (discriminator 2)
WARNING Decoded 0x400e857a: loop() at C:\Users\Kay\code\esphome\esphome_configs\.esphome\build\m5stack-atom-echo-8046b4/.\bedroom-speaker.yaml:234
WARNING Decoded 0x400d9f42: esphome::loop_task(void*) at C:\Users\Kay\code\esphome\esphome_configs\.esphome\build\m5stack-atom-echo-8046b4/src/esphome/components/esp32/core.cpp:69 (discriminator 1)
```

antraynl, 2024-10-23:
Same experience for me on my m5echo: it was working with 2024.9.x, but panics on 2024.10.x:
Also performed clean build files and installed using usb instead of using wireless, same result:

[D][api.connection:1446]: Home Assistant 2024.10.1 (xxx.xxx.xxx.xxx): Connected successfully
[D][voice_assistant:510]: State changed from IDLE to START_MICROPHONE
[D][voice_assistant:516]: Desired state set to WAIT_FOR_VAD
[D][light:036]: 'M5Stack Atom Echo 213bGuru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
[00:50:54]
[00:50:54]Core  1 register dump:
[00:50:54]PC      : 0x4000c46c  PS      : 0x00060e30  A0      : 0x800e58a5  A1      : 0x3ffd09f0  
[00:50:54]A2      : 0x00060930  A3      : 0x00000000  A4      : 0x00000400  A5      : 0x00060930  
[00:50:54]A6      : 0x3fff1c14  A7      : 0x00000040  A8      : 0x800ebdc4  A9      : 0x3ffd0970  
[00:50:54]A10     : 0x3ffbd680  A11     : 0x00000005  A12     : 0x3f403f0c  A13     : 0x000000de  
[00:50:54]A14     : 0x3f404201  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001d  
[00:50:54]EXCVADDR: 0x00060930  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x0000003f  
[00:50:54]
[00:50:54]
[00:50:54]Backtrace: 0x4000c469:0x3ffd09f0 0x400e58a2:0x3ffd0a00 0x400e6683:0x3ffd0a20 0x401f3e55:0x3ffd0ad0 0x401f3eb9:0x3ffd0af0 0x400ea17c:0x3ffd0b10 0x400ed2a2:0x3ffd0b40 0x400da2ba:0x3ffd0b60
[00:50:54]
[00:50:54]
[00:50:54]
[00:50:54]
[00:50:54]ELF file SHA256: fb3f7323cccc7969
[00:50:54]
[00:50:54]Rebooting...

antraynl, 2024-10-24:
2024-10.2 is available addressing an issue with voice assistant, so the issue with 2024-10.1 mentioned here might be solved. Will check tonight.
Details can be found on https://github.com/esphome/esphome/pull/7662