esphome/issues #5847: Compiler Warnings since 2024.5.x

tomatenkopp, 2024-05-25:
### The problem

during the recent updates of my ESP based nsPanel the compiler shows the following warnings.
Everythings works fine, but that should be fixed for future reliability.
Compiling .pioenvs/nspanel1/src/esphome/components/api/api_pb2.o ->see below...
Compiling .pioenvs/nspanel1/bootloader_support/src/esp_image_format.o -> see below...

Processing nspanel1 (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40407.0 (4.4.7) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Creating a new virtual environment for IDF Python dependencies
Installing ESP-IDF's Python dependencies

....

Reading CMake configuration...
Generating assembly for certificate bundle...
Dependency Graph
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/nspanel1/src/esphome/components/adc/adc_sensor.o
Compiling .pioenvs/nspanel1/src/esphome/components/api/api_connection.o
Compiling .pioenvs/nspanel1/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/nspanel1/src/esphome/components/api/api_pb2.o
src/esphome/components/adc/adc_sensor.cpp: In member function 'virtual void esphome::adc::ADCSensor::setup()':
src/esphome/components/adc/adc_sensor.cpp:65:28: warning: 'ADC_ATTEN_DB_11' is deprecated [-Wdeprecated-declarations]
   for (int32_t i = 0; i <= ADC_ATTEN_DB_11; i++) {
                            ^~~~~~~~~~~~~~~
In file included from /data/cache/platformio/packages/framework-espidf/components/driver/include/driver/adc.h:14,
                 from src/esphome/components/adc/adc_sensor.h:10,
                 from src/esphome/components/adc/adc_sensor.cpp:1:
/data/cache/platformio/packages/framework-espidf/components/hal/include/hal/adc_types.h:54:5: note: declared here
     ADC_ATTEN_DB_11 __attribute__((deprecated)) = ADC_ATTEN_DB_12,  ///<This is deprecated, it behaves the same as `ADC_ATTEN_DB_12`
     ^~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp: In member function 'virtual void esphome::adc::ADCSensor::dump_config()':
src/esphome/components/adc/adc_sensor.cpp:121:12: warning: 'ADC_ATTEN_DB_11' is deprecated [-Wdeprecated-declarations]
       case ADC_ATTEN_DB_11:
            ^~~~~~~~~~~~~~~
In file included from /data/cache/platformio/packages/framework-espidf/components/driver/include/driver/adc.h:14,
                 from src/esphome/components/adc/adc_sensor.h:10,
                 from src/esphome/components/adc/adc_sensor.cpp:1:
/data/cache/platformio/packages/framework-espidf/components/hal/include/hal/adc_types.h:54:5: note: declared here
     ADC_ATTEN_DB_11 __attribute__((deprecated)) = ADC_ATTEN_DB_12,  ///<This is deprecated, it behaves the same as `ADC_ATTEN_DB_12`
     ^~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp: In member function 'virtual float esphome::adc::ADCSensor::sample()':
src/esphome/components/adc/adc_sensor.cpp:203:48: warning: 'ADC_ATTEN_DB_11' is deprecated [-Wdeprecated-declarations]
     adc1_config_channel_atten(this->channel1_, ADC_ATTEN_DB_11);
                                                ^~~~~~~~~~~~~~~
In file included from /data/cache/platformio/packages/framework-espidf/components/driver/include/driver/adc.h:14,
                 from src/esphome/components/adc/adc_sensor.h:10,
                 from src/esphome/components/adc/adc_sensor.cpp:1:
/data/cache/platformio/packages/framework-espidf/components/hal/include/hal/adc_types.h:54:5: note: declared here
     ADC_ATTEN_DB_11 __attribute__((deprecated)) = ADC_ATTEN_DB_12,  ///<This is deprecated, it behaves the same as `ADC_ATTEN_DB_12`
     ^~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:218:48: warning: 'ADC_ATTEN_DB_11' is deprecated [-Wdeprecated-declarations]
     adc2_config_channel_atten(this->channel2_, ADC_ATTEN_DB_11);
                                                ^~~~~~~~~~~~~~~
In file included from /data/cache/platformio/packages/framework-espidf/components/driver/include/driver/adc.h:14,
                 from src/esphome/components/adc/adc_sensor.h:10,
                 from src/esphome/components/adc/adc_sensor.cpp:1:
/data/cache/platformio/packages/framework-espidf/components/hal/include/hal/adc_types.h:54:5: note: declared here
     ADC_ATTEN_DB_11 __attribute__((deprecated)) = ADC_ATTEN_DB_12,  ///<This is deprecated, it behaves the same as `ADC_ATTEN_DB_12`
     ^~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:238:91: warning: 'ADC_ATTEN_DB_11' is deprecated [-Wdeprecated-declarations]
   uint32_t mv11 = esp_adc_cal_raw_to_voltage(raw11, &this->cal_characteristics_[(int32_t) ADC_ATTEN_DB_11]);
                                                                                           ^~~~~~~~~~~~~~~
[... 135 lines cut ...]
/data/packages/566b10a3/esphome/nspanel_esphome_core.yaml: In function 'void setup()':
/data/packages/566b10a3/esphome/nspanel_esphome_core.yaml:1623:31: warning: 'ADC_ATTEN_DB_11' is deprecated [-Wdeprecated-declarations]
     internal: true
                               ^              
In file included from /data/cache/platformio/packages/framework-espidf/components/driver/include/driver/adc.h:14,
                 from src/esphome/components/adc/adc_sensor.h:10,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
/data/cache/platformio/packages/framework-espidf/components/hal/include/hal/adc_types.h:54:5: note: declared here
     ADC_ATTEN_DB_11 __attribute__((deprecated)) = ADC_ATTEN_DB_12,  ///<This is deprecated, it behaves the same as `ADC_ATTEN_DB_12`
     ^~~~~~~~~~~~~~~

....

Creating esp32 image...
Successfully created esp32 image.
Linking .pioenvs/nspanel1/firmware.elf
RAM:   [=         ]   7.5% (used 24564 bytes from 327680 bytes)
Flash: [======    ]  62.9% (used 1155077 bytes from 1835008 bytes)
Building .pioenvs/nspanel1/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/nspanel1/firmware.bin"], [".pioenvs/nspanel1/firmware.elf"])
Wrote 0x12b9b0 bytes to file /data/build/nspanel1/.pioenvs/nspanel1/firmware-factory.bin, ready to flash to offset 0x0
```

mattlward, 2025-02-21:
So, I am getting the same form of error.  I have do a "clean build" with no change.  It all seems to work, but I do not like seeing the errors.

Core 2025.1.4
OS 14.2
Supervisor 2025.02.1
ESPHome 2025.2.0

On bare metal Gen8 i5

Thanks Matt

Compiling .pioenvs/esphome-web-8281ec/src/main.cpp.o
In file included from src/esphome/components/adc/adc_sensor.h:9,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
/data/cache/platformio/packages/framework-espidf/components/esp_adc/deprecated/include/esp_adc_cal.h:17:2: warning: #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h" [-Wcpp]
   17 | #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h"
      |  ^~~~~~~
In file included from src/esphome/components/adc/adc_sensor.h:10:
/data/cache/platformio/packages/framework-espidf/components/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
In file included from src/esphome/components/pulse_counter/pulse_counter_sensor.h:10,
                 from src/esphome/components/pulse_counter/automation.h:5,
                 from src/esphome.h:57:
/data/cache/platformio/packages/framework-espidf/components/driver/deprecated/driver/pcnt.h:15:2: warning: #warning "legacy pcnt driver is deprecated, please migrate to use driver/pulse_cnt.h" [-Wcpp]
   15 | #warning "legacy pcnt driver is deprecated, please migrate to use driver/pulse_cnt.h"
      |  ^~~~~~~
Linking .pioenvs/esphome-web-8281ec/firmware.elf
RAM:   [=         ]  10.1% (used 32988 bytes from 327680 bytes)
Flash: [=====     ]  53.3% (used 978257 bytes from 1835008 bytes)

WinfriedTheis-DoE, 2025-06-18:
> Those are just warnings.  The compiler is getting killed because you're out of memory.

Thank you, but how do I resolve that on my HA Yellow?!?