esphome/issues #5849: Could not allocate speaker buffer starting Microphone on M5 Stack Echo

cvwillegen, 2024-05-25:
### The problem

After update to 2024.5.3, M5Stack Atom Echo devices fail to allocate a microphone buffer. This causes the device to not function as intended.

### Which version of ESPHome has the issue?

2024.5.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.5

### What platform are you using?

ESP8266

### Board

m5stack.atom-echo

### Component causing the issue

Microphone / Voice Assistant

### Example YAML snippet

```yaml
substitutions:
  name: "m5stack-atom-echo"
packages:
  #m5stack.atom-echo: github://esphome/media-players/m5stack-atom-echo.yaml@main
  m5stack.atom-echo: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false

api:
  encryption:
    key: "<redacted>"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
~/INFO ESPHome 2024.5.3
INFO Reading configuration /config/esphome/m5stack-atom-echo.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Starting log output from <redacted> using esphome API
INFO Successfully connected to m5stack-atom-echo @ <redacted> in 0.089s
INFO Successful handshake with m5stack-atom-echo @ <redacted> in 0.096s
[10:07:37][D][voice_assistant:220]: Starting Microphone
[10:07:37][W][voice_assistant:086]: Could not allocate speaker buffer
[10:07:37][I][app:100]: ESPHome version 2024.5.3 compiled on May 25 2024, 08:44:30
[10:07:37][I][app:102]: Project m5stack.atom-echo-voice-assistant version 1.0
[10:07:37][C][wifi:580]: WiFi:
[10:07:37][C][wifi:408]:   Local MAC: <redacted>
[10:07:37][C][wifi:413]:   SSID: '<redacted>'[redacted]
[10:07:37][C][wifi:416]:   IP Address: <redacted>
[10:07:37][C][wifi:420]:   BSSID: <redacted>[redacted]
[10:07:37][C][wifi:421]:   Hostname: '<redacted>'
[10:07:37][C][wifi:423]:   Signal strength: <redacted>
[10:07:37][C][wifi:427]:   Channel: <redacted>
[... 526 lines cut ...]
[10:07:40][W][voice_assistant:086]: Could not allocate speaker buffer
[10:07:40][D][voice_assistant:220]: Starting Microphone
[10:07:40][W][voice_assistant:086]: Could not allocate speaker buffer
[10:07:40][D][voice_assistant:220]: Starting Microphone
[10:07:40][W][voice_assistant:086]: Could not allocate speaker buffer
[10:07:40][D][voice_assistant:220]: Starting Microphone
[10:07:40][W][voice_assistant:086]: Could not allocate speaker buffer
[10:07:40][D][voice_assistant:220]: Starting Microphone
[10:07:40][W][voice_assistant:086]: Could not allocate speaker buffer
[10:07:40][D][voice_assistant:220]: Starting Microphone
[10:07:40][W][voice_assistant:086]: Could not allocate speaker buffer
[10:07:40][D][voice_assistant:220]: Starting Microphone
[10:07:40][W][voice_assistant:086]: Could not allocate speaker buffer
[10:07:40][D][voice_assistant:220]: Starting Microphone
[10:07:40][W][voice_assistant:086]: Could not allocate speaker buffer
[10:07:40][D][voice_assistant:220]: Starting Microphone
[10:07:40][W][voice_assistant:086]: Could not allocate speaker buffer
[10:07:40][D][voice_assistant:220]: Starting Microphone
[10:07:40][W][voice_assistant:086]: Could not allocate speaker buffer
[10:07:40][D][voice_assistant:220]: Starting Microphone
[10:07:40][W][voice_assistant:086]: Could not allocate speaker buffer
[10:07:40][D][voice_assistant:220]: Starting Microphone
[10:07:40][W][voice_assistant:086]: Could not allocate speaker buffer
[10:07:40][D][voice_assistant:220]: Starting Microphone
```

technois2, 2024-05-25:
Same issue here.

dmt408, 2024-05-25:
i started having this issue yesterday. if i restart the atom echo a few times it starts to behave properly.

Taomyn, 2024-05-25:
v2024.5.3 also broke my Atom with this problem - reverted ESPHome to 2024.5.2 to fix it.

scharmach, 2024-05-25:
Same issue here, also reverted back to 2025.5.2 for the time being. Since this was the only change merged for 2024.5.3, it must have been broken by this:

https://github.com/esphome/esphome/pull/6800

@jesserockz?

HLandstrom, 2024-05-26:
Same issue here, started happening with 2024.5.3

MonsterVic, 2024-05-26:
M5 Echo, Same issue, #6800 solved the BLE Bootloop error, but gettin "Could not allocate speaker buffer"

bschokker, 2024-05-27:
Exactly the same behavior:

10:07:40][D][voice_assistant:220]: Starting Microphone
[10:07:40][W][voice_assistant:086]: Could not allocate speaker buffer

HA version: 2024.5.5

nem0nic, 2024-05-27:
Commenting to add another confirmation of the same issue.  Unfortunately though when the error starts it hangs up my ESPHome page so I can't get back to the logs to grab and paste here.

nielsnl68, 2024-05-27:
could you try to use pr#6718

```yaml
external_components:
  - source: github://pr#6718
    components:
      - voice_assistant
      - i2s_audio
      - speaker
```

nem0nic, 2024-05-27:
> could you try to use pr#6718
> 
> ```yaml
> external_components:
>   - source: github://pr#6718
>     components:
>       - voice_assistant
>       - i2s_audio
>       - speaker
> ```

Where now do you need me to put that?  I'm not yet a wiz at this.

nielsnl68, 2024-05-28:
You can put it everywhere you want in your yaml file.

nem0nic, 2024-05-28:
Tried putting it at the bottom of the default yaml for the Atom Echo (listed above in comment #1) and it throws the following error

`INFO ESPHome 2024.5.3
INFO Reading configuration /config/esphome/m5stack-atom-echo-30ed84.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
Failed config

wifi: [source /data/packages/c46f54c1/voice-assistant/m5stack-atom-echo.yaml:23]
  on_connect: 
    - delay: 5s
    - ble.disable: {}
  on_disconnect: 
    - ble.enable: {}
  ap: 
  ssid: 11210OT
  password: N!ch0l@SJa
  
  [external_components] is an invalid option for [wifi]. Please check the indentation.
  external_components: 
    - source: github://pr#6718
      components: 
        - voice_assistant
        - i2s_audio
        - speaker`

nem0nic, 2024-05-28:
I put it at the bottom of the yaml (with the correct indentation this time) and this is the error I get:

`INFO ESPHome 2024.5.3
INFO Reading configuration /config/esphome/m5stack-atom-echo-30ed84.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1065, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1052, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 479, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 193, in write_cpp
    return write_cpp_file()
           ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 211, in write_cpp_file
    writer.write_cpp(code_s)
  File "/esphome/esphome/writer.py", line 344, in write_cpp
    copy_src_tree()
  File "/esphome/esphome/writer.py", line 297, in copy_src_tree
    copy_files()
  File "/esphome/esphome/components/esp32/__init__.py", line 697, in copy_files
    shutil.copytree(
  File "/usr/lib/python3.11/shutil.py", line 561, in copytree
    return _copytree(entries=entries, src=src, dst=dst, symlinks=symlinks,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/shutil.py", line 515, in _copytree
    raise Error(errors)
shutil.Error: [('/data/idf_components/b0ad5728/components/esp-sr/docs/myapp/lib64', '/data/build/m5stack-atom-echo-30ed84/components/esp-sr/docs/myapp/lib64', "[Errno 17] File exists: 'lib' -> '/data/build/m5stack-atom-echo-30ed84/components/esp-sr/docs/myapp/lib64'"), ('/data/idf_components/b0ad5728/components/esp-sr/docs/myapp/bin/python3', '/data/build/m5stack-atom-echo-30ed84/components/esp-sr/docs/myapp/bin/python3', "[Errno 17] File exists: '/usr/bin/python3' -> '/data/build/m5stack-atom-echo-30ed84/components/esp-sr/docs/myapp/bin/python3'"), ('/data/idf_components/b0ad5728/components/esp-sr/docs/myapp/bin/python', '/data/build/m5stack-atom-echo-30ed84/components/esp-sr/docs/myapp/bin/python', "[Errno 17] File exists: 'python3' -> '/data/build/m5stack-atom-echo-30ed84/components/esp-sr/docs/myapp/bin/python'")]`

seang96, 2024-05-28:
using pr#6718 and it works again.

nem0nic, 2024-05-28:
I got it sorted where to put it.  Receive a new error but voice assistant still worked.

[20:24:39][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:39][W][component:157]: Component i2s_audio.speaker set Warning flag: unspecified
[20:24:39][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:39][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:39][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:39][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:39][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:39][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:39][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:40][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:40][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:40][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:40][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:40][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:40][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:40][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:40][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:40][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:40][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:41][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:41][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:41][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:41][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:41][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:41][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:41][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:41][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:41][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:41][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:42][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:42][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:42][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:42][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:42][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:42][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:42][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:42][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:42][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:42][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:43][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:43][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:43][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:43][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:43][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:43][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:43][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:43][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:43][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:43][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:44][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:44][W][i2s_audio.speaker:207]: Error writing to I2S: ESP_OK
[20:24:44][W][component:170]: Component i2s_audio.speaker cleared Warning flag

nielsnl68, 2024-05-28:
can you show your Yaml code?

@gnumpi could you have a look above, this is an issue i saw as well using the rtttl component as well after closing.

nem0nic, 2024-05-28:
> can you show your Yaml code?
> 
> @gnumpi could you have a look above, this is an issue i saw as well using the rtttl component as well after closing.

substitutions:
  name: m5stack-atom-echo-30ed84
  friendly_name: Dev Atom
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
external_components:
- source: github://pr#6718
  components:
    - voice_assistant
    - i2s_audio
    - speaker
api:
  encryption:
    key: eJ316/MbwuAeMu3gY1o2cxuwRO9Esxu5FeT/5RlXGV0=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

sampathdassanayake, 2024-05-28:
I have the same issue with 2024.5.4 as well.

gnumpi, 2024-05-28:
> @gnumpi could you have a look above, this is an issue i saw as well using the rtttl component as well after closing.

Regarding pr#6718:
This warning is triggered when the the speaker has been started but it didn't receive any input for more than 100ms. 
The warning message itself is nonsense though, I will try to change it. Probably it is also necessary to increase the threshold value a bit.

nem0nic, 2024-05-28:
I resent the firmware with PR#6718 to 2 of my Atom's. and I'm getting mixed results.  One is working fine but logging stops after it initially coming online (no constant activity logging as it normally does).  The other results in the log entry below.

```
[09:04:40][I][app:100]: ESPHome version 2024.5.4 compiled on May 28 2024, 08:54:55
[09:04:40][I][app:102]: Project m5stack.atom-echo-voice-assistant version 1.0
[09:04:40][C][wifi:580]: WiFi:
[09:04:40][C][esp32_rmt_led_strip:175]: ESP32 RMT LED Strip:
[09:04:40][C][esp32_rmt_led_strip:176]:   Pin: 27
[09:04:41][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Button'
[09:04:41][C][gpio.binary_sensor:016]:   Pin: GPIO39
[09:04:41][C][esp32_ble:374]: ESP32 BLE:
[09:04:42][D][api:102]: Accepted X.X.X.X
[09:04:42][W][component:237]: Component api took a long time for an operation (53 ms).
[09:04:42][D][api.connection:1321]: Home Assistant 2024.5.5 (172.24.1.3): Connected successfully
[09:04:42][D][voice_assistant:447]: State changed from IDLE to START_MICROPHONE
[09:04:42][W][component:237]: Component i2s_audio.microphone took a long time for an operation (51 ms).
[09:04:42][W][component:238]: Components should block for at most 30 ms.
[09:04:45][D][esp32_ble:255]: Disabling BLE...
[09:04:46][D][esp-idf:000][BTC_TASK]: E (11683) BT_OSI: config_save, err_code: 0x1

[09:04:46][W][component:237]: Component esp32_ble took a long time for an operation (232 ms).
[09:04:46][W][component:238]: Components should block for at most 30 ms.
```

weberruss, 2024-05-28:
If I try the pr#6718 :

```
external_components:
  - source: github://pr#6718
    components:
      - voice_assistant
      - i2s_audio
      - speaker

esp_adf:
```

I cant compile

```
esp_adf: [source /config/atom-echo.yaml:275]
  
  Component not found: esp_adf.
```

If I change it to
```
external_components:
  - source: github://pr#6718
    components:
      - voice_assistant
      - i2s_audio
      - speaker
```

I get

```
Failed config

voice_assistant: [source /config/atom-echo.yaml:89]
  id: va
  microphone: echo_microphone
  speaker: echo_speaker
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 2.0
  
  This option requires component esp_adf.
```

weberruss, 2024-05-28:
Ok the following works:

```
external_components:
  - source: github://pr#5230
    components:
      - esp_adf
    refresh: 0s

  - source: github://pr#6718
    components:
      - voice_assistant
      - i2s_audio
      - speaker

esp_adf:

```
Only getting the „ Error writing to I2S: ESP_OK“

nem0nic, 2024-05-29:
Just for my own sanity, we're leaving 

```
external_components:
  - source: github://pr#6718
    components:
      - voice_assistant
      - i2s_audio
      - speaker
``` 

In the build yaml for now since I don't see this issue assigned or in progress yet?

pos-ei-don, 2024-09-16:
Any solution? Seeing the same error, cannot compil toe above suggestions.

Calimerorulez, 2024-11-26:
Still the same problem here, on esphome 2024.11.1

nem0nic, 2024-11-27:
I'm usually the first one to get bitten by every and all updates on it but I updated to 2024.11.1 this morning but don't have that same error we were talking about in May and I am back on the default .yaml without all the workarounds we were using.  I do see I have a ESPHome update that popped up this afternoon though so now I'm worried.

nem0nic, 2024-11-28:
Pulled the trigger and updated to 24.11.2 and updated the atom's and not having the issue.  I'd try the "Clean Build Files" option under the 3 dots on the ride side when in ESP Home Device Compiler and hit "install" again.  Depending on what version you're updating from, we had to do that in the past when we first got it with the buffer problem when the patched fix came out.

dao55, 2024-11-30:
I now have the same issue. Clean build files don't work, fresh install doesn't work, factory reset doesn't work. I tend to install all updates as they happen - and never really seen any issues normally - so I'd be updating from 2024.11.1 from which I'm sure it worked fine. I've also tried all install methods just in case

dao55, 2024-11-30:
Pressing the button gets this from the logs:

[03:19:17][D][binary_sensor:036]: 'Button': Sending state ON
[03:19:20][D][binary_sensor:036]: 'Button': Sending state OFF
[03:19:20][D][voice_assistant:628]: Signaling stop...
[03:19:20][D][voice_assistant:512]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[03:19:20][D][voice_assistant:518]: Desired state set to IDLE
[03:19:20][D][voice_assistant:512]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[03:19:20][D][esp-idf:000]: I (322491) I2S: DMA queue destroyed

[03:19:20][D][voice_assistant:512]: State changed from STOPPING_MICROPHONE to IDLE
[03:19:20][D][voice_assistant:635]: Event Type: 2
[03:19:20][D][voice_assistant:727]: Assist Pipeline ended
[03:19:20][D][light:036]: 'M5Stack' Setting:
[03:19:20][D][light:051]:   Brightness: 60%
[03:19:20][D][light:059]:   Red: 100%, Green: 89%, Blue: 71%
[03:19:21][D][light:036]: 'M5Stack' Setting:
[03:19:21][D][light:051]:   Brightness: 60%
[03:19:21][D][light:059]:   Red: 100%, Green: 89%, Blue: 71%
[03:19:21][D][voice_assistant:512]: State changed from IDLE to START_MICROPHONE
[03:19:21][D][voice_assistant:518]: Desired state set to WAIT_FOR_VAD
[03:19:21][D][voice_assistant:222]: Starting Microphone
[03:19:21][D][voice_assistant:512]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[03:19:21][D][esp-idf:000]: I (323519) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4

[03:19:21][D][voice_assistant:512]: State changed from STARTING_MICROPHONE to WAIT_FOR_VAD
[03:19:21][D][voice_assistant:246]: Waiting for speech...
[03:19:21][D][voice_assistant:512]: State changed from WAIT_FOR_VAD to WAITING_FOR_VAD
[03:19:21][D][voice_assistant:259]: VAD detected speech
[03:19:21][D][voice_assistant:512]: State changed from WAITING_FOR_VAD to START_PIPELINE
[03:19:21][D][voice_assistant:518]: Desired state set to STREAMING_MICROPHONE
[03:19:21][D][voice_assistant:276]: Requesting start...
[03:19:21][D][voice_assistant:512]: State changed from START_PIPELINE to STARTING_PIPELINE
[03:19:21][D][voice_assistant:533]: Client started, streaming microphone
[03:19:21][D][voice_assistant:512]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[03:19:21][D][voice_assistant:518]: Desired state set to STREAMING_MICROPHONE
[03:19:21][D][voice_assistant:635]: Event Type: 1
[03:19:21][D][voice_assistant:638]: Assist Pipeline running
[03:19:21][D][voice_assistant:635]: Event Type: 9

but nothing happens after that

dao55, 2024-12-01:
I've downgraded the Atom to esphome v 2024.10.3  and it's back working now but I really hope there's a fix as despite having a working s3 box, this is the device I actually use the most

dao55, 2024-12-02:
@hs6666 :

There's an addon by khenderick

[(https://community.home-assistant.io/t/legacy-esphome-versions-add-on/568448)](url)

If you install that then choose the version of esphome you want to install (mine was 2024.10.3), install and restart then from the addon page open the addon and start the version. Then click the "open ui" on the addon and all your installed projects will be visible. Choose the M5 Atom and from the 3 dots in the corner choose "clean build files".

First time I tried that I got an error of some kind but that may have been my system because I had it doing other things but if you clean build the files and then install it should work. The M5 did take some time to actually start working again however, that may have been my system again at the time.

Once installed if you have a look at the logs it should say "Info: ESPHome [version]" at the top
With this running you do get a repair notice about a depreciated bit of code "binary_sensor.m5stack_atom_echo_23cddc_assist_in_progress is deprecated" but it's in 2025.04 (Thinking about it, I'm wondering if that has anything to do with the problems but just haven't got the time at the moment to look into it - hence the 3am logs I posted!)

On a related note, I don't often post things so while I'm here I just want to say thanks to @khenderick: for the add on which has been a godsend at times and thanks to the home assistant team for everything especially the little things they do - like telling you something is going to be depreciated 6 months in advance - if more companies did this I'm sure we'd all be a lot less stressed!

JoeCraparotta, 2024-12-29:
issue is still a problem even with latest updates, i also ran into this one as well https://github.com/esphome/firmware/issues/286. I now have 4 of these devices as tiny paper weights

dao55, 2024-12-29:
> issue is still a problem even with latest updates, i also ran into this one as well [esphome/firmware#286](https://github.com/esphome/firmware/issues/286). I now have 4 of these devices as tiny paper weights

If you add this to your yaml file and then clean build it should work as expected.:

esp32:
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0

(I've added it after my api key. compiles with 2024.12.2

JoeCraparotta, 2024-12-29:
> If you add this to your yaml file and then clean build it should work as expected:
>
> ```
> esp32:
>   framework:
>     type: esp-idf
>     version: 4.4.8
>     platform_version: 5.4.0
> ```
>
> (I've added it after [...]

Unfortunately i have tried that, i put that in a few of the devices yaml and i still get the "starting microphone / Could not allocate speaker buffer" error.

I ran clean build files, then Install.

I have also tried flashing these fresh again from https://www.home-assistant.io/voice_control/thirteen-usd-voice-remote/

Device version - 24.7.24 (ESPHome 2024.12.2)
HA - 2024.12.5

Open to any other ideas or things to try.

nem0nic, 2024-12-30:
Last version I was able to get to compile was 2024.12.2.  Every ESP Home
update has like a 95% chance of breaking the Echo's

nem0nic, 2025-01-18:
Will we ever get a working update again for M5stack Atom Echo's?  Every release that comes out I hope we'll be able to update again but it's been months since we've had a working update.  At least if you're not going to support the device advertised on HA website any longer that drove most people to purchase them then be respectful enough to just tell us "We are no longer supporting the M5stack Atom Echo's as voice assistants" and remove the tutorial at https://www.home-assistant.io/voice_control/thirteen-usd-voice-remote/ from the site.  We report the failures each update and get ignored "each update".

`-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Building ESP-IDF components for target esp32
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/data/build/m5stack-atom-echo-236a78-2/.pioenvs/m5stack-atom-echo-236a78-2/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)`

adorobis, 2025-02-02:
> 1. load an older working firmware, then never upgrade it until it becomes incompatible with HA

@hs6666 did you manage to get a working firmware? Where could I find one? Any other news on making it work again as voice assistant?

mateuszdrab, 2025-02-11:
I'm glad I found this, I thought there was an issue with my own configuration for the M5 Stack Echo/RaspiAudio ESP Muse Luxe but turns out I'm not the only one left with completely dysfunctional devices.

mateuszdrab, 2025-02-11:
> have to admit ive lost confidence in the device, I beleive it is very underpowered for use as an assist device
> 
> also they updated the code a while back so that the error during the build was fixed, but i still get the following notice in the logs....
> 
> "WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version."
> 
> also its very unreliable when using it as a regular assist device, i think it was truely introduced as a dev board only
> 
> im finding old raspberry pi 3's with the respeaker HAT a lot more reliable at the lower price point level
> 
> i will eventually be turning mine into a bluetooth proxy for prescence detection in future
> 
> 

The RaspiAudio ESP Muse Luxe is terrible, the ESP chip is not capable of handling two way audio like this.

I'm not even gonna mention the terrible sound quality.

Wyoming Satellite on Raspberry Pi or any other Linux device is the way to go...

This experience is from testing, I never got round to use my ESP voice assistants to work properly because aside of that, wake word detection was difficult, Whisper was very slow and etc. 

I've addressed those now with GPU acceleration, but the ESP modules don't work well, especially now I can't even get them to work at all.

Time to re-employ some RPIs