esphome/issues #5850: Recent update to 24.5.3 broke web server

X-Ryl669, 2024-05-25:
### The problem

I had LibreTiny devices in 24.4.0 version that were working fine, the web server used to display:
[image]

I've updated ESPHome to 24.5.3 today, that went ok, but now, I don't see any item in the web server anymore:
[image]

The devices still work with HA through, and I can still OTA them. Only the web server is not working.

### Which version of ESPHome has the issue?

2024.5.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.5

### What platform are you using?

BK72XX

### Board

IR_remote_I_don_t_know

### Component causing the issue

web_server

### Example YAML snippet

```yaml
# All substitution not shown here, they work perfectly

esphome:
  name: irremotenorth
  friendly_name: ${device_name}

bk72xx:
  board: cb3s
  framework:
    version: dev

# Enable logging
logger:

# Enable Home Assistant API
web_server:
  port: 80
  auth:
    username: admin
    password: ${web_password}

# Enable Home Assistant API
api:
  encryption:
    key: ${api_password}

ota:
  password: ${ota_password}

wifi:
  ssid: ${wifi_ssid}
  password: ${wifi_password}
  manual_ip:
   static_ip: ${ip}
   gateway: ${gw}
   subnet: ${subnet}
   dns1: ${dns}
  # domain: ${domain}
  ##### Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${device_name}"
    password: ${wifi_password}

captive_portal:

remote_transmitter:
  pin: P26
  carrier_duty_percent: 50%

sensor:
  - platform: homeassistant
    name: "Chambre Nord Temperature"
    entity_id: sensor.nspanel1_temperature
    id: nspanel1_temperature
    internal: true

climate:
  - platform: mitsubishi       
    name: "North"
    set_fan_mode: "quiet_4levels"
    supports_cool: true
    supports_heat: true
    supports_dry: true
    supports_fan_only: true
    horizontal_default: "middle"
    vertical_default: "down"    

switch:
  - platform: gpio
    name: "Blue Led"
    pin:
      number: P8
      inverted: false
  - platform: factory_reset
    name: Restart with Factory Default Settings
```

### Anything in the logs that might be useful for us?

```txt
No error message in the log
```

pvossel, 2024-05-26:
Exact same issue for me.

ssieb, 2024-05-26:
Try the fix for https://github.com/esphome/issues/issues/5793

pvossel, 2024-05-26:
I can confirm. Your [workarround ](https://github.com/esphome/issues/issues/5793#issuecomment-2131383089) works @ssieb.
Guess this issue should be merged with #5793.

ssieb, 2024-05-26:
Duplicate of https://github.com/esphome/issues/issues/5793

closed 2024-05-26 by ssieb