esphome/issues #5853: HA unresponsive after starting ESPHome device update

lein1013, 2024-05-26:
### The problem

When I start up update a ESPHome device, after approx 10-30sec HA gets unresponsive and is not reachable any more, MQTT shores offline. 
I could not figure out the root cause yet
hypothesis:
- esphome toolchain does something that crashes HA!?

after some time ~10min HA is up again
my current problem is that I don't find the logs till that happens

### Which version of ESPHome has the issue?

2024.5.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.5

### What platform are you using?

ESP8266

### Board

mainly wemos

### Component causing the issue

na

lein1013, 2024-06-02:
thx, I've updated ESPHome now. Did some updates (esphome device and addons) in parallel with (only) one HA freeze. I'll try to investigate if there is a relation to updating addons.

q20, 2024-07-17:
I'm still seeing freezes with the last few iterations of HA and ESPHome. Today with ESPHome version 2024.7.0 and HA 2024.7.2 (as VM) the same. A forced reboot of HA was required.