esphome/issues #5854: Compilation failed for esp32-c3-devkitm-1 (cmath & cstdint: No such file or directory)

NeilSCGH, 2024-05-26:
### The problem

Hello,

I've created a new esp32-c3-devkitm-1 device but esphome failed to compile, even with the default yaml config.

My process was:
- Plug the ESP32 C3 Super Mini via usb C to my computer
- On Chrome in Esphome, click New devices
- Select a Name, click Connect
- Preparing installation and then "Unable to compile the configuration"

### Which version of ESPHome has the issue?

2024.5.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.4

### What platform are you using?

ESP32

### Board

ESP32 C3 SuperMini

### Component causing the issue

api

### Example YAML snippet

```yaml
esphome:
  name: esp32-c3
  friendly_name: ESP32 C3

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "___redacted___"

ota:
  password: "___redacted___"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-C3 Fallback Hotspot"
    password: "___redacted___"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
Compiling .pioenvs/esp32-c3/src/esphome/components/api/api_connection.cpp.o
In file included from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/api_frame_helper.h:2:10: fatal error: cstdint: No such file or directory
 #include <cstdint>
          ^~~~~~~~~
compilation terminated.
Compiling .pioenvs/esp32-c3/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/esp32-c3/src/esphome/components/api/api_pb2.cpp.o
*** [.pioenvs/esp32-c3/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from src/esphome/components/api/api_frame_helper.cpp:1:
src/esphome/components/api/api_frame_helper.h:2:10: fatal error: cstdint: No such file or directory
 #include <cstdint>
          ^~~~~~~~~
compilation terminated.
*** [.pioenvs/esp32-c3/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
In file included from src/esphome/components/api/proto.h:3,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
src/esphome/core/component.h:3:10: fatal error: cmath: No such file or directory
 #include <cmath>
          ^~~~~~~
compilation terminated.
*** [.pioenvs/esp32-c3/src/esphome/components/api/api_pb2.cpp.o] Error 1
========================== [FAILED] Took 1.74 seconds ==========================
```

ulrichistderbeste, 2024-07-04:
I have the same problem with all types of esp32 c3 especially seed studio xiao esp32 c3.
Any idea to fix that.?

jiakechong1991, 2024-08-05:
我也遇到这个问题了，有大佬知道 为什么 #include <cmath> 会找不到文件而导入失败呢？