esphome/issues #5860: [Climate] Fan mode Middle appears after High

loongyh, 2024-05-28:
### The problem

The fan mode _Middle_ is ordered after _High_ which causes this weird quirk in the frontend:

[image]

Proper ordering should be Auto, Low, Medium, Middle, High.

Looking at the entity's state attributes, _Middle_ is indeed defined after _High_ in `fan_modes`:

[image]

Home Assistant orders the fan modes according to what is passed in the entity's `fan_modes` attribute.
Could this be caused by [Middle being enumed after High in ClimateFanMode](https://github.com/esphome/esphome/blob/497cf8742f495918ed368eafdeafe72d53c5b9e2/esphome/components/climate/climate_mode.h#L60)?

I'm using a [custom component](https://github.com/echavet/MitsubishiCN105ESPHome) as in attached YAML snippet but after skimming through the code I don't see anything relevant to fan mode ordering.

### Which version of ESPHome has the issue?

2024.5.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.5

### What platform are you using?

ESP32

### Component causing the issue

climate

### Example YAML snippet

```yaml
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

logger:
  level: INFO

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 17
  rx_pin: 16

climate:
  - platform: cn105
    name: "AC"
    supports:
      mode: [COOL, DRY]
      fan_mode: [AUTO, LOW, MEDIUM, MIDDLE, HIGH]
      swing_mode: ["OFF", VERTICAL]
    visual:
      min_temperature: 16
      max_temperature: 31
      temperature_step:
        target_temperature: 1
        current_temperature: 0.1
    compressor_frequency_sensor:
      name: AC Compressor Frequency
      icon: mdi:hvac
    vertical_vane_select:
      name: AC Vane
      icon: mdi:arrow-oscillating
```

ssieb, 2024-05-28:
Modes aren't really expected to have an ordering.

loongyh, 2024-05-29:
Hmm do you mean ordering should be done in the frontend regardless of the order passed? It seems there's consensus on this as according to [this issue of the climate card I'm using](https://github.com/nervetattoo/simple-thermostat/issues/185) there was an attempt to implement ordering logic in the frontend.

Still in the context of ESPHome's climate component, is there a reason why _Middle_ is enumed after _High_?

```cpp
...
  /// The fan mode is set to Medium
  CLIMATE_FAN_MEDIUM = 4,
  /// The fan mode is set to High
  CLIMATE_FAN_HIGH = 5,
  /// The fan mode is set to Middle
  CLIMATE_FAN_MIDDLE = 6,
...
```

I'm thinking it's a straightforward fix to simply swap _High_ and _Middle_. However, we need to make sure there are no dependencies on the enum integers by other components.

haraldhh, 2025-10-31:
I'm also very confused when I want to set the fan speed manually to something powerful but not too noisy. Is Middle more than Medium?

kitopopo, 2026-09-28:
Still reproducible in 2026.

I am currently using ESPHome 2026.9.0 with `echavet/MitsubishiCN105ESPHome` on a Mitsubishi MSZ-HJ35VA, and the fan modes are still displayed out of the expected order in Home Assistant.

The real order of the 4 manual fan speeds on this Mitsubishi model is:

`LOW -> MEDIUM -> MIDDLE -> HIGH`

In addition, the component exposes `AUTO` and `QUIET` as separate fan modes.

My ESPHome configuration declares:

```yaml
fan_mode:
  - AUTO
  - QUIET
  - LOW
  - MEDIUM
  - MIDDLE
  - HIGH
```

However, Home Assistant currently displays them as:

`Auto -> Low -> Medium -> High -> Middle -> Quiet`

So `MIDDLE` is still shown after `HIGH`, and `QUIET` is also placed at the end regardless of the order declared in YAML.

Functionally, all fan modes work correctly. This appears to be an ordering/presentation issue only.

I am attaching a current screenshot from Home Assistant showing the problem.

Environment:
- ESPHome 2026.9.0
- Home Assistant 2026
- ESP32-S3 / M5Stack ATOM S3 Lite
- Mitsubishi MSZ-HJ35VA
- External component: `github://echavet/MitsubishiCN105ESPHome`

This looks like the same underlying issue reported here in 2024, so I’m commenting to confirm that it is still present today.

[image]

haraldhh, 2026-09-28:
Hahaa, even the icons don't know which is which of middle and medium.

kitopopo, 2026-09-28:
Exactly 😄

The naming is confusing, but on the Mitsubishi MSZ-HJ35VA they are actually two distinct fan-speed steps.

On the physical remote, the four manual fan speeds are clearly ordered as:

"LOW -> MEDIUM -> MIDDLE -> HIGH"

So in this case:

- "MEDIUM" = speed 2
- "MIDDLE" = speed 3

CN105 also reports them as separate states, so the distinction is real even if the names are unfortunately very similar.