esphome/issues #5861: Error compiling raspiaudio-muse-luxe yaml

Nightstalker81, 2024-05-28:
### The problem

Unable to install 2024.5.4 update on Raspiaudio Muse Luxe device due to compile errors. See logs below.

### Which version of ESPHome has the issue?

2024.5.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.4

### What platform are you using?

ESP32

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.5.4
INFO Reading configuration /config/esphome/raspiaudio-muse-luxe-42455c.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/3552/head
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING `attenuation: 11db` is deprecated, use `attenuation: 12db` instead
WARNING GPIO12 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing raspiaudio-muse-luxe-42455c (board: esp-wrover-kit; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- Wire @ 2.0.0
|-- Improv @ 1.2.3
|-- WiFiClientSecure @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- ESP32-audioI2S @ 2.0.7
|-- FastLED @ 3.3.2
Compiling .pioenvs/raspiaudio-muse-luxe-42455c/libcd9/FastLED/FastLED.cpp.o
Compiling .pioenvs/raspiaudio-muse-luxe-42455c/libcd9/FastLED/bitswap.cpp.o
In file included from .piolibdeps/raspiaudio-muse-luxe-42455c/FastLED/FastLED.h:41,
                 from .piolibdeps/raspiaudio-muse-luxe-42455c/FastLED/FastLED.cpp:2:
.piolibdeps/raspiaudio-muse-luxe-42455c/FastLED/led_sysdefs.h:48:10: fatal error: Arduino.h: No such file or directory

*****************************************************************
* Looking for Arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Arduino.h"
* Web  > https://registry.platformio.org/search?q=header:Arduino.h
*
*****************************************************************

 #include <Arduino.h>
          ^~~~~~~~~~~
compilation terminated.
Compiling .pioenvs/raspiaudio-muse-luxe-42455c/libcd9/FastLED/colorpalettes.cpp.o
*** [.pioenvs/raspiaudio-muse-luxe-42455c/libcd9/FastLED/FastLED.cpp.o] Error 1
In file included from .piolibdeps/raspiaudio-muse-luxe-42455c/FastLED/FastLED.h:41,
                 from .piolibdeps/raspiaudio-muse-luxe-42455c/FastLED/bitswap.cpp:2:
.piolibdeps/raspiaudio-muse-luxe-42455c/FastLED/led_sysdefs.h:48:10: fatal error: Arduino.h: No such file or directory

*****************************************************************
* Looking for Arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Arduino.h"
* Web  > https://registry.platformio.org/search?q=header:Arduino.h
*
*****************************************************************

 #include <Arduino.h>
          ^~~~~~~~~~~
compilation terminated.
*** [.pioenvs/raspiaudio-muse-luxe-42455c/libcd9/FastLED/bitswap.cpp.o] Error 1
In file included from .piolibdeps/raspiaudio-muse-luxe-42455c/FastLED/FastLED.h:41,
                 from .piolibdeps/raspiaudio-muse-luxe-42455c/FastLED/colorpalettes.cpp:4:
.piolibdeps/raspiaudio-muse-luxe-42455c/FastLED/led_sysdefs.h:48:10: fatal error: Arduino.h: No such file or directory

*****************************************************************
* Looking for Arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Arduino.h"
* Web  > https://registry.platformio.org/search?q=header:Arduino.h
*
*****************************************************************

 #include <Arduino.h>
          ^~~~~~~~~~~
compilation terminated.
*** [.pioenvs/raspiaudio-muse-luxe-42455c/libcd9/FastLED/colorpalettes.cpp.o] Error 1
========================= [FAILED] Took 12.92 seconds =========================
```

ssieb, 2024-05-28:
What's your config?  But also see https://esphome.io/components/light/fastled.

ssieb, 2024-05-29:
If you install from that page again, you should get 2024.5.3.  I don't know why it's having an issue on your system.
The config on the site will be changed soon to use the rmt component instead.