esphome/issues #5862: midea doesn't report the Follow Me temp from the wired controller or HA/esphome (or offer an override)

jonoberheide, 2024-05-29:
### The problem

I have a Midea-compatible ducted unit with the white circular/external smart wifi device, plugged into CN40, with a wired controller daisy-chained through it. I'm using the HVAC USB wifi module from Cloudfree with esphome's midea component. Also, an IR transmitter attached to the USB module to transmit Follow Me command periodically.

So far that all works well. I can control the AC at the wired controller or via HA/esphome and the setpoint stays in sync between the two. Similarly, I can use Follow Me at the wired controller, or via HA/esphome and the IR transmitter.

The one part that doesn't work is the T1 room temperature sensor, which is plugged into CN6 on the control board. This temp sensor is wildly off unfortunately (based on the duct's install location), so it reports something like 78 degrees when the room is more like 68. When Follow Me is used (via the wired controller or via the IR transmitter), the AC unit does appear to respect the temperature that was transmitted to it, BUT...it doesn't report that temperature back to HA/esphome via the climate entity.

Based on my research, I think this is a limitation of the UART/wifi interface. I don't believe that temperature reading is available in the protocol (or at least has not been RE'ed in any libraries I've seen).

One possible fix would be to allow the midea component to accept a sensor configuration variable, similar to what climate_ir implements, that will be used to report the current temperature: https://esphome.io/components/climate/climate_ir.html#configuration-variables

This would close the loop on the IR-based Follow Me functionality. The Follow Me would send the temp from the in-room sensor via IR to the AC and then also report that same room temp back to HA via the climate entity.

I'd be happy to take a swing at implementing this, but wanted to get feedback on whether this would be a valid approach first.

### Which version of ESPHome has the issue?

2024.5.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.2

### What platform are you using?

ESP8266

### Board

Cloudfree Ductless HVAC Wi-Fi Module

### Component causing the issue

midea

### Example YAML snippet

```yaml
remote_transmitter:
  pin: GPIO5                       # For iot-uni-stick.
  carrier_duty_percent: 50%        # 50% for IR LED, 100% for direct connect to TSOP IR receiver output.

sensor:
  - platform: homeassistant
    name: "gym temp"
    entity_id: sensor.temp_sensor_gym_temperature   # Sensor from HASS
    internal: true
    filters:
      - throttle: 10s
      - heartbeat: 2min             # Maximum interval between updates.
      - debounce: 1s
    on_value:
      midea_ac.follow_me:
        temperature: !lambda "return x;"
        beeper: true               # Optional. Beep on update.

climate:
  - platform: midea
    id: ${node_id}_climate
    name: ${friendly_node_name}      # Use a unique name.
    autoconf: true              # Autoconfigure most options.
    beeper: true               # Beep on commands.
    visual:                     # Optional. Example of visual settings override.
      min_temperature: 15       # Note these values are in C
      max_temperature: 30
      temperature_step: 1       # This defaults to 0.5. I don't have a need for half-degree increments....
    outdoor_temperature:        # Optional. Outdoor temperature sensor (may display incorrect values after long inactivity).
      name: Outdoor Temp
```

yettavr6, 2024-09-19:
Not that this will be much help, but I just installed an ESP wifi module in my Bosch-branded Midea mini split, and the reported temperature is also wildly off. Current reported as 77 while the Honeywell thermostat reads 74, but it’s been off by 9-10 degrees at times. It seems to be the unit itself and not an ESP issue because I tried a Cielo Wigle adapter from a Mr Cool mini split (also a Midea) and it was the same.

dbaq, 2024-10-19:
hey @jonoberheide,

I have the same exact setup (except the IR tx) as you and I am encountering the same issue. I think your suggested solution is great and I would love for this to be implemented. As an alternative, I believe you can also create a manual  "climate" entity in HA directly and "compose" it with the sensors you want.
What I don't understand is why `midea_ac.follow_me:` does not work over UART and why it requires IR? if the wired thermostat is daisy chained to the wf-60a1 module, it should be doable to send the sensor value via UART right? what am I missing here?

maiksprenger, 2024-10-26:
I might be missing something... but the [Smartlight USB dongle](https://smartlight.me/smart-home-devices/wifi-devices/wifi-dongle-air-conditioners-midea-idea-electrolux-for-home-assistant) is based on ESPHome, and it reports the temperature that the AC measures. My Midea AC displays the measured temperature on mode FAN, and what I get in Home Assistant is the same temp. 
Notably, that USB dongle also lets you connect a wire to the IR sensor of the AC, and then the dongle can send follow me commands to get the AC controller to use another temperature source. So if I'm not misunderstanding something, you might not have to rely on the IR bridge. 
As far as I know, that dongle is ESPhome based, so I don't see why you couldn't do something similar with your dongle.

jonoberheide, 2024-10-26:
@maiksprenger On your first comment, yes the dongle reports the temp that the AC measures. What it doesn't do is report the temperature that it receives from a Follow-Me. So my proposed workaround was to have esphome read/report from an external temp sensor, overriding the AC one.

On the second comment, I think was @dbaq was referring to was being able to use Follow-Me without an IR transmitter.

maiksprenger, 2024-10-27:
Interesting. I tried it out last night, and I get my remote's reported temperature shown as "AC current temperature" once I switch to follow mode. Also, the AC cycled a lot less, only three times a night instead of three times an hour. I might start using that more. 

Not sure if I understand you correctly on your second comment, sorry. I agree with @dbaq that it's surprising that we can't send Follow Me over UART. Maybe it's just that nobody tried. And what I'm saying is that with the Smartlight dongle, you can at least wire it directly to the AC IR receiver, and don't need to have an IR emitter placed elsewhere in the room.

maiksprenger, 2024-10-30:
@jonoberheide I have a Comfee single split unit MSAF5-09HRDN8-QE, and the [Smartlight wifi module](https://smartlight.me/smart-home-devices/wifi-devices/wifi-dongle-air-conditioners-midea-idea-electrolux-for-home-assistant). And you're right, this plugs directly into the unit. 
CN40 is 4 wires, right? It's not unlikely that this is the same serial connection, just a different plug. The USB connector used doesn't mean it's USB, it's just a cheap connection to use apparently.

faaabius, 2024-11-05:
Hello,

was it verified by anyone that it is really impossible to send follow-me via the uart port - or did nobody find the correct sequence?

I am currently working on a deeper insight into the midea and what i found so far is that in the newer wall-mount units there are not so many communication channels and i guess that the telegrams may be transmitted all over the bus. 

My setup is a "Midea All Easy Blue (R290)":
* Standard Display (which holds the interface to uart)
** has 1 wire interface (3 pins. 5V, 0V, I/O) to adapter board
** adapter board has input contact + relais + XYE bus (which is derived from I/O via IC)
*** KJR-120X1 Wall Mount Controller will be connected to XYE bus  (currently not connected, waiting for some parts to modify it from 5V to 12V required for high wall models)

--> the setup is confirmed working with the KJR-120C1 (dual voltage XYE-Bus wall mount remote control)

The wall mount controllers note in the manual, that they use the follow-me function to override the temperature in the unit. This means that at least the XYE-bus is able to route this commands but it seems that nobody just documented these?
When my system is running, i can sniff the traffic on the interfaces.
What i think that there are commands and they can be routed, but are yet unknown because the are not in the mobile app or anywhere documented (why should midea implement port specific filters which may block further extensions of the system?)

maiksprenger, 2024-11-06:
I think you're right. I haven't dug deep into this, but to me it almost seems like the Infrared and UART protocols are the same: https://github.com/esphome/esphome/blob/1e2497748d3a18847df868ac8f4b893800426652/esphome/components/midea/ir_transmitter.h#L48 We know how to send follow me via Infrared, so maybe it's straightforward to get working. I just haven't had the time to dig into it.

Nightraider89, 2025-01-11:
I have no understanding of the programming, but this would be very usefull if the climate entity could report the correct value!