esphome/issues #5864: ESP32-C6 Support is lacking for Devices like the Xiao ESP-32-C6, still even though both Arduino and ESP-IDF both support it by now.

WebDesignerlol, 2024-05-29:
### The problem

Hello, I saw a few posts here that discussed the lack of support for ESP32-C6, presumably because it wasn't officially supported in Arduino or ESP-IDF frameworks but both now do support it officially as of 5.1.0  for ESP-IDF at least, and Arduino lists it as supported as well. 

My use case is to add the esp32-c6-devkitm-1 which should in theory support the new XIAO ESP32-C6 mini boards in a basic way on ESPHome running within Home Assistant. I see errors that tell me that my board is not supported in home assistant version of ESPHome.

Is there a good timeline for when this will become supported? Itching to use this and I know you guys haven't got the Zigbee or Matter side of things fully figured out, but it would be cool to use it as a basic ESP32 in the meantime while future proofing it's existence when Matter and Zigbee are (potentially? hopefully?) added into things with ESPHome, per rumors, lol.

Thank you for the consideration!

### Which version of ESPHome has the issue?

2024.5.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

XIAO ESP32-C6

WebDesignerlol, 2024-05-30:
i made it work on my config in esphome like so:

esphome:
  name: {custom_name}
  friendly_name: {custom_friendly_name}
  platformio_options:
    platform: https://github.com/platformio/platform-espressif32/archive/refs/tags/v6.7.0.zip
    board_build.f_cpu: 160000000L
    board_build.f_flash: 80000000L
    board_build.flash_size: 4MB
    build_flags: "-DBOARD_HAS_PSRAM"
    board_build.arduino.memory_type: qio_opi
esp32:
  board: esp32-c6-devkitm-1
  variant: esp32c6
  framework:
    type: esp-idf
    version: 5.2.1

Hopefully this saves someone some headaches but basically it would have been nice to just put in the:
esp32:
  board: esp32-c6-devkitm-1
  variant: esp32c6
  framework:
    type: esp-idf
    
    and not have to find all the custom parts of things to make it all work. hopefully my haul contributes to this? If seomeone can add these details to the ESP32-C6 and label it as the XAIO Seeed Studio version it would be awesome.

thefjordkeeper, 2024-07-01:
@WebDesignerlol have you had any luck getting this to compile with esp-idf 5.2.1 in ESPHome 2024.6.x recently?

ldrrp, 2024-07-02:
@thefjordkeeper seems to compile fine for 2024.6.4 when nothing else is enabled for it. A few things that worked for my other c3's dont seem to work straight forward on c6. Had to uncomment things slowly to get it to build

Cirehtims, 2024-08-22:
What can we do to help this to be treated sooner ?
I just bought ESP32-C6 boards to switch from wifi to zigbee and extends my presence detection network.

Same request for ESPresence :
https://github.com/ESPresense/ESPresense/issues/1270

Thanks

Bjack795, 2024-09-17:
Hi, is there any available schedule to integrate the esp32c6 in esphome?

max7287, 2024-09-28:
Hi,  I used the configuration for sensor water level 

```
esphome:
  name: "nivel-agua"
  friendly_name: ESP32-c6

esp32:
  board: esp32-c6-devkitm-1
  flash_size: 4MB
  variant: esp32c6
  framework:
    type: esp-idf
    version: "5.3.0"
    platform_version: 6.8.1
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y
```

FirstRulez, 2024-10-08:
Similar problems with the M5NanoC6 which is an ESP32-C6FH4 (at least that's what my slightly blurry eyes are reading!)

dylanschoenmakers, 2024-11-06:
I've made it work with the config below, it's a bit of mix and match from what is mentioned here and the official platformio page on seed: https://wiki.seeedstudio.com/xiao_esp32c6_with_platform_io/

```
esphome:
  name: co2sensor
  friendly_name: CO2 Sensor
  platformio_options:
    platform: https://github.com/mnowak32/platform-espressif32.git#boards/seeed_xiao_esp32c6

esp32:
  board: seeed_xiao_esp32c6
  variant: ESP32C6
  flash_size: 4MB
  framework:
    type: esp-idf
    version: 5.2.1
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y
```

tadeastehan, 2024-11-29:
> I've made it work with the config below, it's a bit of mix and match from what is mentioned here and the official platformio page on seed: [...]

Hello, I am trying to achieve the same thing. I can successfully compile it, but the parts of the code that should work are not working, e.g., the logger component and the LED blinking part of the code (LED_BUILTIN). Thx in advance.

FutureSharks, 2024-11-30:
@tadeastehan just add this to your quoted code there and you get a switch for the on-board LED

```yaml
# Enable logging
logger:
  level: DEBUG

# deleted boring bits

switch:
  - platform: gpio
    pin: GPIO15
    name: "On-board yellow LED"
    inverted: yes
    restore_mode: RESTORE_DEFAULT_OFF
```

tadeastehan, 2024-11-30:
@FutureSharks Hello, thx for the code. I can now blink the led but I can't get the logger component working. I tried this
```yaml
logger:
  level: DEBUG
  baud_rate: 115200
  hardware_uart: UART0
```
but with one luck. When I compile the code I get this output:
```yaml
INFO Successfully uploaded program.
INFO Starting log output from COM8 with baud rate 115200
[10:57:17]I (103) esp_image: segment 0: paddr=00010020 vaddr=42020020 size=0add0h ( 44496) map
[10:57:17]I (120) esp_image: segment 1: paddr=0001adf8 vaddr=40800000 size=05220h ( 21024) load
[10:57:17]I (126) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=180d0h ( 98512) map
[10:57:17]I (148) esp_image: segment 3: paddr=000380f8 vaddr=40805220 size=05404h ( 21508) load
[10:57:17]I (154) esp_image: segment 4: paddr=0003d504 vaddr=4080a630 size=011ach (  4524) load
[10:57:17]I (159) boot: Loaded app from partition at offset 0x10000
[10:57:17]I (196) boot: Set actual ota_seq=1 in otadata[0]
[10:57:17]I (196) boot: Disabling RNG early entropy source...
[10:57:17]I (207) cpu_start: Unicore app
[10:57:17]W (216) clk: esp_perip_clk_init() has not been implemented yet
[10:57:17]I (222) cpu_start: Pro cpu start user code
[10:57:17]I (222) cpu_start: cpu freq: 160000000 Hz
[10:57:17]I (223) cpu_start: Application information:
[10:57:17]I (225) cpu_start: Project name:     co2sensor
[10:57:17]I (230) cpu_start: App version:      2024.10.2
[10:57:17]I (235) cpu_start: Compile time:     Nov 30 2024 10:41:19
[10:57:17]I (241) cpu_start: ELF file SHA256:  d5d6952c8...
[10:57:17]I (247) cpu_start: ESP-IDF:          5.2.1
[10:57:17]I (252) cpu_start: Min chip rev:     v0.0
[10:57:17]I (256) cpu_start: Max chip rev:     v0.99 
[10:57:17]I (261) cpu_start: Chip rev:         v0.1
[10:57:17]I (266) heap_init: Initializing. RAM available for dynamic allocation:
[10:57:17]I (273) heap_init: At 4080C860 len 0006FDB0 (447 KiB): RAM
[10:57:17]I (279) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
[10:57:17]I (285) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
[10:57:17]I (292) spi_flash: detected chip: generic
[10:57:17]I (296) spi_flash: flash io: dio
[10:57:17]I (301) sleep: Configure to isolate all GPIO pins in sleep state
[10:57:17]I (307) sleep: Enable automatic switching of GPIO sleep configuration
[10:57:17]I (314) coexist: coex firmware version: 77cd7f8
[10:57:17]I (320) coexist: coexist rom version 5b8dcfa
[10:57:17]I (325) main_task: Started on CPU0
[10:57:17]I (328) main_task: Calling app_main()
[10:57:17]I (432) main_task: Returned from app_main()
```

tadeastehan, 2024-11-30:
Ok, I got it working.
```yaml
logger:
  baud_rate: 115200
  level: DEBUG
  hardware_uart: USB_SERIAL_JTAG
  ```

lucaskatayama, 2024-12-05:
Well got mine working now: 

```yaml
esphome:
  name: xiao
  friendly_name: xiao
  platformio_options:
    platform: https://github.com/lucaskatayama/platform-espressif32.git#feat/seeed_xiao_esp32c6.json

esp32:
  board: seeed_xiao_esp32c6
  variant: ESP32C6
  framework:
    type: esp-idf
    version: 5.3.0
    platform_version: 6.9.0
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y
...
i2c:
  scl: 23
  sda: 22
  scan: true
  frequency: 200kHz

uart:
  rx_pin: 17
  tx_pin: 16
  baud_rate: 9600
```

# Notes:
1. Did a PR to add the xiao c6 board to platformio, waiting to be approved
https://github.com/platformio/platform-espressif32/pull/1472
That is why `platform: https://github.com/lucaskatayama/platform-espressif32.git#feat/seeed_xiao_esp32c6.json` was set
2. With that board I can use the latest versions of `esp-idf` and platformio (`version: 5.3.1`, `platform_version: 6.9.0`)
3. For I2C, I needed to increase `frequency`, otherwise it crashes with some watchdog error.
4. Didn't need to change `logger`

Tested with:
1. PMS7003 (PMSX003 Particulate Matter Sensor) (UART)
2. ENS160+AHT21  (I2C)

Hope it helps

cbass27, 2024-12-12:
This may be a stupid question but are these example configurations posted just for Wi-Fi connectivity through ESPHome or is there Zigbee support with ESPHome/Home Assistant as well?

gwerhart0800, 2024-12-24:
First off, thanks to all those that have contributed to this thread. I apologize for the long winded post, but I spend a bit of time drilling down into the issue and I am not sure how to resolve it. I am trying to configure a Xaio ESP32C6 for i2c.

For reference, my yaml config is was lifted from examples in these comments:
~~~
esphome:
  name: xiao
  friendly_name: xiao
  platformio_options:
    platform: https://github.com/lucaskatayama/platform-espressif32.git#feat/seeed_xiao_esp32c6.json

esp32:
  board: seeed_xiao_esp32c6
  variant: ESP32C6
  framework:
    type: esp-idf
    version: 5.3.0
    platform_version: 6.9.0
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y

i2c:
  scl: 23
  sda: 22
  scan: true
  frequency: 200kHz
~~~
When I add the i2c key to the yaml config, I get a compilation error:
~~~
src/esphome/components/i2c/i2c_bus_esp_idf.cpp: In member function 'virtual void esphome::i2c::IDFI2CBus::setup()':
src/esphome/components/i2c/i2c_bus_esp_idf.cpp:21:42: error: 'I2C_NUM_1' was not declared in this scope; did you mean 'I2C_NUM_0'?
   21 |   next_port = (next_port == I2C_NUM_0) ? I2C_NUM_1 : I2C_NUM_MAX;
      |                                          ^~~~~~~~~
      |                                          I2C_NUM_0
*** [.pioenvs/xiao/src/esphome/components/i2c/i2c_bus_esp_idf.cpp.o] Error 1
~~~
The code in question is in i2c_bus_esp_idf.cpp:
~~~
 #if SOC_I2C_NUM > 1
   next_port = (next_port == I2C_NUM_0) ? I2C_NUM_1 : I2C_NUM_MAX;
 #else
   next_port = I2C_NUM_MAX;
 #endif
~~~
The soc_caps.h file for the esp32c6 has this definition:
~~~
// ESP32-C6 has 1 I2C
#define SOC_I2C_NUM                 (2U) // I2C_NUM = HP_I2C + LP_I2C
#define SOC_HP_I2C_NUM              (1U)
~~~
There is also a definition for the low power i2c interface later in the soc_caps.h file.
~~~
/*-------------------------- LP_I2C CAPS -------------------------------------*/
// ESP32-C6 has 1 LP_I2C
#define SOC_LP_I2C_NUM              (1U)
~~~
So, the i2c_bus_esp_idf.cpp code is looking at SOC_I2C_NUM > 1, but the hal/i2c_types.h code that defines I2C_NUM_1 only defines it if SOC_HP_I2C_NUM is >= 2:
~~~
/**
 * @brief I2C port number, can be I2C_NUM_0 ~ (I2C_NUM_MAX-1).
 */
typedef enum {
    I2C_NUM_0 = 0,              /*!< I2C port 0 */
#if SOC_HP_I2C_NUM >= 2
    I2C_NUM_1,                  /*!< I2C port 1 */
#endif /* SOC_HP_I2C_NUM >= 2 */
#if SOC_LP_I2C_NUM >= 1
    LP_I2C_NUM_0,               /*< LP_I2C port 0 */
#endif /* SOC_LP_I2C_NUM >= 1 */
    I2C_NUM_MAX,                /*!< I2C port max */
} i2c_port_t;
~~~
So, SOC_HP_I2C_NUM  is == 1 and thus I2C_NUM_1 is not defined in the enum. Since this SOC only has one high power i2c port, it would seem that mix of high and low power i2c interfaces is not properly handled?

It appears that v5.2.X does not have the high power and low power definitions, but my attempts to use V5.2.3 results in this message when trying to build:
~~~
  ESP-IDF 5.2.3 is not available with pioarduino; you may need to specify 'release'.
~~~
Since others seem to have gotten their code to compile with the i2c bus defined in the yaml, can anyone tell me how you did it?

Thanks!

gwerhart0800, 2024-12-25:
Added comment ... I finally got the ESP-IDF version down graded to 5.2.2 and that works with the i2c definition. I would definitely steer clear of v5.3.X until they fix the issues related to supporting both high and low power i2c on the esp32c6.

gwerhart0800, 2024-12-25:
lucaskatayama Thank you! Clearly, I need to spend more time looking at the LP processor core as I am planning to use the Xaio boards for battery powered sensors. Got to work on my soldering skills to make connections to the pads on the back ... or go with pogo pins.

lucaskatayama, 2024-12-26:
> lucaskatayama Thank you! Clearly, I need to spend more time looking at the LP processor core as I am planning to use the Xaio boards for battery powered sensors. Got to work on my soldering skills to make connections to the pads on the back ... or go with pogo pins.

Please share your learning... I need to learn it too. I want to use it with solar panels...

tadeastehan, 2024-12-28:
> Well got mine working now:
> 
> ```yaml
> esphome:
>   name: xiao
>   friendly_name: xiao
>   platformio_options:
>     platform: https://github.com/lucaskatayama/platform-espressif32.git#feat/seeed_xiao_esp32c6.json
> 
> esp32:
>   board: seeed_xiao_esp32c6
>   variant: ESP32C6
>   framework:
>     type: esp-idf
>     version: 5.3.0
>     platform_version: 6.9.0
>     sdkconfig_options:
>       CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y
> ...
> i2c:
>   scl: 23
>   sda: 22
>   scan: true
>   frequency: 200kHz
> 
> uart:
>   rx_pin: 17
>   tx_pin: 16
>   baud_rate: 9600
> ```
> 
> # Notes:
> 
>     1. Did a PR to add the xiao c6 board to platformio, waiting to be approved
>        [feat(board): added seeed_xiao_esp32c6 board platformio/platform-espressif32#1472](https://github.com/platformio/platform-espressif32/pull/1472)
>        That is why `platform: https://github.com/lucaskatayama/platform-espressif32.git#feat/seeed_xiao_esp32c6.json` was set... **Please give a thumbs there to help**
> 
>     2. With that board I can use the latest versions of `esp-idf` and platformio (`version: 5.3.1`, `platform_version: 6.9.0`)
> 
>     3. For I2C, I needed to increase `frequency`, otherwise it crashes with some watchdog error.
> 
>     4. Didn't need to change `logger`
> 
> 
> Tested with:
> 
>     1. PMS7003 (PMSX003 Particulate Matter Sensor) (UART)
> 
>     2. ENS160+AHT21  (I2C)
> 
> 
> Hope it helps

Hello! I'm currently working with the Seeeduino Xiao ESP32-C6 and have a quick question: at what I2C frequency do the watchdogs typically start to complain? I’m using an I2C device that requires a frequency of 100 kHz, and I want to ensure I won’t run into any issues.

obucklin, 2025-01-23:
Hello, I was wondering if anyone has gotten this to work with the arduino framework. My HA instance is on a Raspberry Pi 3B and it won't run esp-idf. Any tips would be appreciated. thanks!

filips, 2025-02-17:
In case anybody here runs into stability and connectivity issues on the XIAO ESP32-C6, I found out that you manually have to enable the antenna switch ([source](https://wiki.seeedstudio.com/xiao_esp32c6_getting_started/)).

This can be done by simply setting GPIO3 as an unexposed output:

```yaml
output:
  - platform: gpio
    id: wifi_switch_enable
    pin: GPIO3
```

This will enable the antenna switch, which is by default configured to use the built-in ceramic antenna. After doing this, my WiFi signal levels increased by ~20 dBm, and a lot of connectivity issues disappeared. I also had random reboots (Software Reset Digital Core and Voltage Unstable Reset), that have seemingly resolved, although I have not tested for prolonged time as I just discovered this. I wonder if the RF power ending up at the switch could cause some of this erratic behavior.

filips, 2025-02-18:
> So it seems in order to add support for the ESP32-C6, support needs to be added in platformio, there's a PR [here](https://github.com/platformio/platform-espressif32/pull/1472), but this PR depends on [this issue](https://github.com/platformio/platform-espressif32/issues/1225), which is very long and about support/involvement from Espressif 🤔

I think the issue there is that the PR includes both the `arduino` and `esp-idf` in the supported frameworks for the SEEED XIAO ESP32-C6, which blocks the merging in general. I have suggested that `arduino` be removed until the upstream issues there are resolved, so we at least can use the correct board definition without resorting to using forks of various libraries.

ximex, 2025-02-25:
maybe related to: https://github.com/esphome/issues/issues/6639

filips, 2025-03-26:
The board support has been merged into https://github.com/platformio/platform-espressif32/pull/1547 development branch now. Since as far as I can tell esphome uses `pioarduino/platform-espressif32` by default, we'll have to wait for them to integrate the newest upstream changes from `platformio/platform-espressif32` now. 

After then the board should also be included here: https://github.com/esphome/esphome/blob/4bb59ce1d16bcb81bab5a2e4ad54f759fc2846e1/esphome/components/esp32/boards.py

tadeastehan, 2025-03-26:
What is the best way to use `arduino` framework to leverage OTA updates that work only with `arduinoo` framework? I tried running this:
```yaml
esp32:
  board: seeed_xiao_esp32c6
  variant: ESP32C6
  flash_size: 4MB
  framework:
    type: arduino
    version: recommended
    platform_version: 6.9.0
    # sdkconfig_options:
    #   CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y
``` 
But i get this error:
```
*** missing SConscript file 'C:\\Users\\Acer\\.platformio\\packages\\framework-arduinoespressif32@3.20005.220925\\tools\\platformio-build-esp32c6.py'
File "C:\Users\Acer\.platformio\packages\framework-arduinoespressif32@3.20005.220925\tools\platformio-build.py", line 161, in <module>
```
So we may need to create a `platformio-build-esp32c6.py` file.

randybb, 2025-04-26:
@Zipponia it is exactly telling what you have to do - `rmt_channel:` is supported only on arduino, under idf 5 it is not supported, so remove it.

Zipponia, 2025-04-26:
As far as I see the framework arduino doesn’t yet support the XIAO esp32 C6

Or does it ?

Anyone managed to get it working ?

krissfr, 2025-04-28:
The board is supported by pioarduino ( https://github.com/pioarduino/platform-espressif32/blob/main/boards/seeed_xiao_esp32c6.json ), but still missing from https://github.com/esphome/esphome/blob/dev/esphome/components/esp32/boards.py .
i hope it will be supported soon :)

ximex, 2025-06-01:
no. the ESP32C6 will only work with esp-idf AFAIK
Is there anything why you need the arduino framework? AFAIK support for this should get dropped sooner or later

Zipponia, 2025-06-01:
> no. the ESP32C6 will only work with esp-idf AFAIK Is there anything why you need the arduino framework? AFAIK support for this should get dropped sooner or later

Only need to drive WS2812 leds.

Is there a way to do this now with the Idf framework then ?

Zipponia, 2025-06-08:
Thank you for the sketch

I'm using this board https://wiki.seeedstudio.com/rgb_matrix_for_xiao/ together with the XIAO Esp32 C6

So My full sketch is this 

`esphome:
  name: fru
  friendly_name: fru

esp32:
  board: esp32-c6-devkitc-1
  framework:
    type: esp-idf
    version: 5.5.0
    release: "20250510"
    platform_version: "https://github.com/pioarduino/platform-espressif32.git#develop"

    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "my key"

ota:
  - platform: esphome
    password: "my pass"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 192.168.1.138
    gateway: 192.168.1.1
    subnet: 255.255.255.0

status_led:
  pin: GPIO15

sensor:  
  # Wi-Fi Diagnostics
  - platform: wifi_signal # Reports the WiFi signal strength/RSSI in dB
    name: "WiFi Signal dB"
    id: wifi_signal_db
    internal: true
    update_interval: 3s
    entity_category: "diagnostic"

  - platform: copy # Reports the WiFi signal strength in %
    source_id: wifi_signal_db
    name: "WiFi Signal Percent"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "Signal %"
    entity_category: "diagnostic"
    device_class: ""

  - platform: uptime
    name: "Uptime"
    id: sensor_uptime
    update_interval: 300s
    entity_category: "diagnostic"

binary_sensor:
  - platform: status
    name: "Status"
    id: sensor_status

  - platform: template
    name: "API connected"
    id: sensor_api_connected
    internal: True
    entity_category: 'diagnostic'
    device_class: 'connectivity'
    lambda: return global_api_server->is_connected();

light:
  - platform: esp32_rmt_led_strip
    rgb_order: GRB
    name: "RGB-LED"
    pin: GPIO0
    num_leds: 60
    chipset: ws2812
    restore_mode: RESTORE_DEFAULT_OFF
    effects:
      - pulse:
          name: "Strobo"
          transition_length: 0s
          update_interval: 0.1s
      - pulse:
          name: "Fast"
          transition_length: 0s
          update_interval: 0.5s
      - pulse:
          name: "Normal"
          transition_length: 0s
          update_interval: 1s
      - pulse:
          name: "Slow"
          transition_length: 0s
          update_interval: 2s
      - random:
          name: "Random"
          transition_length: 5s
          update_interval: 7s`

it compiles and uploads

No led turns on.

With the normal Arduino code from the Arduino IDE I'm able to use all the leds

afsy-d, 2025-07-12:
> Thank you for the sketch
> 
> I'm using this board https://wiki.seeedstudio.com/rgb_matrix_for_xiao/ together with the XIAO Esp32 C6
> 
> So My full sketch is this
> 
> `esphome: name: fru friendly_name: fru
> 
> esp32: board: esp32-c6-devkitc-1 framework: type: esp-idf version: 5.5.0 release: "20250510" platform_version: "https://github.com/pioarduino/platform-espressif32.git#develop"
> 
> ...
> light:
> 
>     * platform: esp32_rmt_led_strip
>       rgb_order: GRB
>       name: "RGB-LED"
>       pin: GPIO0
>       num_leds: 60
>       chipset: ws2812
>       restore_mode: RESTORE_DEFAULT_OFF
>       effects:
>       
> 
> 
> it compiles and uploads
> 
> No led turns on.
> 
> With the normal Arduino code from the Arduino IDE I'm able to use all the leds

Having a similar issue with 'WS2813 Mini' LEDs on the C6, broke my head over it for a couple of hours before stumbling on this thread. It seems I am not alone.