esphome/issues #5865: platform: integration not accumulating values correctly

devjklein, 2024-05-30:
### The problem

I am working on a rainfall sensor for my esphome based weather station. I am  using pulse_meter to track the rainfall rate and the total rainfall. I want to use an integration sensor to track the total rainfall over the past hour and the past 24 hours, however it is not working as expected. I have attached my yaml for the sensor below as well as the log output on a test board while manually triggering 5 pulses.

I expect the integration sensor to accumulate the pulses within the specified time unit, but that does not appear to be the case. I do not understand the low values I am getting in my log.

### Which version of ESPHome has the issue?

2024.5.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.5

### What platform are you using?

ESP32

### Board

ESP32-WROOM-32

### Component causing the issue

integration

### Example YAML snippet

```yaml
sensor:
  - platform: pulse_meter
    pin:
      number: GPIO32
      mode: INPUT_PULLUP
    name: "${loc} rainfall per hour"
    id: rain_meter
    icon: "mdi:weather-rainy"
    internal_filter: 500ms
    timeout: 45min
    total:
      name: "${loc} Total Daily Rainfall"
      icon: "mdi:beaker-outline"

# Integration sensor for total rainfall over the past hour
  - platform: integration
    name: "${loc} Rainfall Past Hour"
    sensor: rain_meter
    time_unit: h

  # Integration sensor for total rainfall over the past 24 hours
  - platform: integration
    name: "${loc} Rainfall Past 24 Hours"
    sensor: rain_meter
    time_unit: d
```

### Anything in the logs that might be useful for us?

```txt
[13:47:05][D][sensor:094]: 'Test Total Daily Rainfall': Sending state 1.00000 pulses with 0 decimals of accuracy
[13:47:09][D][sensor:094]: 'Test Total Daily Rainfall': Sending state 2.00000 pulses with 0 decimals of accuracy
[13:47:09][D][sensor:094]: 'Test rainfall per hour': Sending state 18.54452 pulses/min with 2 decimals of accuracy
[13:47:09][D][sensor:094]: 'Test Rainfall Past Hour': Sending state 0.06578 pulses/minh with 4 decimals of accuracy
[13:47:09][D][sensor:094]: 'Test Rainfall Past 24 Hours': Sending state 0.00274 pulses/mind with 4 decimals of accuracy
[13:47:13][D][sensor:094]: 'Test Total Daily Rainfall': Sending state 3.00000 pulses with 0 decimals of accuracy
[13:47:13][D][sensor:094]: 'Test rainfall per hour': Sending state 15.02780 pulses/min with 2 decimals of accuracy
[13:47:13][D][sensor:094]: 'Test Rainfall Past Hour': Sending state 0.08439 pulses/minh with 4 decimals of accuracy
[13:47:13][D][sensor:094]: 'Test Rainfall Past 24 Hours': Sending state 0.00352 pulses/mind with 4 decimals of accuracy
[13:47:17][D][sensor:094]: 'Test Total Daily Rainfall': Sending state 4.00000 pulses with 0 decimals of accuracy
[13:47:17][D][sensor:094]: 'Test rainfall per hour': Sending state 13.65925 pulses/min with 2 decimals of accuracy
[13:47:17][D][sensor:094]: 'Test Rainfall Past Hour': Sending state 0.10186 pulses/minh with 4 decimals of accuracy
[13:47:17][D][sensor:094]: 'Test Rainfall Past 24 Hours': Sending state 0.00424 pulses/mind with 4 decimals of accuracy
[13:47:21][D][sensor:094]: 'Test Total Daily Rainfall': Sending state 5.00000 pulses with 0 decimals of accuracy
[13:47:21][D][sensor:094]: 'Test rainfall per hour': Sending state 16.91177 pulses/min with 2 decimals of accuracy
[13:47:21][D][sensor:094]: 'Test Rainfall Past Hour': Sending state 0.11693 pulses/minh with 4 decimals of accuracy
[13:47:21][D][sensor:094]: 'Test Rainfall Past 24 Hours': Sending state 0.00487 pulses/mind with 4 decimals of accuracy
```

gabest11, 2024-05-31:
It's doing a real integral, aka area under the curve, not very useful in this case. 

I also found out this myself with a rain sensor. My solution was to increment a global variable and define a template sensor based on that.

```
binary_sensor:
  - platform: remote_receiver
    name: "Rain 01"
    rc_switch_raw:
      code: '000001110101011101110110'
    filters:
      - delayed_off: 1000ms
    on_press:
      then:
        - lambda: |-
            id(counter_rain_01) += 1;
            ESP_LOGD("main", "counter_rain_01 = %d", id(counter_rain_01));              

globals:
 - id: counter_rain_01
   type: int
   restore_value: true
   initial_value: '0'

sensor:
  - platform: template
    name: "Rain 01"
    id: rain_01_total_id
    update_interval: 60s
    unit_of_measurement: "mm"
    icon: "mdi:weather-rainy"
    device_class: "precipitation"
    state_class: "total_increasing"
    accuracy_decimals: 3
    lambda: |-
      return id(counter_rain_01);
    filters:
      - multiply: 0.88888888888

button:
  - platform: template
    name: "Rain 01 reset"
    on_press:
      then:
        - lambda: |-
            id(counter_rain_01) = 0;
            ESP_LOGD("main", "counter_rain_01 RESET");
        - sensor.template.publish:
            id: rain_01_total_id
            state: !lambda 'return id(counter_rain_01);'
```

ssieb, 2024-05-31:
The output of the pulse meter is a rate, which doesn't make sense to accumulate and neither does the total.  Copy the total value to a global variable at the reset point and then your sensor can show the difference between the current total and that value.  But it would probably be better to let HA do the accumulating anyway.
If you need more help, then come ask on discord or one of the forums.

closed 2024-05-31 by ssieb

devjklein, 2024-05-31:
@gabest11 @ssieb Thank you so much for the input and sample code, I was realizing my mistake with further testing. I'm experimenting with the statistics sensor in Home Assistant to provide the rainfall over the past hour, 24 hours, etc.