esphome/issues #5866: HARDWARE_UART: USB_CDC fails in framework esp-idf

NonaSuomy, 2024-05-30:
### The problem

ESP-IDF not handling the usb-cdc option.

Works with `type: arduino` just not `type: esp-idf`

### Which version of ESPHome has the issue?

DEV

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

DEV

### What platform are you using?

ESP32

### Board

ESP32-S3-N16R8 esp32-s3-devkitc-1 style

### Component causing the issue

logger: hardware_uart: usb_cdc

### Example YAML snippet

```yaml
esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 16MB
  framework:
    #type: arduino
    type: esp-idf

# Enable logging
logger:
  level: VERY_VERBOSE
  hardware_uart: USB_CDC
```

### Anything in the logs that might be useful for us?

```txt
Compiling .pioenvs/esp32-s3-n16r8-001/esptimer/src/system_time.o
/config/.esphome/platformio/packages/framework-espidf/components/esp_system/port/soc/esp32s3/usb_console.c:77:1: error: static assertion failed: "usb_osglue_int is not multicore capable"
 Static_assert(SOC_CPU_CORES_NUM == 1, "usb_osglue_int is not multicore capable");
 ^~~~~~
*** [.pioenvs/esp32-s3-n16r8-001/esp_system/port/soc/esp32s3/usb_console.o] Error 1
========================= [FAILED] Took 84.28 seconds =========================
```

### Additional information

https://github.com/esphome/esphome/pull/3105

ssieb, 2024-05-30:
The default is USB_SERIAL_JTAG, doesn't that work?

NonaSuomy, 2024-05-31:
It seems that it should be compatible with both:
https://github.com/esphome/esphome/blob/1f301df51d4389f16e6a059c2ac87186c9631be8/esphome/components/logger/__init__.py#L86
`VARIANT_ESP32S3: [UART0, UART1, USB_CDC, USB_SERIAL_JTAG],`
```
On chips with an integrated USB Serial/JTAG Controller, it is possible to use the part of this controller that implements a serial port (CDC) to implement the serial console, instead of using UART with an external USB-UART bridge chip.
```

It doesn't give that error though with:
```yaml
esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 16MB
  framework:
    #type: arduino
    type: esp-idf

# Enable logging
logger:
  level: VERY_VERBOSE
  hardware_uart: USB_SERIAL_JTAG
```