esphome/issues #5867: Cannot update to Esphome 2024.5.4 from 2.24.5.3

bert269, 2024-05-31:
### The problem

No issues with upgrading to previous releases.
Getting these messages in the log, after trying to upgrade.

### Which version of ESPHome has the issue?

2024.5.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.5.5

### What platform are you using?

ESP8266

### Component causing the issue

Upgrade to later version.

### Anything in the logs that might be useful for us?

```txt
"Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:239
integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 6:47:35 PM (1 occurrences)
Last logged: 6:47:35 PM

[547629843648] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 168, in async_install
    await async_update_addon(self.hass, slug=self._addon_slug, backup=backup)
  File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 59, in _wrapper
    data = await funct(*argv, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 136, in async_update_addon
    return await hassio.send_command(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 593, in send_command
    raise HassioAPIError(error.get(ATTR_MESSAGE))
homeassistant.components.hassio.handler.HassioAPIError: Can't install ghcr.io/esphome/esphome-hassio:2024.5.4: 500 Server Error for http+docker://localhost/v1.44/images/create?tag=2024.5.4&fromImage=ghcr.io%2Fesphome%2Fesphome-hassio&platform=linux%2Farm64: Internal Server Error ("Get "https://ghcr.io/v2/": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)")

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 239, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2738, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2779, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 975, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1047, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 461, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 170, in async_install
    raise HomeAssistantError(f"Error updating {self.title}: {err}") from err
homeassistant.exceptions.HomeAssistantError: Error updating ESPHome: Can't install ghcr.io/esphome/esphome-hassio:2024.5.4: 500 Server Error for http+docker://localhost/v1.44/images/create?tag=2024.5.4&fromImage=ghcr.io%2Fesphome%2Fesphome-hassio&platform=linux%2Farm64: Internal Server Error ("Get "https://ghcr.io/v2/": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)")"
```

ssieb, 2024-05-31:
That looks like a networking issue or github problem.

bert269, 2024-06-02:
how do I proceed from here. Is there a nother forum I can reach out to with this problem, you maybe aware of?

ssieb, 2024-06-02:
You could try re-installing the add-on.

bert269, 2024-06-02:
That did not work....I uninstalled the existing ESPHome addon and tried to
re-install the latest. Not happening: After about 30minutes I decided to go
out and reboot the HA server (yellow).

After the reboot, it seems that ESPHOME was installed, but when I tried to
start it, I got this message:
[image: image.png]

I uninstalled ESPHOME again and rebooted the HAOS. Then tried to re-install
it and the same thing happened - the installation just hangs and is not
working.

 - I tried to reinstall it again and it still is not installed. Now all my
ESP devices are inaccessible.

Is there an older version I can try to re-install?
Please advise as this is critical and I am not even home for the next few
weeks.

jsuanet, 2024-06-03:
You can find older versions of ESPHome in this repro https://github.com/khenderick/esphome-legacy-addons