esphome/issues #5869: New ESP32-S3-Box-3 device cannot be set up with ESPHome site

admn-monkey, 2024-05-31:
### The problem

The device loads the firmware successfully, however the site never prompts for wi-fi credentials. There's no way to configure it for wi-fi so it just starts and says "No Wi-Fi". The troubleshooting QR code link does not provide a method to get it set up either. Reference documentation and YouTube videos clearly show a prompt after loading where wi-fi credentials can be put in. Without that prompt, flashing the device through the website leaves the device unable to be used. 

### Which version of ESPHome has the issue?

2024.5.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.5

### What platform are you using?

ESP32

### Board

Espressif

### Component causing the issue

Website

vidvisionify, 2024-06-01:
Seems like the same issue as #5845

admn-monkey, 2024-06-02:
> Seems like the same issue as #5845

No, I don't believe it is. In the case I've reported, the Wi-Fi configuration never shows up. Requires a specific sequence of steps to work around that are completely silly. 

## Steps to get the Wi-Fi prompt

1. Connect USB to PC then to ESP32-S3-BOX-3.  
2. Press the blue connect button on the site.  
3. Select the port on which Box-3 is connected and OK   
4. Select install  
5. After the installation is complete, press next
	1. If the wi-fi setting option does not appear, disconnect the USB from the PC.
	2. Reconnect to PC.
	3. Repeat steps 1-3
	4. Do not choose to install ; start the Console and then click Back. 
	5. It should show Wi-Fi settings. 
	6. If not then repeat these last steps

closed 2024-10-15 by github-actions[bot]