esphome/issues #5870: Can't call ble_client.ble_write from ble_client.on_connect

hachi, 2024-05-31:
### The problem

According to @jesserockz The problem seems to be that the ble_client triggering the on_connect automation maintains node_state independently of ble_client.ble_write action, and as a result you get told that the client is not connected when you call ble_write

A workaround can include using a delay action before the ble_write, to return control to the scheduler, but in the case I was trying to use it was not working because I wanted on_connect to finish before actions in other automation chains run.

### Which version of ESPHome has the issue?

Version: 2024.5.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32-IDF

### Board

WESP32

### Component causing the issue

ble_client

### Example YAML snippet

```yaml
ble_client:
  - mac_address: 9C:1D:58:93:5B:DE
    id: air_filter
    auto_connect: false
    on_connect:
      then:
        - lambda: |-
            ESP_LOGI("ble_client_lambda", "Client connected to HPA-250B Air filter");
        - lambda: |-
            ESP_LOGI("ble_client_lambda", "Sending handshake to HPA-250B Air Filter");
        - ble_client.ble_write:
            id: air_filter
            service_uuid: 0000fff0-0000-1000-8000-00805f9b34fb
            characteristic_uuid: 0000ffe9-0000-1000-8000-00805f9b34fb
            value: [0x4d, 0x41, 0x43, 0x2b, 0x9c, 0x1d, 0x58, 0x93, 0x5b, 0xde]
```

### Anything in the logs that might be useful for us?

```txt
[16:01:03][I][esp32_ble_client:227]: [0] [9C:1D:58:93:5B:DE] Connected
[16:01:03][I][ble_client_lambda:136]: Client connected to HPA-250B Air filter
[16:01:03][I][ble_client_lambda:138]: Sending handshake to HPA-250B Air Filter
[16:01:03][W][ble_client.automation:141]: Cannot write to BLE characteristic - not connected
```

ssieb, 2024-06-01:
The problem is this in the client base:
`bool connected() { return this->state_ == espbt::ClientState::ESTABLISHED; }`
That state doesn't get set until all the "nodes" which includes the triggers are run.

hachi, 2024-11-16:
How do we keep this issue open until fixed?