esphome/issues #5873: ESPHome devices disappear from the dashboard

hardl1, 2024-06-02:
### The problem

About 3 weeks ago I installed my first ESPHome device under Homeassistant.
I was able to see the device from the ESPHome dashboard and edit it there. Additional devices have since been added.

Since yesterday there are suddenly no more devices on the dashboard. The Yaml files are visible in the file editor and the devices are also present in HA.
I restarted ESPHome, server and browser, with no change.
Then deleted the ESPHome add-on and reinstalled it, also beta.
The devices were back, but all with an update request.
After the first update the devices are all gone again.

What should I do?

### Which version of ESPHome has the issue?

 2024.5.4 + 2024.5.4 beta

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core 2024.5.5

### What platform are you using?

ESP8266

### Board

D1 mini

ssieb, 2024-06-02:
Is this the same as https://github.com/esphome/issues/issues/5802?  Can you check your browser console?

hardl1, 2024-06-02:
Looks like it, but I have no idea what that means.
[image]

hardl1, 2024-06-02:
Same problem with 3 different browsers.

hardl1, 2024-06-02:
The highlighted line could be the problem, but I don't know how or what I could change.
[image]

ssieb, 2024-06-02:
That's not the same issue as the other one.
What does that error message say?

hardl1, 2024-06-03:
I found my mistake. 
I created a "New Device" esp129.yaml.
When copying from frost.yaml I mixed up the names.
In 129.yaml the name was: frost and in frost.yaml the name was esp129.
Your tip about the browser error showed me the way, thank you.

closed 2024-06-03 by hardl1