esphome/issues #5881: Ethernet not working anymore with ESPHome 2024.5.x (LILYGO T-ETH-Lite ESP32, RTL8201)

pixelwave, 2024-06-05:
### The problem

I updated to the newest **ESPHome version 2024.5.2** and my ethernet connection using a **LILYGO T-ETH-Lite ESP32** (RTL8201) does not work anymore.

I tested it with a basic script (see attached). I downgraded to **ESPHome version 2023.12.5** and there ethernet works again. 

Issue discussed in the home assistant forum (https://community.home-assistant.io/t/lilygo-t-eth-lite-esp32-not-connecting-via-ethernet/732020/5).

### Which version of ESPHome has the issue?

2024.5.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

LILYGO T-ETH-Lite ESP32

### Component causing the issue

ethernet

### Example YAML snippet

```yaml
esphome:
  name: eth-test
  friendly_name: eth-test

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "..."

ota:
  password: "..."

ethernet:
  type: RTL8201
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 0
  power_pin: GPIO12
```

xNUTx, 2024-06-05:
I can confirm the issue.

Discord troubleshooting and debugging session: https://discord.com/channels/429907082951524364/1247796225869090876

Adding the following to your config fixes the issue:
`external_components:
  - source: github://esphome/esphome@2024.5.0b1
    components: [ ethernet ]`

There are 2 changes visibly included in b2 that when applied in a PR inclusion neither break the ethernet component, but something in b2 certainly does.

xNUTx, 2024-06-05:
@jesserockz suggested using a PR that got merged this morning, which reverts one of the changes and applies it differently (#6704)

Try building with this instead:

```yaml
external_components:
  - source: github://esphome/esphome#6836
    components: [ ethernet ]
``` 

This results in a working build on 2024.5.5 for me.

pixelwave, 2024-06-06:
@xNUTx Thanks a lot! I can confirm it works with this component:

```
external_components:
  - source: github://esphome/esphome#6836
    components: [ ethernet ]
```

Hope this gets fixed soon in an official ESPHome update.

jesserockz, 2024-06-06:
Resolved by esphome/esphome#6836

closed 2024-06-06 by jesserockz