esphome/issues #5884: [sntp] Only first server is using with esp-idf

asergunov, 2024-06-06:
### The problem

The `CONFIG_LWIP_DHCP_MAX_NTP_SERVERS` is not changed so it's defaulted to `1`.

This means the second and third servers are silently ignored.

### Which version of ESPHome has the issue?

Version: 2024.5.5

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Component causing the issue

sntp

### Example YAML snippet

```yaml
time:
  - platform: sntp
    servers: 
      - 0.pool.ntp.org
      - 1.pool.ntp.org
      - 192.168.1.1
text_sensor:
  - platform: template
    name: SNTP IP 1
    update_interval: 5s
    entity_category: DIAGNOSTIC
    lambda: |
      auto&& ret = esp_sntp_getserver(0);
      if(ret)
        return esphome::network::IPAddress(ret).str();
      return {};
  - platform: template
    name: SNTP IP 2
    update_interval: 5s
    entity_category: DIAGNOSTIC
    lambda: |
      auto&& ret = esp_sntp_getserver(1);
      if(ret)
        return esphome::network::IPAddress(ret).str();
      return {};
  - platform: template
    name: SNTP IP 3
    update_interval: 5s
    entity_category: DIAGNOSTIC
    lambda: |
      auto&& ret = esp_sntp_getserver(2);
      if(ret)
        return esphome::network::IPAddress(ret).str();
      return {};
  - platform: template
    name: SNTP Server 1
    update_interval: 5s
    entity_category: DIAGNOSTIC
    lambda: |
      auto&& ret = esp_sntp_getservername(0);
      if(ret) return {ret};
      return {};
  - platform: template
    name: SNTP Server 2
    update_interval: 5s
    entity_category: DIAGNOSTIC
    lambda: |
      auto&& ret = esp_sntp_getservername(1);
      if(ret) return {ret};
      return {};
```

### Additional information

Only first one is set. The rest are empty

ssieb, 2024-06-06:
Duplicate of https://github.com/esphome/issues/issues/3123

closed 2024-06-06 by ssieb

asergunov, 2024-06-07:
#3123 actually describes few issues on 8266. I have this one fixed just by
```yaml
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_LWIP_SNTP_MAX_SERVERS: '3'
```