esphome/issues #5885: Component not found: globals.

Tihomir971, 2024-06-06:
### The problem

After upgrading to 2024.5.5, and Home Assistant to 2024.6.0 Global Variables stop working: "Component not found: globals."

### Which version of ESPHome has the issue?

2024.5.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.0

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
globals:
# Value to consider completely dry. Auto-calibrated.
  - id: dryValue
    type: double
    restore_value: no
    initial_value: '2.2'
  # Value to consider completely wet. Auto-calibrated.
  - id: wetValue
    type: double
    restore_value: no
    initial_value: '1.1'
```

ssieb, 2024-06-06:
Show the compile log including the entire error message.

Tihomir971, 2024-06-07:
Code:
esphome:
  name: "soil-moisture-esp32"
  friendly_name: Soil Moisture

esp32:
  board: esp32dev

# Enable logging
logger:
#  level: DEBUG 
  
# Enable Home Assistant API
api:

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

  manual_ip:
    static_ip: 192.168.1.187
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1
    dns2: 8.8.8.8  

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Moisture Fallback Hotspot"
    password: "rTo0MDeTDHXV"
    ap_timeout: 15min    #Set to >1min to prevent going to fallback when HA restarting after an update and run battery down
    
#captive_portal:

 # Example configuration entry
globals:
# Value to consider completely dry. Auto-calibrated.
  - id: dryValue
    type: double
    restore_value: no
    initial_value: '2.2'
  # Value to consider completely wet. Auto-calibrated.
  - id: wetValue
    type: double
    restore_value: no
    initial_value: '1.1'

sensor:
  - platform: adc
    pin: GPIO36
    id: plant_moisture_up
    name: "Plant Moisture Up"
    unit_of_measurement: "%"
    device_class: "moisture"
 #   icon: "mdi:water-percent"
    accuracy_decimals: 0
    attenuation: 12db
    force_update: true
    update_interval: 1s
    filters:
#      - exponential_moving_average
      - median:
          window_size: 15
          send_every: 15
          send_first_at: 5
      - lambda: |-
          if (x > id(dryValue)) {
            return 0;
          } else if (x < id(wetValue)) {
            return 100;
          } else {
            return round((id(dryValue) - x) / (id(dryValue) - id(wetValue)) * 100);
          }

Validation error:
INFO ESPHome 2024.5.5
INFO Reading configuration /config/esphome/soil-moisture-esp32.yaml...
ERROR Unable to load component globals:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 176, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 936, in exec_module
  File "<frozen importlib._bootstrap_external>", line 1074, in get_code
  File "<frozen importlib._bootstrap_external>", line 1004, in source_to_code
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/esphome/esphome/components/globals/__init__.py", line 77
    data = {
           ^
SyntaxError: '{' was never closed
Failed config

globals: [source /config/esphome/soil-moisture-esp32.yaml:41]
  
  Component not found: globals.
  - id: dryValue
    type: double
    restore_value: False
    initial_value: 2.2
  - id: wetValue
    type: double
    restore_value: False
    initial_value: 1.1

ssieb, 2024-06-07:
You either have a modified component or data corruption.

Tihomir971, 2024-06-07:
I didn't change yaml code.
If I remove globals and lambda everything work fine.
Where can be data corruption? Should try to reinstall ESPHome integration

ssieb, 2024-06-07:
Yes, re-install the add-on.  It looks like part of that file is missing.

closed 2024-10-07 by Tihomir971