esphome/issues #5888: modbus_controller switch write "on" does not work

MStapelfeldt, 2024-06-07:
### The problem

I have a switch to turn on and off a Growatt MIC600 Inverter
```
 switch:
   - platform: modbus_controller
     name: "${friendly_name} OnOff"
     register_type: holding
     address: 0
     entity_category: config
     icon: "mdi:toggle-switch"
```
It's possible to turn the inverter off, but not back on again. I found out, using a select instead of switch is working to switch off and on.
```
select:
  - platform: modbus_controller
    id: onoffsel1
    name: "${friendly_name} OnOffSel"
    address: 0
    value_type: U_WORD
    icon: "mdi:toggle-switch"
    optionsmap:
      "Off": 0
      "On": 1
```
Because both times the value which is send is "1", I would expect both, Select and Switch would work the same way.

### Which version of ESPHome has the issue?

2024.5.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.5

### What platform are you using?

ESP8266

### Board

D1-Mini, ESP12, ESP07

### Component causing the issue

modbus_controller

### Example YAML snippet

```yaml
substitutions:
  device_description: Growatt Inverter
  friendly_name:  Growatt Inverter
  name: Shine X 1
  
esphome:
  name: shine_x_1

esp8266:
  board: d1_mini

# Enable logging
logger:
  baud_rate: 0
# Enable Home Assistant API
api:
  encryption:
    key: "n0Soo81ScxlWI9YR9jNlatoDS7HA2VrNOABnEXccukQ="

ota:
  password: !secret ota_passwd

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${name} Fallback Hotspot"
    password: "Yourpassword"

captive_portal:

time:
  - platform: homeassistant
    id: homeassistant_time

uart:
  id: mod_bus
  tx_pin: 1
  rx_pin: 3
  baud_rate: 115200
  
modbus:
  id: modbus1
  uart_id: mod_bus
  
modbus_controller:
  - id: growatt
# the Modbus device addr
    address: 0x1
    modbus_id: modbus1
    setup_priority: -10  
    update_interval: 5s

number:
    - platform: modbus_controller
      name: "${friendly_name} Max Output Power %"
      id: poweroutput
      address: 3
      value_type: U_WORD
      min_value: 0
      max_value: 100
      step: 1
 
    - platform: modbus_controller
      name: "${friendly_name} Reactive Power Output %"
      id: reactive_poweroutput
      address: 4
      value_type: U_WORD
      min_value: 0
      max_value: 100
      entity_category: config
      step: 1

select:
  - platform: modbus_controller
    id: onoffsel
    name: "${friendly_name} OnOffSel"
    address: 0
    value_type: U_WORD
    # entity_category: config
    icon: "mdi:toggle-switch"
    optionsmap:
      "Off": 0
      "On": 1

text_sensor:
  - platform: modbus_controller
    name: "${friendly_name} Firmware Version"
    address: 9
    register_count: 3
    register_type: holding
    #internal: true
    entity_category: diagnostic

  - platform: template
    name: "${friendly_name} Status"
    icon: mdi:eye
    entity_category: diagnostic
    lambda: |-
      if (id(status).state == 1) {
        return {"Normal"};
      } else if (id(status).state == 0)  {
        return {"Waiting"};
      } else {
        return {"Fault!"};
      }

 switch:
   - platform: modbus_controller
     name: "${friendly_name} OnOff"
     register_type: holding
     address: 0
     entity_category: config
     icon: "mdi:toggle-switch"
```

### Additional information

The only diffrence in sending the value is `register_type: holding` vs `value_type: U_WORD`, both should be 16bit, so I guess it's the way it's written. Also switch "Off" (value 0) works with switch and select.
```
- U_WORD (unsigned 16 bit integer from 1 register = 16bit)
- Holding Registers - Holding registers are the most universal 16-bit register. They may be read and/or written. Modbus Function Code 3 (Read Holding Registers) will be used.
```
Same issue is when using `register_type: coil`

ssieb, 2024-06-07:
Add `debug:` to the uart and see if it's the same data sent in both cases.

MStapelfeldt, 2024-06-08:
Sry for the late replay. Here is the debug output:
From Switch:
```
[22:52:58][D][uart_debug:114]: <<< 01:04:02:01:8E:38:C4
[22:53:00][D][switch:012]: 'Growatt Inverter OnOff' Turning ON.
[22:53:00][D][switch:055]: 'Growatt Inverter OnOff': Sending state ON
[22:53:00][D][uart_debug:114]: >>> 01:0F:00:00:00:01:01:01:EF:57
[22:53:00][D][uart_debug:114]: >>> 01:0F:00:00:00:01:01:01:EF:57
[22:53:00][D][uart_debug:114]: >>> 01:0F:00:00:00:01:01:01:EF:57
[22:53:01][D][uart_debug:114]: >>> 01:0F:00:00:00:01:01:01:EF:57
[22:53:01][D][uart_debug:114]: >>> 01:0F:00:00:00:01:01:01:EF:57
[22:53:01][W][modbus_controller:030]: Modbus device=1 set offline
[22:53:01][D][modbus_controller:040]: Modbus command to device=1 register=0x00 countdown=0 no response received - removed from send queue
[22:53:03][D][uart_debug:114]: >>> 01:03:00:00:00:01:84:0A
```
From Select:
```
[23:08:51][D][select:062]: 'Growatt Inverter OnOffSel' - Setting
[23:08:51][D][select:115]: 'Growatt Inverter OnOffSel' - Set selected option to: On
[23:08:51][D][modbus_controller.select:048]: Found value 1 for option 'On'
[23:08:51][D][uart_debug:114]: >>> 01:06:00:00:00:01:48:0A
[23:08:51][D][uart_debug:114]: <<< 01:06:00:00:00:01:48:0A
[23:08:53][D][uart_debug:114]: >>> 01:01:00:00:00:01:FD:CA
[23:08:53][D][uart_debug:114]: >>> 01:01:00:00:00:01:FD:CA
[23:08:53][D][uart_debug:114]: >>> 01:01:00:00:00:01:FD:CA
[23:08:53][D][uart_debug:114]: >>> 01:01:00:00:00:01:FD:CA
[23:08:54][D][uart_debug:114]: >>> 01:01:00:00:00:01:FD:CA
[23:08:54][W][modbus_controller:030]: Modbus device=1 set offline
[23:08:54][D][modbus_controller:040]: Modbus command to device=1 register=0x00 countdown=0 no response received - removed from send queue
[23:08:54][D][uart_debug:114]: >>> 01:03:00:00:00:01:84:0A
[23:08:54][W][modbus_controller:064]: Modbus device=1 back online
[23:08:54][D][modbus_controller.select:014]: New select value 1 from payload
[23:08:54][D][select:015]: 'Growatt Inverter OnOffSel': Sending state On (index 1)
[23:08:54][D][uart_debug:114]: <<< 01:03:02:00:01:79:84
[23:08:54][D][uart_debug:114]: >>> 01:03:00:03:00:02:34:0B
```
There is a big diffrence if I picked the right output...

ssieb, 2024-06-08:
Is that log with the config having `register_type: holding` for the switch?

MStapelfeldt, 2024-06-08:
Yes, it's same yaml as posted above.

ssieb, 2024-06-08:
That doesn't make sense.  It's sending the wrong command.
What's the uart output if you add `use_write_multiple: false` to the switch?

MStapelfeldt, 2024-06-08:
It's this:
```
[01:45:47][D][switch:012]: 'Growatt Inverter OnOff' Turning ON.
[01:45:47][D][switch:055]: 'Growatt Inverter OnOff': Sending state ON
[01:45:47][D][uart_debug:114]: >>> 01:05:00:00:FF:00:8C:3A
[01:45:47][D][uart_debug:114]: >>> 01:05:00:00:FF:00:8C:3A
[01:45:48][D][uart_debug:114]: >>> 01:05:00:00:FF:00:8C:3A
[01:45:48][D][uart_debug:114]: >>> 01:05:00:00:FF:00:8C:3A
[01:45:48][D][uart_debug:114]: >>> 01:05:00:00:FF:00:8C:3A
[01:45:48][W][modbus_controller:030]: Modbus device=1 set offline
[01:45:48][D][modbus_controller:040]: Modbus command to device=1 register=0x00 countdown=0 no response received - removed from send queue
[01:45:50][D][uart_debug:114]: >>> 01:03:00:00:00:01:84:0A
```

ssieb, 2024-06-08:
It's definitely thinking that the register type is set to "coil".  Double check the config.  Enable verbose logging as that will provide a bit more info.

ssieb, 2024-06-09:
Also, set `bitmask: 1`

MStapelfeldt, 2024-06-09:
> Enable verbose logging

not sure how to do this...

MStapelfeldt, 2024-06-09:
current config switch:

```

switch:
  - platform: modbus_controller
    name: "${friendly_name} OnOff"
    skip_updates: 5
    use_write_multiple: false
    register_type: holding
    address: 0
    bitmask: 1
    entity_category: config
    icon: "mdi:toggle-switch"
```

And log:

```

[02:07:02][D][switch:016]: 'Growatt Inverter OnOff' Turning OFF.
[02:07:02][D][switch:055]: 'Growatt Inverter OnOff': Sending state OFF
[02:07:02][D][uart_debug:114]: >>> 01:06:00:00:00:00:89:CA
[02:07:02][D][uart_debug:114]: <<< 01:06:00:00:00:00:89:CA
[02:07:04][D][uart_debug:114]: >>> 01:03:00:03:00:02:34:0B

[02:08:18][D][switch:012]: 'Growatt Inverter OnOff' Turning ON.
[02:08:18][D][switch:055]: 'Growatt Inverter OnOff': Sending state ON
[02:08:18][D][uart_debug:114]: >>> 01:06:00:00:00:01:48:0A
[02:08:18][D][uart_debug:114]: <<< 01:06:00:00:00:01:48:0A
[02:08:19][D][uart_debug:114]: >>> 01:03:00:03:00:02:34:0B
```

It looks like this was right with bitmask 1. It's now working to turn off and on!

I still don't know why it's working with select, but not without bitmask...

ssieb, 2024-06-09:
Because the select is setting a specific value.  The switch doesn't know what value to send for on, so it just sets all the bits unless you tell it otherwise.

closed 2024-06-09 by ssieb

MStapelfeldt, 2024-06-09:
ok, so sending off works when writing all bits to 0 but on needs only one bit to 1.

Thank you a lot for you help!