esphome/issues #5889: Pylontech 3000A invalid line: found only 13 items

RonnyKempe, 2024-06-07:
### The problem

Plyontech Component seems to have problems parsing the Answer from my 4 Year old Pylontech Stack.
This is returned from the Master Battery
19:19:48	[D]	[pylontech:081]	
invalid bat_num in line Power Volt   Curr   Tempr  Tlow   Thigh  Vlow   Vhigh  Base.St  Volt.St  Curr.St  Temp.St  Coulomb  Time                 B.V.St   B.T.St  
19:19:48	[W]	[pylontech:085]	
invalid line: found only 13 items in 1     50831  0      30000  28000  29000  3378   3392   Idle     Normal   Normal   Normal   105%     2024-06-08 01:21:48  Normal   Normal  
19:19:48	[W]	[pylontech:085]	
invalid line: found only 13 items in 2     50820  0      30000  28000  29000  3379   3391   Idle     Normal   Normal   Normal   106%     2024-06-08 01:21:47  Normal   Normal  
19:19:48	[W]	[pylontech:085]	
invalid line: found only 13 items in 3     50818  0      29000  27000  27000  3377   3390   Idle     Normal   Normal   Normal   107%     2024-06-08 01:21:47  Normal   Normal  

But expected seems to be 
   // Power Volt  Curr Tempr Tlow  Thigh  Vlow Vhigh Base.St Volt.St Curr.St Temp.St Coulomb Time                B.V.St B.T.St MosTempr M.T.St
   // 1    50548  8910 25000 24200 25000  3368 3371  Charge  Normal  Normal  Normal  97%     2021-06-30 20:49:45 Normal Normal 22700    Normal

Whats missing is "MosTempr" and "M.T.St"

### Which version of ESPHome has the issue?

2024.5.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

all

### What platform are you using?

ESP8266

### Board

wemos

### Component causing the issue

Pylontech

### Example YAML snippet

```yaml
uart:
  id: uart_bus
  tx_pin: 1
  rx_pin: 3
  baud_rate: 115200
  rx_buffer_size: 4096
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
pylontech:
  - id: pylontech0
    uart_id: uart_bus
    update_interval: 20s

sensor:
  - platform: pylontech
    battery: 1
    voltage:
      id: bat1_voltage
      name: "Battery1 Voltage"
    current:
      id: bat1_current
      name: "Battery1 Current"
    coulomb:
      name: "Battery1 State of Charge"
    temperature:
      name: "Battery1 Temperature"
  
  - platform: pylontech
    battery: 2
    voltage:
      id: bat2_voltage
      name: "Battery2 Voltage"
    current:
      id: bat2_current
      name: "Battery2 Current"
    coulomb:
      name: "Battery2 State of Charge"
    temperature:
      name: "Battery2 Temperature"

  - platform: pylontech
    battery: 3
    voltage:
      id: bat3_voltage
      name: "Battery3 Voltage"
    current:
      id: bat3_current
      name: "Battery3 Current"
    coulomb:
      name: "Battery3 State of Charge"
    temperature:
      name: "Battery3 Temperature"

  - platform: pylontech
    battery: 1

    voltage_low:
      name: "Battery1 min cell voltage"
    voltage_high:
      name: "Battery1 max cell voltage"

text_sensor:
  - platform: pylontech
    battery: 1
    base_state:
      id: bat1_base_state
      name: "Battery1 Base State"
    voltage_state:
      id: bat1_voltage_state
      name: "Battery1 Voltage State"
    current_state:
      id: bat1_current_state
      name: "Battery1 Current State"
    temperature_state:
      id: bat1_temperature_state
      name: "Battery1 Temperature State"

  - platform: pylontech
    battery: 2
    base_state:
      id: bat2_base_state
      name: "Battery2 Base State"
    voltage_state:
      id: bat2_voltage_state
      name: "Battery2 Voltage State"
    current_state:
      id: bat2_current_state
      name: "Battery2 Current State"
    temperature_state:
      id: bat2_temperature_state
      name: "Battery2 Temperature State"

  - platform: pylontech
    battery: 3
    base_state:
      id: bat3_base_state
      name: "Battery3 Base State"
    voltage_state:
      id: bat3_voltage_state
      name: "Battery3 Voltage State"
    current_state:
      id: bat3_current_state
      name: "Battery3 Current State"
    temperature_state:
      id: bat3_temperature_state
      name: "Battery3 Temperature State"
```

### Anything in the logs that might be useful for us?

```txt
19:24:08	[W]	[component:237]	
Component pylontech took a long time for an operation (188 ms).
19:24:08	[W]	[component:238]	
Components should block for at most 30 ms.
19:24:08	[D]	[pylontech:081]	
invalid bat_num in line 
pylon>pw
19:24:08	[D]	[pylontech:081]	
invalid bat_num in line 
@
19:24:08	[D]	[pylontech:081]	
invalid bat_num in line Power Volt   Curr   Tempr  Tlow   Thigh  Vlow   Vhigh  Base.St  Volt.St  Curr.St  Temp.St  Coulomb  Time                 B.V.St   B.T.St  
19:24:08	[W]	[pylontech:085]	
invalid line: found only 13 items in 1     50761  -238   30000  28000  29000  3375   3386   Dischg   Normal   Normal   Normal   105%     2024-06-08 01:26:08  Normal   Normal  
19:24:08	[W]	[pylontech:085]	
invalid line: found only 13 items in 2     50766  -219   29000  28000  29000  3377   3387   Dischg   Normal   Normal   Normal   106%     2024-06-08 01:26:07  Normal   Normal  
19:24:08	[W]	[pylontech:085]	
invalid line: found only 13 items in 3     50749  -210   29000  27000  27000  3373   3386   Dischg   Normal   Normal   Normal   107%     2024-06-08 01:26:07  Normal   Normal  
19:24:08	[W]	[pylontech:085]	
invalid line: found only 1 items in 4     -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       
19:24:08	[W]	[pylontech:085]	
invalid line: found only 1 items in 5     -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       
19:24:08	[W]	[pylontech:085]	
invalid line: found only 1 items in 6     -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       
19:24:08	[W]	[pylontech:085]	
invalid line: found only 1 items in 7     -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       
19:24:08	[W]	[pylontech:085]	
invalid line: found only 1 items in 8     -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       
19:24:08	[D]	[uart_debug:158]	
<<< "\rpylon>"
19:24:08	[D]	[pylontech:081]	
invalid bat_num in line 
Command completed successfully
19:24:08	[D]	[pylontech:081]	
invalid bat_num in line 
$$
19:24:28	[D]	[uart_debug:158]	
>>> "pwr\n"
```

camel1cz, 2024-08-12:
Have the same problem + I had to change the line delimiter to CR (\r or 0x0D) for correct read out of lines from pylontech console.
The integration seems to be hardcoded to one specific version of pylontech firmware.

I started to code support for more versions, but ran into some design problems combinded with lack of time and give up for now...

As quick fix:
1) create in your /config/esphome new folder (my is called my_components)
2) clone the pylontech extension into it, so you'll see file /config/esphome/my_components/pylontech/pylontech.cpp
3) in you yaml add this section to override original pylontech component with your local copy. I placed it after esphome: and esp32: sections:

`external_components:
  - source: 
      type: local
      path: my_components
    components: [pylontech]`

In the code, I changed in the pylontech.cpp:
a) (I don't think you need it, as your lines are read correctly) constant for line ending:
`static const uint8_t ASCII_LF = 0x0D;`

b) parsing line change to:
`  const int parsed = sscanf(                                                                                   // NOLINT
      buffer.c_str(), "%d %d %d %d %d %d %d %d %7s %7s %7s %7s %d%% %*d-%*d-%*d %*d:%*d:%*d %*s %*s %5*s %*s",  // NOLINT
      &l.bat_num, &l.volt, &l.curr, &l.tempr, &l.tlow, &l.thigh, &l.vlow, &l.vhigh, l.base_st, l.volt_st,      // NOLINT
      l.curr_st, l.temp_st, &l.coulomb/*, mostempr_s*/);                                                           // NOLINT`

c) and few lines below, change:
`if (parsed != 14) {`
to
`if (parsed != 13) {`

RonnyKempe, 2024-10-02:
Thank you, i works.

elpamyelhsa, 2025-02-09:
Just got a US5000-B with a build date of Dec-2024 with a newer firmware. I had to update the code for support. If I get time to submit a change maybe the console command **info** could be used to identify firmware versions in the code.

**pylontech.cpp**
`const int parsed = sscanf(
      buffer.c_str(), "%d %d %d %d %d %d %d %d %d %d %d %d %7s %7s %7s %7s %d%% %*d-%*d-%*d %*d:%*d:%*d 
      %*s %*s %d %5s %*s", &l.bat_num, &l.volt, &l.curr, &l.tempr, &l.tlow, &l.tlowid, &l.thigh, &l.thighid, &l.vlow, 
      &l.vlowid, &l.vhigh, &l.vhighid, l.base_st, l.volt_st, l.curr_st, l.temp_st, &l.coulomb, &l.btime, &l.bvst, &l.btst, 
      mostempr_s, &l.mtst);`

and

`if (parsed != 19) {`

as well as **pylontech.h**
`int bat_num = 0, volt, curr, tempr, tlow, tlowid, thigh, thighid, vlow, vlowid, vhigh, vhighid, coulomb, btime, bvst, mostempr, btst, mtst;`

elpamyelhsa, 2025-02-09:
Frimware from US5000-B with build date of Dec-2024 debug output

**[uart_debug:158]: >>> "pwr\n"**
```
pwr\n
\r@\r\r\n
Power Volt   Curr   Tempr  Tlow   Tlow.Id  Thigh  Thigh.Id Vlow   Vlow.Id  Vhigh  Vhigh.Id Base.St  Volt.St  Curr.St  Temp.St  Coulomb  Time                 B.V.St   B.T.St  MosTempr M.T.St   SysAlarm.St\r\r\n
1     50661  17050  35000  33100  12       34500  0        3377   0        3378   6        Charge   Normal   Normal   Normal   90%      2025-02-09 14:17:44  Normal   Normal  34900    Normal   Normal  \r\r\n
2     50648  14770  35200  33300  12       34600  0        3372   7        3378   11       Charge   Normal   Normal   Normal   89%      2025-02-09 14:17:42  Normal   Normal  35800    Normal   Normal  \r\r\n
3     50659  14575  35400  34400  12       35000  0        3377   0        3378   4        Charge   Normal   Normal   Normal   90%      2025-02-09 14:17:43  Normal   Normal  35700    Normal   Normal  \r\r\n
4     50649  15000  35800  35200  8        35800  0        3376   1        3378   0        Charge   Normal   Normal   Normal   89%      2025-02-09 14:17:42  Normal   Normal  36200    Normal   Normal  \r\r\n
5     -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       \r\r\n
6     -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       \r\r\n
7     -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       \r\r\n
8     -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       \r\r\n
9     -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       \r\r\n
10    -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       \r\r\n
11    -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       \r\r\n
12    -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       \r\r\n
13    -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       \r\r\n
14    -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       \r\r\n
15    -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       \r\r\n
16    -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       \r\n
\rCommand completed successfully\r\n
\r$$\r\n
```