esphome/issues #5891: The task_wdt should be configurable

amluto, 2024-06-09:
### The problem

I'm trying to get an AirGradient Open working reliably on ESPHome, and it's unstable -- it reboots due to task_wdt on a regular basis, enough so that the device is essentially nonfunctional.  I'm not the only one with problems like this -- here's a whole thread with no resolution:

https://forum.airgradient.com/t/airgradient-pro-with-esphome-display-keeps-turning-back-on/641/55

Having poked around a bit and read far too much about everyone's experiences here, I suspect the problem is straightforward: the MCU is overloaded.  task_wdt is, by default, set up to watch the *idle* task, and if the system doesn't go idle for 5 seconds, it reboots.  The offending configurations seem to be using two software serial UARTs, and if there is nothing specifically set up to feed the TWDT, boom.

On the one hand, it ought to be possible to find every offending code path that could take too long and make sure to explicitly feed the watchdog.  On the other hand, this is silly -- all the code is working just fine, if occasionally laggy due to bursts of activity, and being able to cleanly configure the TWDT from the YAML config would be great.

Moreover, this hardware has an off-chip hardware watchdog, so there is no need for the TWDT!

Hence my feature request: can we please have some options to change the TWDT timeout and, optionally, to turn it off entirely?

(Note: I've personally reproduced this issue with both the arduino and the esp-idf frameworks.)

### Which version of ESPHome has the issue?

2024.5.5

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Anything in the logs that might be useful for us?

```txt
E (11282) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (11282) task_wdt:  - loopTask (CPU 0)
E (11282) task_wdt: Tasks currently running:
E (11282) task_wdt: CPU 0: IDLE
E (11282) task_wdt: Aborting.

(Not much to see.  The TWDT rebooted machine.  Some of the forum posts carefully dug up some stack traces, but the general theme seems to be that nothing is actually wrong other than often being in the middle of bitbanging a UART.)
```

### Additional information

Here are the hardware specs:

https://www.airgradient.com/documents/spec-sheets/Spec_Sheet_AirGradient_Open_Air_O-1PP.pdf

Note the pair of chatty air quality sensors and the hardware watchdog.

ssieb, 2024-06-09:
If you're using an ESP32, then it's not software serial.  And there's no way you should be trying to run two software serial ports on an 8266.

amluto, 2024-06-09:
> If you're using an ESP32, then it's not software serial. And there's no way you should be trying to run two software serial ports on an 8266.

Yeah, I found that after reading yet more docs.  This is ESP32-C3-MINI.

But I think my main point stands: _something_ is causing the TWDT to reboot, and it would be really nice to be able to adjust the TWDT configuration.

ssieb, 2024-06-09:
And I think that it's much better to find out what's causing it and fix that.  Anything that triggers the WDT is a bug and I don't want to just hide it.

amluto, 2024-06-10:
For whatever reason, I currently can't trigger the TWDT failure with what I thought was the failing configuration.  (But it was only failing some of the time.  I don't know precisely what's up.)

But I still think the TWDT behavior is odd.  I don't have much embedded programming experience, but I have plenty of kernel programming experience, and the idea of forcibly rebooting a system because it didn't go idle for five seconds seems quite problematic to me, for quite a few reasons:

1. 5 seconds may not be long enough to reliably establish a wifi connection.  Rebooting before wifi comes up means that debugging at a distance is essentially impossible.

2. The trigger condition is bizarre.  It's not detecting that anything hung or that critical tasks aren't running or that a task is running forever without scheduling -- it's detecting that a CPU was fully utilized for 5 seconds.  On any general purpose OS, even a real-time one, there's nothing wrong with being fully utilized, for 5 seconds or even indefinitely.  esphome isn't a general purpose OS, and it seems like esphome ought not to be fully utilized, but I can imagine a user that intentionally does something CPU bound.

3. The remedy seems suboptimal.  If esphome considers a TWDT even to be an error, then maybe esphome should try to take a useful action.  Logging something seems useful.  Rebooting if the system is non-idle for too long _and_ there is no successful data upload via API for some time period seems potentially useful.  Finding a way to log the error and a little trace of what esphome was doing seems useful.

oliv3r, 2025-01-15:
Just bumped into this and I agree, the wdt should be configureable from YAML, sometimes we know things take time, and for this case 10seconds would be more ideal. In the end, the purpose of the WDT is to reset the board, for the most part if something is 'hung' and the wdt never gets tickled again. 5s is a choice, and for most cases this choice is fine. But especially for debugging, configuration is really needed.

In my case, is my board hanging/stuck or is it just busy. The serial console just shows init stuff, and then a reboot with a reason 'wdt'. So for me to debug this, I need to extend the delay to see if that 'hides' the problem so I can dig deeper.

dwmw2, 2025-02-10:
I'm seeing a similar problem on an ESP32S3 [PD Stepper](https://github.com/joshr120/PD-Stepper) board.:
```
[15:02:18]E (82918) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
[15:02:18]E (82918) task_wdt:  - loopTask (CPU 0)
[15:02:18]E (82918) task_wdt: Tasks currently running:
[15:02:18]E (82918) task_wdt: CPU 0: IDLE0
[15:02:18]E (82918) task_wdt: CPU 1: IDLE1
[15:02:18]E (82918) task_wdt: Aborting.
[15:02:18]E (82918) task_wdt: Print CPU 0 (current core) backtrace
[15:02:18]
[15:02:18]
[15:02:18]
[15:02:18]
[15:02:18]Backtrace: 0x4037abc7:0x3fca53d0 0x4203713d:0x3fca53f0 0x4037d97f:0x3fca5410
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4037abc7: xt_utils_wait_for_intr at /home/ANT.AMAZON.COM/dwmw/.platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at /home/ANT.AMAZON.COM/dwmw/.platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/components/esp_hw_support/cpu.c:111
WARNING Decoded 0x4203713d: esp_vApplicationIdleHook at /home/ANT.AMAZON.COM/dwmw/.platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/components/esp_system/freertos_hooks.c:59
WARNING Decoded 0x4037d97f: prvIdleTask at /home/ANT.AMAZON.COM/dwmw/.platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/components/freertos/FreeRTOS-Kernel/tasks.c:4439

```