esphome/issues #5892: ESPHome Sprinkler Controller, activating Main switch when no valves enabled causes valves to enable (unexpected behavior)

staffy, 2024-06-09:
### The problem

As stated in title, when no valves/zones are enabled (presumably because the user does not want them to activate for some reason) and the main switch is activated (which could be caused by for example an automation running once per day or so), then all valves are enabled and a full irrigation cycle is run, which is unexpected. If one or more valves are enabled, then only these valves are activated as expected.

The expected behavior with no valves enabled would be to immediately finish the cycle with no valves activated and no configuration changes.

### Which version of ESPHome has the issue?

2024.5.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

4 relay board

### Component causing the issue

Sprinkler

### Example YAML snippet

```yaml
switch:
  - platform: gpio
    pin: GPIO32
    id: valve_1
    internal: true
  - platform: gpio
    pin: GPIO33
    id: valve_2
    internal: true
#  - platform: gpio
#    pin: GPIO25
#    id: valve_3
#    internal: true
#  - platform: gpio
#    pin: GPIO26
#    id: valve_4
#    internal: true

sprinkler:
  - id: sprinkler_controller
    main_switch: "Bevattning Huvudswitch"
    auto_advance_switch: "Bevattning Auto Advance"
    valve_overlap: 1s
    valves:
      - valve_switch: "Bevattning Hörnet"
        enable_switch: "Bevattning Hörnet aktiv"
        run_duration_number:
          name: "Hörnet Körtid"
          unit_of_measurement: min
          initial_value: 10
        valve_switch_id: valve_1
        
      - valve_switch: "Bevattning Altan"
        enable_switch: "Bevattning Altan aktiv"
        run_duration_number:
          name: "Altan Körtid"
          unit_of_measurement: min
          initial_value: 10
        valve_switch_id: valve_2
```

### Anything in the logs that might be useful for us?

```txt
[20:50:54][I][app:100]: ESPHome version 2024.5.4 compiled on Jun  9 2024, 20:17:35
[20:50:54][I][app:102]: Project esphome.web version 1.0
[20:50:54][C][status_led:019]: Status LED:
[20:50:54][C][status_led:020]:   Pin: GPIO23
[20:50:54][C][wifi:580]: WiFi:
[20:50:54][C][wifi:408]:   Local MAC: xxx
[20:50:54][C][wifi:413]:   SSID: [redacted]
[20:50:54][C][wifi:416]:   IP Address: xxx
[20:50:54][C][wifi:420]:   BSSID: [redacted]
[20:50:54][C][wifi:421]:   Hostname: 'esphome-web-ff737c'
[20:50:54][C][wifi:423]:   Signal strength: -74 dB ▂▄▆█
[20:50:54][C][wifi:427]:   Channel: 11
[20:50:54][C][wifi:428]:   Subnet: 255.255.255.0
[20:50:54][C][wifi:429]:   Gateway: 192.168.1.1
[20:50:54][C][wifi:430]:   DNS1: 192.168.1.1
[20:50:54][C][wifi:431]:   DNS2: 0.0.0.0
[20:51:00][C][sprinkler:068]: Sprinkler Switch 'Bevattning Hörnet'
[20:51:03][D][switch:016]: 'Bevattning Hörnet aktiv' Turning OFF.
[20:51:03][D][switch:055]: 'Bevattning Hörnet aktiv': Sending state OFF
[20:51:04][D][switch:016]: 'Bevattning Altan aktiv' Turning OFF.
[20:51:04][D][switch:055]: 'Bevattning Altan aktiv': Sending state OFF
[20:51:09][D][esp32.preferences:114]: Saving 2 preferences to flash...
[20:51:09][D][esp32.preferences:143]: Saving 2 preferences to flash: 0 cached, 2 written, 0 failed
[20:51:16][D][sensor:094]: 'WiFi RSSI': Sending state -75.00000 dBm with 0 decimals of accuracy
[20:51:24][D][switch:012]: 'Bevattning Huvudswitch' Turning ON.
[20:51:24][D][switch:012]: 'Bevattning Auto Advance' Turning ON.
[20:51:24][D][switch:055]: 'Bevattning Auto Advance': Sending state ON
[20:51:24][D][switch:012]: 'Bevattning Hörnet aktiv' Turning ON.
[20:51:24][D][switch:055]: 'Bevattning Hörnet aktiv': Sending state ON
[20:51:24][D][switch:012]: 'Bevattning Altan aktiv' Turning ON.
[20:51:24][D][switch:055]: 'Bevattning Altan aktiv': Sending state ON
[20:51:24][D][sprinkler:1427]: CYCLE is starting valve 0 for 60 seconds, cycle 1 of 1
[20:51:24][D][switch:012]: 'valve_1' Turning ON.
[20:51:24][D][switch:055]: 'valve_1': Sending state ON
[20:51:24][D][switch:055]: 'Bevattning Huvudswitch': Sending state ON
[20:51:24][W][component:237]: Component api took a long time for an operation (65 ms).
[20:51:24][W][component:238]: Components should block for at most 30 ms.
[20:51:24][D][switch:055]: 'Bevattning Hörnet': Sending state ON
[20:52:09][D][esp32.preferences:114]: Saving 3 preferences to flash...
[20:52:09][D][esp32.preferences:143]: Saving 3 preferences to flash: 0 cached, 3 written, 0 failed
[20:52:16][D][sensor:094]: 'WiFi RSSI': Sending state -73.00000 dBm with 0 decimals of accuracy
[20:52:23][D][sprinkler:1296]: Marking valve 0 complete
[20:52:23][D][sprinkler:1427]: CYCLE is starting valve 1 for 60 seconds, cycle 1 of 1
[20:52:23][D][switch:012]: 'valve_2' Turning ON.
[20:52:23][D][switch:055]: 'valve_2': Sending state ON
[20:52:23][D][switch:055]: 'Bevattning Altan': Sending state ON
[20:52:24][D][switch:016]: 'valve_1' Turning OFF.
[20:52:24][D][switch:055]: 'valve_1': Sending state OFF
[20:52:24][D][switch:055]: 'Bevattning Hörnet': Sending state OFF
[20:53:16][D][sensor:094]: 'WiFi RSSI': Sending state -74.00000 dBm with 0 decimals of accuracy
[20:53:22][D][sprinkler:1296]: Marking valve 1 complete
[20:53:23][D][switch:016]: 'valve_2' Turning OFF.
[20:53:23][D][switch:055]: 'valve_2': Sending state OFF
[20:53:23][D][switch:055]: 'Bevattning Altan': Sending state OFF
[20:53:24][D][sprinkler:1449]: All valves stopped, including pumps
[20:53:24][D][switch:055]: 'Bevattning Huvudswitch': Sending state OFF
[20:54:16][D][sensor:094]: 'WiFi RSSI': Sending state -72.00000 dBm with 0 decimals of accuracy
[20:55:16][D][sensor:094]: 'WiFi RSSI': Sending state -75.00000 dBm with 0 decimals of accuracy
```

kbx81, 2024-06-10:
This is not unexpected behavior as it is clearly stated [in the documentation](https://esphome.io/components/sprinkler#sprinkler-start-full-cycle-action).

staffy, 2024-06-10:
Very clearly stated, thanks for pointing that out! But it still seems strange to me, I cannot see a reason why anyone would want that behavior?

kbx81, 2024-06-10:
Consider this situation:

No zones are enabled. A user wishes to run a cycle of the system. The user walks up to the hardware and presses a button to start the system. Nothing happens. User is confused and assumes something is broken. (Chaos ensues...)

Note that this scenario plays out exactly the same way if you're using a switch or button on a dashboard to activate the system.

Part of the motivation for the behavior was to minimize situations where the controller just does nothing. It's not obvious why the controller won't start unless you happen to be watching the device's logs as you're interacting with it. This behavior is quite confusing to an end-user who just wants to switch on their sprinklers (and who knows nothing about how to troubleshoot hardware...or at least isn't thinking about it at the time).

I understand what you're trying to do, but--trust me--if it worked the way you're suggesting/expecting, you will end up wasting a bunch of time one day by confusing _yourself. Ask me how I know._ 😅 You will spend a lot less time tweaking your automation to cope with this behavior than you will _waste_ trying to figure out why your controller won't turn on when you try to use it one day and forget that no zones are enabled. 😇

AndreKR, 2025-04-11:
I get your reasoning, but can we at least have a config option to change this behavior?

I turned off all my valves in autumn (but kept the automation for the main switch on) and I just found out that it has been watering into the drain all winter because the valves turned on by themselves!

kbx81, 2025-04-12:
Use the `standby_switch` if you want to shut down your system for the winter.

closed 2025-04-12 by kbx81

AndreKR, 2025-04-12:
> Use the standby_switch if you want to shut down your system for the winter

If I show the standby_switch in the dashboard instead of the automation, how is that better? I would still need to have the same message, explaining that it is impossible to turn off all valves at the same time and instead the standby switch has to be used.

kbx81, 2025-04-12:
The valve enable switches are not intended as a way to stop the controller from running a cycle; put another way, if you don't want a cycle to run, don't tell the controller to run a cycle.
OTOH, if you want to "lock out" the controller, that's what the standby switch is for -- winter shutdown or perhaps maintenance.
As you are using an automation to start your system, it sounds like this could be solved in your automation: if all of your valves are disabled, just don't trigger the controller to start. Now you don't need any extra controls on your dashboard. 🙂

AndreKR, 2025-04-13:
> this could be solved in your automation: if all of your valves are disabled, just don't trigger the controller to start

That is actually a quite decent workaround. I can use that in the schedule automation, for the UI button and even for the physical button. Only downside is that the physical button can no longer trigger a complete cycle locally, Home Assistant has to be available to start a cycle, but I can live with that.

I still find it extremely counterintuitive, but this solves my immediate problem.