esphome/issues #5896: Unable to compile S3 Voice Assistant on Windows

OlwinFroon, 2024-06-11:
### The problem

I followed the instructions [HERE](https://esphome.io/guides/installing_esphome.html) to install Python and ESPHome on my Windows machine.

Then I tried to adopt a previously flashed ESP32 S3 Box 3.

The compiling process crashed with the message 'ImportError: failed to find libmagic.  Check your installation' and 'Component not found: image'

### Which version of ESPHome has the issue?

2024.5.5

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
substitutions:
  name: esp32-s3-box-3-redacted
  friendly_name: S3 Box A
packages:
  esphome.voice-assistant: github://esphome/firmware/wake-word-voice-assistant/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: 4ZtOEJ/bUR7/gcLp9vR33JprrzI8mO0YipdLhC72k+E=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Additional information

I installed Python 3.12.4 amd-64 for Windows.

Further digging revealed a related post [HERE](https://github.com/esphome/issues/issues/5427) which might help fixing the problem.

OlwinFroon, 2024-06-11:
Somebody suggested to fix this with `pip install python-magic-bin` (version 0.4.14).

Didn't work, though.

The new error message is now `magic.magic.MagicException: b'could not find any valid magic files!'`.

guillempages, 2024-09-01:
https://github.com/esphome/esphome/pull/7365 should fix this.
You can already use it (until it gets merged into core) by adding
```
external_components:
  - source: github://pr#7365
    components: [image]
```
to your yaml configuration

closed 2024-10-05 by clydebarrow