esphome/issues #5897: Keeping states - in RAM and/or Flash - description not clear or not working

starbrights, 2024-06-11:
### The problem

I am try to run (on BK7321 - maybe that is part of problem) a rollershutter. That all works fine but I could find out
* what state can survive an reboot
* what state can survive a repower

I found a few different things about it and it is very time consuming to test (need to flash, need to cut power/wifi) but although I invest quite some time I don't get a final clue.

It seem that *(esp_)restore_from_flash* is not supported for my chip. Ok.
But I found also (in globals) the *restore_value* - these doesn't seem to work for me. Not for reboot nor for repower (power off/on).
On the other hand, if I do an *App.safe_reboot();* I can observe that in this case the position of the cover (which is an internal variable obviously) is saved. I also get in log: 
`[D][lt.preferences:132]: Saving 2 preferences to flash: 0 cached, 2 written, 0 failed`
But which one and how to control that is unclear.

### Which version of ESPHome has the issue?

ESPHome version 2024.5.4

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

-

### What platform are you using?

BK72XX

### Board

moes curtain switch

### Component causing the issue

cover, output, binary_sensor, switch

### Example YAML snippet

```yaml
####################################################
# short up/down: cover up/down
# any key if cover is moving: stop
# stop while not moving: cover to pref_pos
#
# long stop: set a new pref_pos
# long up:   switch up (overrule cover)
# long down: switch down (overrule cover)
#
# long long up: reboot
# long long stop: toggle timer (middle red)
# long long down: toogle LED
####################################################
esphome:
  name: moes
  on_boot:
    priority: -10
    then:
      - output.turn_on: gpio_backlight

bk72xx:
  board: generic-bk7231n-qfn32-tuya

logger:

web_server:

captive_portal:

[... 306 lines cut ...]
              - lambda: 'return id(timer_active);'
            then:
              - cover.open: my_cover
            else:
              - logger.log: "Timer is inactive, skipping action."

interval:
  - interval: 30s
    then:
      - mqtt.publish:
          topic: moes/timer_active
          payload: !lambda 'return String(id(timer_active)).c_str();'

      - mqtt.publish:
          topic: moes/pos
          payload: !lambda 'return String(id(my_cover).position).c_str();'

      - mqtt.publish:
          topic: moes/pref
          payload: !lambda 'return String(id(pref_pos)).c_str();'

      - mqtt.publish:
          topic: moes/backlight
          payload: !lambda 'return String(id(backlight_active)).c_str();'
```

ssieb, 2024-06-11:
You didn't enable restoring for the globals.
This isn't an issue for github.  If you need help, come ask on the esphome discord server or one of the forums.

closed 2024-06-11 by ssieb

starbrights, 2024-06-11:
I added enable_restore, but at least after repower that values was gone.

starbrights, 2024-06-11:
I found an issue:  I could find this in log:
`17:38:33][I][ota:117]: Boot seems successful, resetting boot loop counter.
[17:38:33][D][lt.preferences:104]: Saving 4 preferences to flash...
[17:38:33][D][lt.preferences:132]: Saving 4 preferences to flash: 0 cached, 4 written, 0 failed
`
but only once and always around 5min after bootup. Changing position of cover or other variables that are configured to be as restore_value doesn't trigger a new "save".