esphome/issues #5899: 2024.5.5 release: const.py is missing CONF_WEB_SERVER_ID variable

RobertD502, 2024-06-11:
### The problem

After updating to ESPHome `2024.5.5` and attempting to update one of my devices, it is failing as the `const.py` file is missing the `CONF_WEB_SERVER_ID` variable.

The following error message is encountered:

``` log
INFO ESPHome 2024.5.5
INFO Reading configuration /config/esphome/petlibro-water-fountain.yaml...
ERROR Unable to import component datetime:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 176, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/257f044b/esphome/components/datetime/__init__.py", line 6, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_WEB_SERVER_ID' from 'esphome.const' (/esphome/esphome/const.py)
Failed config

datetime: [source /config/esphome/petlibro-water-fountain.yaml:67]
  
  Component not found: datetime.
  - platform: template
    name: Filter install date
    id: filter_install_dt
    type: datetime
    optimistic: True
    initial_value: 
      year: 2024
      month: 1
      day: 1
      hour: 0
      minute: 1
      second: 1
    restore_value: True
    on_value: 
      then: 
        - component.update: filter_life_remaining
button.template: [source /config/esphome/petlibro-water-fountain.yaml:289]
  platform: template
  name: Reset Filter
  on_press: 
    then: 
      - 
        Unable to find action with the name 'datetime.datetime.set'.
        datetime.datetime.set: 
          id: filter_install_dt
          datetime: !lambda |-
            return id(homeassistant_time).utcnow();
```

I entered into the container and, sure enough, the `CONF_WEB_SERVER_ID` is missing from the `const.py` file. To ensure that it wasn't some weird bug that occurred during my update of the add-on, I also looked at the `2024.5.5` release's source code and can see that the `CONF_WEB_SERVER_ID` variable is missing.

## Note:

I am using the `esp-idf` framework, so I am having to pull in the some of the esphome components manually.

``` yaml
esp32:
  board: esp32-c3-devkitm-1
  variant: ESP32C3
  framework:
    type: esp-idf

external_components:
  - source: github://esphome/esphome@dev
    components: [ datetime, homeassistant, time ]
```

Since I am pulling in, for example, the `datetime` component manually, its __init__ file is importing the `CONF_WEB_SERVER_ID` variable, but the PR that includes this change & the `const.py` change ([#6445](https://github.com/esphome/esphome/pull/6445)) was not part of the `2024.5.5` release (https://github.com/esphome/esphome/compare/2024.5.4...2024.5.5).

Just to be sure, since I am using the esp-idf framework I have to pull in those components (that are part of esphome) as external components, correct? Prior to defining them as external components (for example, in the `2024.5.4` version of ESPHome), my builds failed for this device with an error stating that those components are missing.

### Which version of ESPHome has the issue?

2024.5.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.2

### What platform are you using?

ESP32-IDF

### Board

esp32-c3-devkitm-1

### Component causing the issue

datetime

ssieb, 2024-06-11:
You're pulling in a newer version of the component that has changes that use new constants, but you don't have the core changes that include those constants.  (Not possible with external components.)  So you can't do this.  Why are you even trying to do this?

closed 2024-06-11 by ssieb

RobertD502, 2024-06-11:
> You're pulling in a newer version of the component that has changes that use new constants, but you don't have the core changes that include those constants. (Not possible with external components.) So you can't do this. Why are you even trying to do this?

As mentioned above, unless I define those components as external components, the build fails with an error stating that the `datetime`, `homeassistant`, and `time` components are missing - based on my findings at the time, since I am defining the framework as `esp-idf`, I have to define those components as external components. Otherwise, ESPHome complains that the components are missing. Unless I am misunderstanding, how else am I supposed to utilize the datetime, homeassistant, and time components with the framework set to esp-idf?

ssieb, 2024-06-11:
If you use esp-idf, you often have to do a clean build if you add or remove components from the config.