esphome/issues #5900: save of preferences are just triggered once after 5min

starbrights, 2024-06-11:
### The problem

I have a rollerblind switch that works well thanks to the great ESPHOME framework. 
But there is an issue restore ... but now I know it is not restore but saving preferences. Checking the logs it turns out that this happens exactly once after 300s, but not anymore after changing the variables or position of color.
First I was confused about that results was so mixed. Sometimes restore works, sometimes not. But now it is clear, it depends on whether changes happens before "save" in the first 5min.

### Which version of ESPHome has the issue?

ESPHome version 2024.5.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

-

### What platform are you using?

BK72XX

### Board

moes curtain switch

### Component causing the issue

cover, output, binary_sensor, switch

### Example YAML snippet

```yaml
...
globals:
  - id: pref_pos
    type: int
    initial_value: '55'
    restore_value: 'true'
  - id: timer_active
    type: bool
    initial_value: 'true'
    restore_value: 'true'
  - id: backlight_active
    type: bool
    initial_value: 'true'
```

### Anything in the logs that might be useful for us?

```txt
[17:33:41][D][main:413]: run on_boot
[17:33:41][I][app:062]: setup() finished successfully!
[17:33:41][I][app:100]: ESPHome version 2024.5.4 compiled on Jun 11 2024, 17:32:14
[17:33:41][C][status_led:019]: Status LED:
[17:33:41][C][status_led:020]:   Pin: 26
[17:33:41][C][wifi:580]: WiFi:
....
[17:38:09][D][cover:076][async_tcp]: 'moes1' - Setting
[17:38:09][D][cover:084][async_tcp]:   Position: 50%
[17:38:09][D][switch:012][async_tcp]: 'Relay down' Turning ON.
[17:38:09][D][switch:055][async_tcp]: 'Relay down': Sending state ON
[17:38:10][D][switch:016]: 'Relay up' Turning OFF.
[17:38:10][D][switch:016]: 'Relay down' Turning OFF.
[17:38:10][D][switch:055]: 'Relay down': Sending state OFF
[17:38:10][D][cover:170]: 'moes1' - Publishing:
[17:38:10][D][cover:173]:   Position: 50%
[17:38:10][D][cover:186]:   Current Operation: IDLE
[17:38:13][D][binary_sensor:036]: 'binary_switch_stop': Sending state ON
[17:38:18][D][binary_sensor:036]: 'binary_switch_stop': Sending state OFF
[17:38:18][D][main:322]: Very long press stop detected
[17:38:18][D][main:313]: set new pref_pos 50
[17:38:30][D][sensor:093]: 'Uptime': Sending state 297.06799 s with 0 decimals of accuracy
[17:38:33][I][ota:117]: Boot seems successful, resetting boot loop counter.
[17:38:33][D][lt.preferences:104]: Saving 4 preferences to flash...
[17:38:33][D][lt.preferences:132]: Saving 4 preferences to flash: 0 cached, 4 written, 0 failed
[17:39:06][D][binary_sensor:036]: 'binary_switch_stop': Sending state ON
[17:39:08][D][binary_sensor:036]: 'binary_switch_stop': Sending state OFF
[17:39:08][D][main:308]: Long press stop detected
[17:39:08][D][main:300]: timer_active: 0
```

mick96, 2024-07-17:
Same issue is happening here with time based cover on bk7231N.
It seems like _sync()_ it is only called once while it is supposed to be scheduled every _flash_intervall_rate_.

starbrights, 2024-07-18:
I didn't dive into it, I am not so familiar with c++programming. This and deep sleep is what I see is missing at the BK72xx plattform.