esphome/issues #5901: The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h

bsaurusrex, 2024-06-12:
### The problem

Build error with remote_base:

### Which version of ESPHome has the issue?

2024.5.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.1

### What platform are you using?

ESP32

### Board

esp32-c6-devkitc-1

### Component causing the issue

remote_transmitter

### Example YAML snippet

```yaml
remote_transmitter:
  pin: GPIO10
  id: ir_transmitter
  carrier_duty_percent: 50%
```

### Anything in the logs that might be useful for us?

```txt
In file included from src/esphome/components/remote_base/remote_base.h:12,
                 from src/esphome/components/remote_base/abbwelcome_protocol.h:5,
                 from src/esphome.h:44,
                 from src/main.cpp:3:
/data/cache/platformio/packages/framework-espidf/components/driver/deprecated/driver/rmt.h:18:2: warning: #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h" [-Wcpp]
   18 | #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h"
      |  ^~~~~~~
src/esphome/components/remote_base/abbwelcome_protocol.h: In member function 'std::string esphome::remote_base::ABBWelcomeData::to_string(uint8_t) const':
src/esphome/components/remote_base/abbwelcome_protocol.h:147:64: error: format '%X' expects argument of type 'unsigned int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  147 |       info = str_sprintf(this->get_three_byte_address() ? "[%06X %s %06X] Type: %02X" : "[%04X %s %04X] Type: %02X",
      |                                                             ~~~^
      |                                                                |
      |                                                                unsigned int
      |                                                             %06lX
  148 |                          this->get_source_address(), this->get_retransmission() ? "»" : ">",
      |                          ~~~~~~~~~~~~~~~~~~~~~~~~~~             
      |                                                  |
      |                                                  uint32_t {aka long unsigned int}
src/esphome/components/remote_base/abbwelcome_protocol.h:147:72: error: format '%X' expects argument of type 'unsigned int', but argument 4 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  147 |       info = str_sprintf(this->get_three_byte_address() ? "[%06X %s %06X] Type: %02X" : "[%04X %s %04X] Type: %02X",
      |                                                                     ~~~^
      |                                                                        |
      |                                                                        unsigned int
      |                                                                     %06lX
  148 |                          this->get_source_address(), this->get_retransmission() ? "»" : ">",
  149 |                          this->get_destination_address(), this->get_message_type());
      |                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                
      |                                                       |
      |                                                       uint32_t {aka long unsigned int}
src/esphome/components/remote_base/abbwelcome_protocol.h:147:94: error: format '%X' expects argument of type 'unsigned int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  147 |       info = str_sprintf(this->get_three_byte_address() ? "[%06X %s %06X] Type: %02X" : "[%04X %s %04X] Type: %02X",
      |                                                                                           ~~~^
      |                                                                                              |
      |                                                                                              unsigned int
      |                                                                                           %04lX
  148 |                          this->get_source_address(), this->get_retransmission() ? "»" : ">",
      |                          ~~~~~~~~~~~~~~~~~~~~~~~~~~                                           
      |                                                  |
      |                                                  uint32_t {aka long unsigned int}
src/esphome/components/remote_base/abbwelcome_protocol.h:147:102: error: format '%X' expects argument of type 'unsigned int', but argument 4 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  147 |       info = str_sprintf(this->get_three_byte_address() ? "[%06X %s %06X] Type: %02X" : "[%04X %s %04X] Type: %02X",
      |                                                                                                   ~~~^
      |                                                                                                      |
      |                                                                                                      unsigned int
      |                                                                                                   %04lX
  148 |                          this->get_source_address(), this->get_retransmission() ? "»" : ">",
  149 |                          this->get_destination_address(), this->get_message_type());
      |                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                              
      |                                                       |
      |                                                       uint32_t {aka long unsigned int}
cc1plus: some warnings being treated as errors
*** [.pioenvs/office-aircon/src/main.cpp.o] Error 1
```

### Additional information

I believe this new rmt drive is esp-idf 5.0+

esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  flash_size: 8MB
  #partitions: partitions_zb.csv
  framework:
    type: esp-idf
    version: 5.2.1
    platform_version: 6.7.0

@jesserockz

jesserockz, 2024-06-12:
Don't ping maintainers. And also don't create issues for platforms/variants/chips that are not yet supported by the latest release.

closed 2024-06-12 by jesserockz

Goosneck, 2024-06-20:
I have the same issue (esp-idf 5.2.1 with esp32-c6-devkitc-1). What knowledge would I need in order to solve (or "bypass it") to be able to transmit and receive IR signals (Just generally lost for the past 2 days)?