esphome/issues #5903: M5stack Atom Echo's button doesn't trigger Assist

Defozo, 2024-06-12:
### The problem

I uploaded the binary provided on https://www.home-assistant.io/voice_control/thirteen-usd-voice-remote/.
I believe the Atom Echo should trigger voice recognition when the button is pressed but this doesn't happen.

### Which version of ESPHome has the issue?

2024.5.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.2

### What platform are you using?

ESP32

### Board

M5Stack Atom Echo

### Additional information

I can confirm the button works hardware-wise as I enabled it in HASS and the presses are correctly detected.

JustEnoughDucks, 2024-06-18:
Is it correctly sending your voice over?

I can use a debug session and the Atom Echo will hear me and interpret commands, but indeed the button doesn't trigger the voice assistant

Defozo, 2024-06-19:
Yes, it's successfully transmitting the voice over. Even though I haven't listened to the sample myself, the recognition via Whisper appears to be working exceptionally well.

pablowako, 2025-06-17:
bump
This issue persists, at least for me.

Both the center button and the side button have no use, and don't trigger anything. Cant even see it in the logs

CodyReichert, 2025-06-18:
I noticed this too. Pretty sure it is a bug in [the default yaml config](https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml), but I didn't install from that source so I'm not sure if it uses the same thing (you'd see a reference to that default yaml on github in your device's yaml), but basically everything else works just fine, so a config issue would make sense.