esphome/issues #5904: Will not update firmware. Platform tag required

dualtronic, 2024-06-13:
### The problem

Cannot flash firmware on M5 Stack ESP32 (bluetooth proxy).  Requires Platform tag, but also fails when including that tag because the online configuration already uses new platform (ESP32:) description tag.

### Which version of ESPHome has the issue?

2024.6.0b2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.2

### What platform are you using?

ESP32

### Board

ESP32 (M5Stack Atom)

### Component causing the issue

ESPHome

### Example YAML snippet

```yaml
substitutions:
  name: esp32-bluetooth-proxy-9d2178
  friendly_name: Bluetooth Proxy 9d2178
packages:
  esphome.bluetooth-proxy: github://esphome/firmware/bluetooth-proxy/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: DHrH6oy3Sx2YBWzk0jnPsk9qLM6oOHzc1vdOaLcgsQo=
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
Installation trace below:

INFO ESPHome 2024.6.0b2
INFO Reading configuration /config/esphome/esp32-bluetooth-proxy-9d2178.yaml...
Failed config

At least one platform must be specified for 'ota'; add 'platform: esphome' for original OTA functionality
```

ssieb, 2024-06-13:
I think you're misunderstanding the error.  What did you try to do that didn't work?

alexruffell, 2024-06-13:
I am also getting the same error on all nodes I tried to update. One of them had a different, but related, error:

```
INFO ESPHome 2024.6.0b2
INFO Reading configuration /config/esphome/fume-extractor.yaml...
Failed config

ota.unknown: [source /config/esphome/fume-extractor.yaml:41]
  
  'ota' requires a 'platform' key but it was not specified.
  password: !secret iot_wifi_password
```

I can't find any docs (possibly because they are still beta?) that state what the platform key is and how to configure it.

ssieb, 2024-06-13:
https://next.esphome.io/changelog/2024.6.0.html#ota-platforms

dualtronic, 2024-06-18:
@alexruffell  Your suggestion fixed my problem. Thank you!  I had no idea about this changed config requirement.
 I hope there is an easy migration pathway for other users like me who don't keep up with these changes, otherwise there will be alot people with the same problem.

cociweb, 2024-06-20:
It might be better to pin this issue out as it is breaking change and also breaks the auto updates everyone....

kastro123, 2024-06-20:
@alexruffell I had the same problem. Thank you for hint, it's working now.

tomf247, 2024-06-20:
Same issue for me. IS commenting out the two lines in the ota section is the correct solution? I get it that it works now, but will it bork something later?

alexruffell, 2024-06-20:
@tomf247 `safe_mode` is no longer part of OTA (it was shortly before my post which is why documentation had not been updated yet) so it will error out if you add it. As for the password, that is up to you whether you want one or not. If you do, you will need to update the node via USB first to establish one.

osokthedevil, 2024-06-20:
replaced you ota with this

ota:
  platform: esphome

save and update

mworsnop, 2024-06-20:
This is really a bug! I shouldn't have to be going through 45 different YAML and adding platform ESP home. If this was so important there should've been a global update or something. I'm sure glad I don't have 200 of them.

alexruffell, 2024-06-20:
@mworsnop Growing pains, but not a bug. There is a way for you to avoid this in the future by putting common parts of your YAML in what I believe is called packages that you then reference in your devices YAML. This way, one update takes care of them all.

BTW, in yesterday's release party (see YouTube) they did briefly discuss why they opted not to modify everyone's device YAML and I agree with them that it was best not to. It did not take me long to adjust my 58 ESPHome devices. It's not that bad with basic copy & paste...

joshuaboniface, 2024-06-20:
The options were not just "silently rewrite everyones YAML" and this. The third, reasonable option was to treat the absence of any child items in `ota:` as the old-style config, perhaps warn the user of the future deprecation of this configuration syntax, and then leave that fallback in place for some time.

Sure, it "did not take long" to update my X devices (except, it does, because mass-install is not a thing and I have to wait for each one to recompile, upload, and restart... but that's its own issue), but the fact we had to do it at all, after presumably - as I did - getting that very cryptic error, Googling it, finding this issue, and then scrolling to find the workaround. That's frustrating to have to do due to what should be a normal update, especially when said update is coming from HomeAssistant itself and the normal in-place update just fails. And what about those who aren't using the ESPHome addon, who haven't adopted configs that can be easily edited?

alexruffell, 2024-06-20:
The breaking change was highlighted in the Release Notes:

https://esphome.io/changelog/2024.6.0.html

I don't want to start a debate on why they did or did not do something, or whether it could have been done differently, however breaking changes are often a necessary evil for faster progress. I ran into this issue when documentation was not even finished as it was a few days into beta and I was able to resolve the issue in a matter of minutes searching Google for what the error stated. I am not an expert either but I do have close to 60 nodes so I have gone through this type of issue before. Anyhow, I very easily fixed/updated all my nodes, including ones with the other breaking change regarding one_wire & dallas_temp.

As for Repairs, I am guessing that it isn't as easy as you make it seem, and I appreciate the fact that they did NOT modify my YAML automatically... but that is just my preference :)

GSzabados, 2024-06-20:
I truly understand your point, but there has been nothing done to highlight this issue.

2024.6.0 was released, breaking changes are **down 3 pages in the release notes**. Phrased in the most cryptic way as they could be. 

> Separate OTABackend from OTA component https://github.com/esphome/esphome/pull/6459 by [@kbx81](https://github.com/kbx81) (new-integration) (breaking-change)

I have looked at my wireless logs when updated to 2024.6.0, all worked, no highlighted message or anything.
This morning updated to 2024.6.1 and boom, all the updates are failing from HA. Then the logs are failing in ESPHome. It is not really intuitive process, when you know that you are going to bomb everybody's configurations which uses ota, and which one is not?

alexruffell, 2024-06-20:
@GSzabados - Strange... the OTA change was introduced with 2024.6.0 as far as I know. I made the YAML changes during beta as otherwise I could not compile.

You do make a good point on "cryptic way" the GitHub PRs are worded. They may sound clear and simple to developers but I often have a hard time understanding what they are about. Many PRs barely have anything in them so there it is even harder to figure out what the change is about. It would really be nice if the devs added a tiny bit more "general audience" detail on what the PR is about if we are expected to look at the PRs to see if the changes will cause issues to our nodes. Anyhow, most of this work is done by people donating their time and effort so how can one complain? I just appreciate all the hard work and enjoy it a lot even if sometimes it takes much effort to get stuff working.

alexruffell, 2024-06-20:
> The third, reasonable option was to treat the absence of any child items in `ota:` as the old-style config, perhaps warn the user of the future deprecation of this configuration syntax, and then leave that fallback in place for some time.

Yes, this is a nice suggestion but I wonder how feasible it is when you have to modify the compiler for that to happen. What about a YAML validator, similar to the configuration.yaml one, that can alert users to issues before even trying to compile? This may also give them more options on helping the user instead of doing it at compile time. Regardless, I am sure more can be done but it takes people and money... maybe some kind user/developer will take it on :)

> Sure, it "did not take long" to update my X devices (except, it does, because mass-install is not a thing and I have to wait for each one to recompile, upload, and restart... but that's its own issue),...

True... I actually made a basic Node red flow that triggers the update of them all with a simple button press so once I copy & pasted the OTA change to all my nodes, I pressed the button to trigger the mass update. Also, many of my nodes use templates so I only had to update the template (ie.: ~20 Sonoff S31 outlets share a template so one change took care of them all). The "Update All" on the dashboard never works for me as the tab in Chrome freezes up/crashes after a few have run, and since it doesn't keep track of what was updated already, it just starts over... that is why I went with NR (but others use scripts, automations, etc. - I just do everything I can in NR).

daernsinstantfortress, 2024-06-21:
Thanks very much - easy and quick fix (and next time read the release notes - missed the one_wire config change too!)

One thing to note - HA now has a great system for warning of upcoming deprecation to avoid situations like this. Generally a change like this would be implemented in two stages with the new configuration option (`platform: esphome`) being optional for a release or two, and a notification to the user telling them that they need to make the change. And then, at a future release, the new configuration becomes mandatory and will fail if not updated. Indeed, in many cases HA also has a "fix my configuration" button to automatically make the change.

Now that esphome and HA are under the same umbrella, perhaps there's some future opportunity for improvement around this sort of breaking change that is essential in a rapidly developing application, but can be very offputting to non-technical users?

rkhanso, 2024-06-21:
> @mworsnop Growing pains, but not a bug. There is a way for you to avoid this in the future by putting common parts of your YAML in what I believe is called packages that you then reference in your [...]

One problem I see here is that we don't have clairvoyance and know what will be the next section of the code getting a breaking update so we can put those sections in the code into a !secrets type file ahead of time - to later reference and make global changes for those upcoming breaking fixes we don't know about.

alexruffell, 2024-06-21:
@rkhanso You do have clairvoyance to some extent by reading all changes, and above all the breaking changes, both always diligently listed in the release notes. All these notes also point to github where you can dig deeper on each change. Sometimes side effects of a change may have gone unforeseen but
while unfortunate, that is typical in software development that moves at a fast pace. I will gladly take this uncertainty in favor of monthly new features.
Please note that you *don't* need to update your nodes upon each release if they work as you desire, at least not as frequently... however, if you skip x months, when you do update, you will have to review all changelogs of the versions you skipped (easier on github as they are all on one page - search "esphome releases github" in google).

As for the secrets file, it sounds like your understanding is mistaken. It will not protect your code from any changes but rather just give you a place to store login and password information. Other than keeping this info out of view (not in device code you might share) it also allows you to update a login and password (wifi for example) in one spot. Maybe you were referencing templates or packages which allow you to store common code in all your devices in one file so that you can keep device yaml shorter and make any updates easier... but those too will not prevent the effects of changes. Not updating if you don't need to is the only way I know of.

gordon861, 2024-06-24:
What a total pain in the A$$.

Tried to update three boards with the 'ota:', yea just add a couple lines, it's simple. 

One worked, an ESP8266 connected to some LEDs as a test.
Two failed and cannot be connected to now, one of them was a Bluetooth Proxy which I can just plug in and reflash from scratch.  But the other is a 'ESP-01S with a relay' and no USB port so I'll need to dig out the board for it and try to remember how I set it up over a year ago.

All this might be simple if you know this stuff, but the great thing about ESPhome was you could normally muddle through and pick things up via the web and YouTube.  A big issue with this change is that a lot of the videos that newbis follow to get things working will be out of date.

ssieb, 2024-06-24:
> One thing to note - HA now has a great system for warning of upcoming deprecation to avoid situations like this. Generally a change like this would be implemented in two stages with the new [...]

There's no way to do either of these.
`platform:` can't be optional and there would be no way to use the new ota component.
There's also no way to automatically fix the config since the `ota:` section could be in a package somewhere and messing with people's configs was strongly agreed against by the devs anyway.

ssieb, 2024-06-24:
> Two failed and cannot be connected to now, one of them was a Bluetooth Proxy which I can just plug in and reflash from scratch. But the other is a 'ESP-01S with a relay' and no USB port so I'll need to dig out the board for it and try to remember how I set it up over a year ago.

Failed how?  The ota component wasn't changed.  This was just a config structure change.

gordon861, 2024-06-24:
> Failed how? The ota component wasn't changed. This was just a config structure change.

I know my info was lacking, I tried to reflash the bluetooth proxy board and that seemed to go through now.

The ESP-01S is just unreachable right now - 
```
INFO Successfully compiled program.
INFO Resolving IP address of relay-test2.local
ERROR Error resolving IP address of relay-test2.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname
```
 To be fair it's an old setup that I put together when I first started messing and could probably be improved anyways, so I think it's a rebuild.

gordon861, 2024-06-24:
And now, after sitting here for 30mins, it's just come alive, not done anything else to it. (4am need sleep).

Serebrin, 2024-06-25:
Велике вам дякую!!!! Не міг вирішити цю проблему починаючи із версії 6.0! Зараз все запрацювало! Дякую...