esphome/issues #5907: `BETA` Error compiling with 2024.6.0b2: `AttributeError: 'EStr' object has no attribute 'get'`

edwardtfn, 2024-06-14:
### The problem

I'm getting an error when trying to compile my ESP32 project with 2024.6.0b2.
This same code is part of a NSPanel custom firmware in used by many people, and I'm afraid they all will face this same issue, as I've tried with 4 different devices and got the same error (even when creating a new device, it fails to compile).
It fails both with Arduino and ESP-IDF (default for this project).

### Which version of ESPHome has the issue?

2024.6.0b2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.2

### What platform are you using?

ESP32-IDF

### Board

Sonoff NSPanel (esp32dev)

### Component causing the issue

Looks to be in the core package constructor

### Example YAML snippet

```yaml
substitutions:
  device_name: "hall-panel"
  friendly_name: "Painel do hall"
  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password

packages:
  #nspanel_package: !include packages/Blackymas/nspanel_esphome.yaml
  remote_package:
    url: https://github.com/Blackymas/NSPanel_HA_Blueprint
    ref: main
    files:
      - nspanel_esphome.yaml # Base package including TFT transfer
  #    # - advanced/esphome/nspanel_esphome_core.yaml # Core without TFT transfer
  #    # - nspanel_esphome_addon_climate_cool.yaml
  #    # - nspanel_esphome_addon_climate_heat.yaml
  #    # - nspanel_esphome_addon_climate_dual.yaml
  #    # - advanced/esphome/nspanel_esphome_advanced.yaml
  #    # - nspanel_esphome_upload_tft.yaml # Use only if you have issues with TFT transfer
  #  refresh: 300s

ota:
  - id: !extend ota_std
    platform: esphome

#esp32:
#  framework: 
#    type: arduino
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.6.0b2
INFO Reading configuration /config/esphome/troubleshooting.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1079, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1057, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 1093, in read_config
    res = load_config(command_line_substitutions)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 947, in load_config
    return _load_config(command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 937, in _load_config
    return validate_config(config, command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 774, in validate_config
    config = do_packages_pass(config)
             ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/components/packages/__init__.py", line 179, in do_packages_pass
    recursive_package = do_packages_pass(package_config)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/components/packages/__init__.py", line 179, in do_packages_pass
    recursive_package = do_packages_pass(package_config)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/components/packages/__init__.py", line 180, in do_packages_pass
    config = merge_config(recursive_package, config)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config_helpers.py", line 101, in merge_config
    return merge(full_old, full_new)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config_helpers.py", line 52, in merge
    res[k] = merge(old[k], v) if k in old else v
             ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config_helpers.py", line 52, in merge
    res[k] = merge(old[k], v) if k in old else v
             ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config_helpers.py", line 52, in merge
    res[k] = merge(old[k], v) if k in old else v
             ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config_helpers.py", line 58, in merge
    ids = {
          ^
  File "/esphome/esphome/config_helpers.py", line 61, in <dictcomp>
    if (v_id := v.get(CONF_ID)) and isinstance(v_id, str)
                ^^^^^
AttributeError: 'EStr' object has no attribute 'get'
```

ssieb, 2024-06-14:
You can't convert from a non-platform component to a platform component using `!extend`.

edwardtfn, 2024-06-14:
The issue is not related to OTA. When I copy the entire content from the remote package to the local yaml, it compiles fine with exactly same code.
Looks like the issue is related to using `packages`. I have the same issue in another totally different project.
It happens both with remote packages or local packages.

edwardtfn, 2024-06-14:
Build flags, under ESPHome, is breaking the thing.
Any build flag I try to use gets to that error.

edwardtfn, 2024-06-17:
This code looks to be changes by #6748 (https://github.com/esphome/esphome/pull/6748/files#diff-547d7b08834621f8c0b2e0a14608b313bf9a4e8ebcb3fd291e80d3c7a306a5c0R61)

edwardtfn, 2024-06-18:
I found another case where this happens, now with an extend in a select. I've removed all the `build_flags`, but then found the same issue in another place.

This is my device's yaml:
```yaml
# packages:
#   pre_build_package: !include packages/Blackymas/prebuilt/nspanel_esphome_prebuilt.yaml

########################
##### EXPERIMENTAL #####
########################

#####################################################################################################
##### NSPANEL ESPHOME created by Blackymas - https://github.com/Blackymas/NSPanel_HA_Blueprint  #####
##### ESPHOME PRE-BUILT                                                                         #####
##### PLEASE only make changes if it is necessary and also the required knowledge is available. #####
##### For normal use with the Blueprint, no changes are necessary.                              #####
#####################################################################################################
---
substitutions:
  device_name: nspanel
  wifi_ssid: nspanel
  wifi_password: NSPanel_Blueprint

  ##### DON'T CHANGE THIS ######
  NSPANEL_HA_BLUEPRINT_PREBUILT: "true"
  ##############################

packages:
#  core_package: !include packages/Blackymas/esphome/nspanel_esphome_core.yaml
#  upload_tft_package: !include packages/Blackymas/esphome/nspanel_esphome_addon_upload_tft.yaml
  remote_package:
    url: https://github.com/Blackymas/NSPanel_HA_Blueprint
    ref: dev
    refresh: 1h
    files:
      - esphome/nspanel_esphome_core.yaml
      - esphome/nspanel_esphome_addon_upload_tft.yaml

api:
  services:
    - service: firmware_update
      variables:
        url: string
      then:
        - ota.http_request.flash:
            md5_url: !lambda return fw_url->state + ".md5";
            url: !lambda return fw_url->state;
        - lambda: ESP_LOGE("prebuilt.api.service.firmware_update", "Firmware update failed!");

button:
  - name: Factory reset
    platform: factory_reset
    disabled_by_default: true
    internal: false

  - id: bt_firmware_update
    name: Firmware update
    platform: template
    on_press:
      then:
        - ota.http_request.flash:
            md5_url: !lambda return fw_url->state + ".md5";
            url: !lambda return fw_url->state;
        - lambda: ESP_LOGE("prebuilt.button.bt_firmware_update", "Firmware update failed!");

captive_portal:
  id: ap_captive_portal

dashboard_import:
  package_import_url: github://Blackymas/NSPanel_HA_Blueprint/prebuilt/nspanel_esphome_prebuilt.yaml@main
[... 127 lines cut ...]
This is the select how it is in the remote file `nspanel_esphome_addon_upload_tft`:
```yaml
select:
  - id: tft_file_model
    name: Update TFT display - Model
    platform: template
    options:
      - "Use nextion_update_url"
      - "NSPanel Blank"
      - "NSPanel EU"
      - "NSPanel US"
      - "NSPanel US Landscape"
      - "NSPanel EU (CJK languages)"
      - "NSPanel US (CJK languages)"
      - "NSPanel US Landscape (CJK languages)"
    initial_option: "NSPanel EU"
    optimistic: true
    restore_value: true
    internal: false
    entity_category: config
    disabled_by_default: false
    icon: mdi:file-sync
```

So, looks like this package creation is broken in some cases. I don't know exactly what is the issue, but I'm seeing in more than one place now.

edwardtfn, 2024-06-18:
@gtrabanco, the version on `dev` should work for you, as I've removed almost all the points of failure.

edwardtfn, 2024-06-19:
What is in common between the two cases I found is that in both I was trying to use some kind of `!extend` to a list, not a string. The `!extend` is not explicitly used for `esphome` case, but in the end it was modifying a list also.

edwardtfn, 2024-06-21:
- esphome/esphome#6952

closed 2024-06-21 by kbx81

mortenx, 2024-09-05:
Uploading: [============================================================] 100% Done...

INFO Upload took 4.02 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1014, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1001, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 523, in command_run
    return show_logs(config, args, port)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 383, in show_logs
    from esphome.components.api.client import run_logs
  File "/esphome/esphome/components/api/client.py", line 8, in <module>
    from aioesphomeapi import APIClient
  File "/usr/local/lib/python3.11/dist-packages/aioesphomeapi/__init__.py", line 7, in <module>
    from .client import APIClient
  File "/usr/local/lib/python3.11/dist-packages/aioesphomeapi/client.py", line 91, in <module>
    from .connection import APIConnection, ConnectionParams, handle_timeout
  File "aioesphomeapi/connection.py", line 24, in init aioesphomeapi.connection
  File "/usr/local/lib/python3.11/dist-packages/aioesphomeapi/_frame_helper/__init__.py", line 3, in <module>
    from .noise import APINoiseFrameHelper
  File "aioesphomeapi/_frame_helper/noise.py", line 10, in init aioesphomeapi._frame_helper.noise
  File "/usr/local/lib/python3.11/dist-packages/chacha20poly1305_reuseable/__init__.py", line 7, in <module>
    from cryptography.hazmat.primitives.ciphers.aead import (
  File "/usr/local/lib/python3.11/dist-packages/cryptography/hazmat/primitives/ciphers/__init__.py", line 11, in <module>
    from cryptography.hazmat.primitives.ciphers.base import (
  File "/usr/local/lib/python3.11/dist-packages/cryptography/hazmat/primitives/ciphers/base.py", line 10, in <module>
    from cryptography.hazmat.bindings._rust import openssl as rust_openssl
ImportError: libssl.so.1.1: cannot open shared object file: No such file or directory