esphome/issues #5908: Dallas onewire temperature sensor discovery no longer working after update to 2024.6 beta

rccoleman, 2024-06-14:
### The problem

I've been successfully using a Dallas onewire temperature sensor on my ESP8266 boards for quite a while, but after upgrading from 2024.5.x to 2024.6.x (beta) and updating the config, my sensor is no longer detected.  I've been using automatic discovery as I only have a single sensor hooked up to a single GPIO, and now I have to explicitly specify the address of the sensor (which I got from seeing what was discovered in 2024.5.5).

### Which version of ESPHome has the issue?

2024.6.0b2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.2

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

one_wire

### Example YAML snippet

```yaml
I had the following snippet working find for a long time on 2024.5.x and prior ESPHome versions:

# Temp Sensor on D02/GPIO4
dallas:
  - pin: 4
    update_interval: 60min
    id: kit_water_temp

# Individual sensors
sensor:
  - platform: dallas
    index: 0
    name: Kitchen Water Temp
    filters:
      - lambda: return x * (9.0/5.0) + 32.0;
    unit_of_measurement: "°F"
    accuracy_decimals: 2

interval:
  - interval: 2sec
    then:
      component.update: kit_water_temp
```
I modified the config to work with 2024.6.x beta and simplified it a bit, but it no longer detects my onewire sensor:
```
# Temp Sensor on D02/GPIO4
one_wire:
  - platform: gpio
    pin: 4
    
# Individual sensors
sensor:
  - platform: dallas_temp
    update_interval: 2sec
    name: Kitchen Water Temp
    filters:
      - lambda: return x * (9.0/5.0) + 32.0;
    unit_of_measurement: "°F"
    accuracy_decimals: 2
```
When I add `address: 0x480000006a42a528`, it detects the sensor and reports correct data.
```

### Anything in the logs that might be useful for us?

```txt
2024.6.0b2 not working:

INFO ESPHome 2024.6.0b2
INFO Reading configuration /config/kitchen-water-temp-new.yaml...
INFO Starting log output from 192.168.1.53 using esphome API
INFO Successfully connected to kit-water-temp @ 192.168.1.53 in 0.005s
INFO Successful handshake with kit-water-temp @ 192.168.1.53 in 0.018s
[11:19:10][I][app:100]: ESPHome version 2024.6.0b2 compiled on Jun 14 2024, 11:06:57
[11:19:10][C][wifi:599]: WiFi:
[11:19:10][C][wifi:427]:   Local MAC: 48:E7:29:6E:AD:1A
[11:19:10][C][wifi:432]:   SSID: [redacted]
[11:19:10][C][wifi:435]:   IP Address: 192.168.1.53
[11:19:10][C][wifi:438]:   BSSID: [redacted]
[11:19:10][C][wifi:440]:   Hostname: 'kit-water-temp'
[11:19:10][C][wifi:442]:   Signal strength: -71 dB ▂▄▆█
[11:19:10][C][wifi:446]:   Channel: 1
[11:19:10][C][wifi:447]:   Subnet: 255.255.255.0
[11:19:10][C][wifi:448]:   Gateway: 192.168.1.1
[11:19:10][C][wifi:449]:   DNS1: 192.168.1.1
[11:19:10][C][wifi:450]:   DNS2: 0.0.0.0
[11:19:10][C][logger:185]: Logger:
[11:19:10][C][logger:186]:   Level: DEBUG
[11:19:10][C][logger:188]:   Log Baud Rate: 115200
[11:19:10][C][logger:189]:   Hardware UART: UART0
[11:19:10][C][gpio.one_wire:016]: GPIO 1-wire bus:
[11:19:10][C][gpio.one_wire:017]:   Pin: GPIO4
[11:19:10][W][gpio.one_wire:078]:   Found no devices!
[11:19:10][C][restart.button:017]: Restart Button 'Kitchen Restart'
[11:19:10][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[11:19:10][W][dallas.temp.sensor:031]:   Unable to select an address
[11:19:10][C][mdns:115]: mDNS:
[11:19:10][C][mdns:116]:   Hostname: kit-water-temp
[11:19:10][C][esphome.ota:073]: Over-The-Air updates:
[11:19:10][C][esphome.ota:074]:   Address: kit-water-temp.local:8266
[11:19:10][C][esphome.ota:075]:   Version: 2
[11:19:10][C][safe_mode:018]: Safe Mode:
[11:19:10][C][safe_mode:019]:   Boot considered successful after 60 seconds
[11:19:10][C][safe_mode:021]:   Invoke after 10 boot attempts
[11:19:10][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[11:19:10][C][api:139]: API Server:
[11:19:10][C][api:140]:   Address: kit-water-temp.local:6053
[11:19:10][C][api:144]:   Using noise encryption: NO
```
2024.5.5 working:
```
[11:24:25][I][app:100]: ESPHome version 2024.5.5 compiled on Jun 14 2024, 11:23:40
[11:24:25][C][wifi:580]: WiFi:
[11:24:25][C][wifi:408]:   Local MAC: 48:E7:29:6E:AD:1A
[11:24:25][C][wifi:413]:   SSID: [redacted]
[11:24:25][C][wifi:416]:   IP Address: 192.168.1.53
[11:24:25][C][wifi:419]:   BSSID: [redacted]
[11:24:25][C][wifi:421]:   Hostname: 'kit-water-temp'
[11:24:25][C][wifi:423]:   Signal strength: -72 dB ▂▄▆█
[11:24:25][C][wifi:427]:   Channel: 1
[11:24:25][C][wifi:428]:   Subnet: 255.255.255.0
[11:24:25][C][wifi:429]:   Gateway: 192.168.1.1
[11:24:25][C][wifi:430]:   DNS1: 192.168.1.1
[11:24:25][C][wifi:431]:   DNS2: 0.0.0.0
[11:24:25][C][logger:185]: Logger:
[11:24:25][C][logger:186]:   Level: DEBUG
[11:24:25][C][logger:188]:   Log Baud Rate: 115200
[11:24:25][C][logger:189]:   Hardware UART: UART0
[11:24:25][C][restart.button:017]: Restart Button 'Kitchen Restart'
[11:24:25][C][dallas.sensor:075]: DallasComponent:
[11:24:25][C][dallas.sensor:076]:   Pin: GPIO4
[11:24:25][C][dallas.sensor:077]:   Update Interval: 3600.0s
[11:24:25][D][dallas.sensor:082]:   Found sensors:
[11:24:25][D][dallas.sensor:084]:     0x480000006a42a528
[11:24:25][C][dallas.sensor:089]:   Device 'Kitchen Water Temp'
[11:24:25][C][dallas.sensor:089]:     Device Class: 'temperature'
[11:24:25][C][dallas.sensor:089]:     State Class: 'measurement'
[11:24:25][C][dallas.sensor:089]:     Unit of Measurement: '°F'
[11:24:25][C][dallas.sensor:089]:     Accuracy Decimals: 2
[11:24:25][C][dallas.sensor:091]:     Index 0
[11:24:25][C][dallas.sensor:097]:     Address: 0x480000006a42a528
[11:24:25][C][dallas.sensor:098]:     Resolution: 12
[11:24:25][C][mdns:115]: mDNS:
[11:24:25][C][mdns:116]:   Hostname: kit-water-temp
[11:24:25][C][ota:096]: Over-The-Air Updates:
[11:24:25][C][ota:097]:   Address: kit-water-temp.local:8266
[11:24:25][C][ota:103]:   OTA version: 2.
[11:24:25][C][api:139]: API Server:
[11:24:25][C][api:140]:   Address: kit-water-temp.local:6053
[11:24:25][C][api:144]:   Using noise encryption: NO
```
```

SeByDocKy, 2024-06-15:
For me, with this new version, my ESP32 S3 crash after a couple of second

rccoleman, 2024-06-15:
Seems unrelated to this issue. Feel free to open a new one.

ssieb, 2024-06-15:
@rccoleman that's very strange.  Try using this:
```yaml
one_wire:
  - platform: gpio
    pin: 4
    id: dallas

button:
  - platform: template
    name: Test 1-wire bus
    then:
      - lambda: |-
          id(dallas).search();
          id(dallas).dump_config();
```
See what you get in the logs when you do that.

@SeByDocKy get serial logs of the crash, then come ask on discord or file a different issue.

rccoleman, 2024-06-15:
Hmm, looks like it worked with that?
```
[15:17:03][D][button:010]: 'Test 1-wire bus' Pressed.
[15:17:03][C][gpio.one_wire:016]: GPIO 1-wire bus:
[15:17:03][C][gpio.one_wire:017]:   Pin: GPIO4
[15:17:03][C][gpio.one_wire:080]:   Found devices:
[15:17:03][C][gpio.one_wire:082]:     0x480000006a42a528 (DS18B20)
```

I still got the "No devices found!" during startup, and the polling succeeded because I specified the address.

ssieb, 2024-06-15:
Can you get serial logs from boot?  And can you come find me on the esphome discord server, so we be more interactive?

rccoleman, 2024-06-15:
Sure

roblamoreaux, 2024-06-17:
I noticed this as well, but was able to go back a version to find the device IDs to put in for my devices. 
After adding the device IDs for the three DS18x20s I have it listed the ids during One_wire init.

It would be nice if the index: x option worked still or there was an easier way to assign which device is which after boot so we don't have to compile, run, find ids, update code, compile and run again.

Rob

I did save logs if needed, but they are on another computer

rccoleman, 2024-06-17:
@ssieb provided a patch that a patch on Discord that just retries discovery 5 times and it works for me. Obviously just a workaround, though.

https://discord.com/channels/429907082951524364/866427615564922920/1251665034191310879

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: onewire
    components: [ gpio ]
    refresh: 1min
```

djexplode6, 2024-06-19:
Same here. Had Dallas temps working fine on.5. Changed config to use one-wire for .6. One-wire reports nothing on the bus. I have the devices defined with their ids and get the following errors:  

[17:53:18][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[17:53:18][W][dallas.temp.sensor:074]: 'Refridgerator' - reading scratch pad failed bus reset
[17:53:18][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed
[17:53:18][D][sensor:094]: 'Refridgerator': Sending state nan °C with 1 decimals of accuracy
[17:53:34][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[17:53:34][W][dallas.temp.sensor:074]: 'Freezer' - reading scratch pad failed bus reset
[17:53:34][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed
[17:53:34][D][sensor:094]: 'Freezer': Sending state nan °C with 1 decimals of accuracy

Related Config:

one_wire:
  - platform: gpio
     pin: GPIO13

sensor:
  - platform: dallas_temp
     address: (long device address)
     name: “Refrigerator Temp”
   - platform: dallas_temp
     address: (long device address)
     name: “Freezer Temp”

kbrgmn, 2024-06-20:
Indeed, the sensor which worked on 2024.5 is not being reported anymore with 2024.6.0, even when updated for the breaking changes according to the changelog:

```yaml
one_wire:
  - platform: gpio
    pin: 4

sensor:
  - platform: dallas_temp
    name: "Temperature"
```

rccoleman, 2024-06-20:
I added a logic analyzer trace here for one of my devices: https://discord.com/channels/429907082951524364/1251665034191310879/1252826630561464452

TheRealFalseReality, 2024-06-20:
Would support for using `index:` come back? It really helps for not having to find the addresses, especially if using just 1 Dallas probe.

rccoleman, 2024-06-20:
if you're just using one, a functional discovery mechanism would work without `index`.  Adding this from above works around the discovery issue and I no longer need to specify the address:

> @ssieb provided a patch that a patch on Discord that just retries discovery 5 times and it works for me. Obviously just a workaround, though. [...]

Otherwise, it sounded like `index` wasn't a very reliable way to identify individual devices on a bus that contains more than one.

TheRealFalseReality, 2024-06-20:
> if you're just using one, a functional discovery mechanism would work without `index`. Adding this from above works around the discovery issue and I no longer need to specify the address: [...]

I'll look into. I'm using a total of 6 ESP32 devices, 2 of which are using 2 dallas sensors (to monitor 2 adjacent aquariums). I actually use `index: 0` and another sensors with `index: 1` . They have always been discovered and to my surprise, they have never switched, as I feared they may have. Really convenient way to have dallas_sensors "plug-and-play". I was able to just have the code set-up for 2 sensors for each device, and whenever I was to add another one, all I had to add was wire it and restart the esp32 device...

Here's an example and the source code if you wanted to take a look:

```
sensor:
  # Temperature Probe 1
  - platform: dallas
    id: temp_1
    icon: mdi:thermometer-lines
    index: 0
    name: "Temperature"
    state_class: "measurement"
    device_class: "temperature"
    filters:
      - calibrate_linear:
          method: least_squares
          datapoints:
            - 0.0 -> ${cal_0}
            - 100.0 -> ${cal_100}
    on_value: 
      then:
        - component.update: temp_range

  # Temperature Probe 2
  - platform: dallas
    id: temp_2
    icon: mdi:thermometer-lines
    index: 1
    name: "Temperature 2"
    state_class: "measurement"
    device_class: "temperature"
    filters:
      - calibrate_linear:
          method: least_squares
          datapoints:
            - 0.0 -> ${cal_0_2}
            - 100.0 -> ${cal_100_2}
    disabled_by_default: true
    on_value: 
      then:
        - component.update: temp_range_2
```

https://github.com/TheRealFalseReality/aquapi/blob/main/common/temperature_dallas.yaml

brostudiodev, 2024-06-20:
I have tried the workaroud 

 Couldn't find any component that can be used for 'one_wire::OneWireBus'. Are you missing a hub declaration?.

> if you're just using one, a functional discovery mechanism would work without `index`. Adding this from above works around the discovery issue and I no longer need to specify the address: [...]
I have tried this one but it is not installing - error: Couldn't find any component that can be used for 'one_wire::OneWireBus'. Are you missing a hub declaration?.

And yes my dallas temp sensors failed due 2024.6 update :(

Makin-Things, 2024-06-20:
I just updated and have the same problem. I already had the correct address in the config yet it still cannot be found. Now I am without a working heater when it is frigging cold. So how do I get my heater working again?

@jesserockz I seriously don't understand how a breaking change with open issues against it could possibly be merged and released.

SeByDocKy, 2024-06-20:
I can also confirm that that the new one-wire integration let crash my ESP32 S3..... 
I tested also the n-1 commit and was working fine until the one-wire commit. 

Actually this ESP32 S3 is in production in my solar station. I have to make an access to the serial port (in the current enclosing box) to output serial logs to help to debug...

schinddo, 2024-06-20:
I had also similar problems with the 2024.6.1 release and I urgently need a working one-wire component so I downgraded to 2024.5.5 release and its working fine now.

brostudiodev, 2024-06-20:
> I just updated and have the same problem. I already had the correct address in the config yet it still cannot be found. Now I am without a working heater when it is frigging cold. So how do I get my [...]

For this case it is better to go back to previous version (use Settings/System/Restore points (or similar -> I have different lang than EN in my instance) and revert back the change. It is better to wait once HA team fixed this...

Makin-Things, 2024-06-20:
> For this case it is better to go back to previous version (use Settings/System/Restore points (or similar -> I have different lang than EN in my instance) and revert back the change. It is better to [...]

It was easier to go back to dumb heating.

SeByDocKy, 2024-06-20:
I would also vote for a revert to the dallas component until the one-wire is fully tested and debugged....

mathieucarbou, 2024-06-20:
> ```yaml
> https://github.com/ssieb/esphome
> ```

I've looked at the diff (https://github.com/ssieb/esphome/commit/0a704810a88a502232c0129ce5fdb60d9e2f695a#diff-fe5aa506e8e456fd0236a4fd142f41d72c8c358d54007854f5738e40e805b41b) and this is exactly what I also need to do with the implementation for DS18 on ESP32 I am using, which relies on the RMT Peripheral and don't have any delay calls. Scanning the devices once right after a boot is a hit and miss and I need to redo the scan.

I've seen that the previous impl of 1-wire in ESPHome was also relying heavily on delays, in the I/O methods plus one of 500us put just in the setup before scanning. Now, the search is happening right aways:

before:

```
// clear bus with 480µs high, otherwise initial reset in search_vec() fails
  pin_->pin_mode(gpio::FLAG_INPUT | gpio::FLAG_PULLUP);
  delayMicroseconds(480);

  one_wire_ = new ESPOneWire(pin_);  // NOLINT(cppcoreguidelines-owning-memory)

  std::vector<uint64_t> raw_sensors;
  raw_sensors = this->one_wire_->search_vec();
```

now:

```
void GPIOOneWireBus::setup() {
  ESP_LOGCONFIG(TAG, "Setting up 1-wire bus...");
  this->search();
}
```

Maybe this delay was helping this hit and miss ?

saschaludwig, 2024-06-20:
> Indeed, the sensor which worked on 2024.5 is not being reported anymore with 2024.6.0, even when updated for the breaking changes according to the changelog:
> 
> ```yaml
> one_wire:
>   - platform: gpio
>     pin: 4
> 
> sensor:
>   - platform: dallas_temp
>     name: "Temperature"
> ```

same here, worked on 2024.5.5
broken on 2024.6.1

gjgeary, 2024-06-20:
I am seeing this issue with DS18B20 on ESP32.  I have another ESP8266 with 2 DS18B20 sensors that is working just fine.  So far.  Fingers crossed.

roblamoreaux, 2024-06-20:
I am going to try to get better logs tonight or this weekend on my 2 DS1820 sensor board since it only has sensors and a display unlike the other boards. 
Is there anything I can set to get better logs?  
     When I set very_verbose there were a lot of other messages but not enough about the 1-wire stuff to be useful.

Of note about using Index. Because of the way discovery is done as long as all devices are up the index will always be same address order unless a part is replaced. So while I find Index useful I can see it is not the greatest solution. I'd love it if the index could be changed as a configuration parameter so it doesn't need a recompile to change which device is where.

Links2004, 2024-06-20:
can confirm this being a problem with ESP32 and ESP32-C3 
e.g.: https://github.com/esphome/issues/issues/5936

any more debugging info that can I provide for my setup, or is there good info already to fix it?

SeByDocKy, 2024-06-20:
and ESP32-S3 ...

gabest11, 2024-06-20:
I also have a problem with the C3 and this temperature sensor. Tested all available gpio pins and found that half of them do not work.

GPIO0 OK
GPIO1 OK
GPIO3 OK
GPIO4 FAIL
GPIO5 FAIL
GPIO6 FAIL
GPIO7 FAIL
GPIO10 OK

[This mod did not help](https://github.com/esphome/issues/issues/5908#issuecomment-2180690029)

peregrines2, 2024-06-20:
Same here :(
ESP8266:
worked on 2024.5.5
broken on 2024.6.1

ConvergintMarkMartens, 2024-06-20:
Confirmed Broken.  Same Here..
ESP32
GPIO13
Worked on 2024.5.5
Broken on 2024.6.1

gabest11, 2024-06-21:
Could someone test this?

```
external_components:
  - source: github://gabest11/esphome@one_wire
    components: [ gpio ]
```

joe-cole1, 2024-06-21:
Same issue here. I have two DS18B20 sensors on an ESP32. The one on pin 22 works. Pin 14 does not work.

thargy, 2024-06-21:
I have 9  DS18B20 sensors on an ESP32 on pin 14. I've been using the sensors for over a year without issue before the release of 2024.6. It worked for several hours before the above issue appeared, and I lost all temperature readings. This caused my hot water tank to overheat in the middle of the night and woke me up at 4AM!

Even though I have ESPHome installed via HAOS, I also have a local installation on my dev machine, so it was relatively quick for me to roll back by reverting my local version and pushing the firmware from my PC rather than the HAOS add-on.  I have huge sympathy for less techie users trying to figure out how to do a rollback via HAOS itself...

On the plus side, this has reminded me to check for `NaN`s in the temperature readings and act appropriately (i.e. turn off immersion heaters when I can't read the temperatures!)

Mistakes do happen, particularly in open-source software run by volunteers. Further, the suggestion of splitting the `dallas` component up to allow for more 1-wire components in future is sensible. My goal is not to dump on @ssieb or @jesserockz, who I have no doubt are as disappointed as any of us (and probably more so). No one wants to release broken code.

However, it is abundantly clear that this issue did not only separate out the 1-wire logic. Instead, it changed a lot of the existing logic and timings for the existing temperature sensor. As a result, the description of the breaking change was far from complete - and is a classic example of doing too much in one step. Indeed, it also dropped `index` and moved `update_interval`, neither of which surfaced clearly in the release notes, making it hard for users to understand what has broken and why. The `update_interval` change is buried even further.

ESPHome is a hugely popular project, and DS18B20 sensors are a popular sensor and common use case. Many users will use ESPHome via HAOS, and rolling back an addon version is far from trivial.

At the very least, I submit that the existing 'Dallas' component should have been left in place for a period with a compile warning or similar to indicate it was deprecated. This would have allowed users to migrate to the new 1-wire system in their own time and to easily roll back when they identified this problem.

Can I respectfully encourage @ssieb to focus on creating a PR that allows the legacy `dallas` component to coexist with the new `one_wire` & `dallas_temp` components so that it can be released ASAP and restore existing functionality? That way, the numerous timing issues and bugs can be addressed without prolonging the disruption.

Similarly, could I encourage @jesserockz to consider requiring deprecation periods over breaking changes when accepting pull requests? I appreciate that it can sometimes cause more work, but it is a reasonable balance to protect less technical users and one adopted by the HA devs in general.

I have tried very hard to write this in a constructive manner, and I hope it reads as such.

schinddo, 2024-06-21:
> I've seen that the previous impl of 1-wire in ESPHome was also relying heavily on delays in the I/O methods plus one of 500us just in the setup before scanning. Now, the search is happening right away:
> 
> before:
> 
> ```
> // clear bus with 480µs high, otherwise initial reset in search_vec() fails
>   pin_->pin_mode(gpio::FLAG_INPUT | gpio::FLAG_PULLUP);
>   delayMicroseconds(480);
> 
>   one_wire_ = new ESPOneWire(pin_);  // NOLINT(cppcoreguidelines-owning-memory)
> 
>   std::vector<uint64_t> raw_sensors;
>   raw_sensors = this->one_wire_->search_vec();
> ```

Maybe the DS18B20 sensor needs to be reset before you can use it, unlike other sensors.

In the DS18B20 manual, in chapter "Hardware Configuration", there is a delay or 480 µs mentioned for reset:

> If the bus is held low for more than 480μs, all components on the bus will be reset.

ssieb, 2024-06-21:
Yes, I missed the setup call and it seems that it only matters in some cases.  This change was tested by several people on various platforms with no issues found.  If more people would test beta releases, issues like this would be much more likely to be found.  Test out this branch that has several fixes and works for several people that have been having issues:
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: onewire
    components: [ gpio ]
    refresh: 1min
```

joos81, 2024-06-21:
> Yes, I missed the setup call and it seems that it only matters in some cases. This change was tested by several people on various platforms with no issues found (including people that couldn't use the previous version of the component). If more people would test beta releases, issues like this would be much more likely to be found. Test out this branch that has several fixes and works for several people that have been having issues:
> 
> ```yaml
> external_components:
>   - source:
>       type: git
>       url: https://github.com/ssieb/esphome
>       ref: onewire
>     components: [ gpio ]
>     refresh: 1min
> ```
> 
> Let me know if you see anything other than 0 retries in the config logging.

This resolves the issue for me on two ESP32 devices with Dallas sensors on GPIO15.
I don't have any errors or warnings in the logs anymore and I get proper temperature readings.

kaufe, 2024-06-21:
> Yes, I missed the setup call and it seems that it only matters in some cases. This change was tested by several people on various platforms with no issues found (including people that couldn't use [...]

Same for me, resolved also the Issue on 2 ESP32 devices with Dallas sensors ON GPIO14. Thanks @ssieb 
Don`t see any Errors or warning in the logs.

peregrines2, 2024-06-21:
> Yes, I missed the setup call and it seems that it only matters in some cases. This change was tested by several people on various platforms with no issues found (including people that couldn't use [...]

Did not work for me on 2 ESP8266 GPIO2

ssieb, 2024-06-21:
> Did not work for me on 2 ESP8266 GPIO2

Need more info than that.   And GPIO2 is not a good pin to use.  Did it work on that before?

peregrines2, 2024-06-21:
> > Did not work for me on 2 ESP8266 GPIO2
> 
> Need more info than that. And GPIO2 is not a good pin to use. Did it work on that before?

It did work perfectly before (about 6 months now).
Updating esphome broke it yesterday.

ssieb, 2024-06-21:
And you still didn't provide any details...

Pulpet90, 2024-06-21:
After update to 2024.6.3, I can see my sensors finally

```
[09:58:08][C][gpio.one_wire:016]: GPIO 1-wire bus:
[09:58:08][C][gpio.one_wire:017]:   Pin: GPIO5
[09:58:08][C][gpio.one_wire:080]:   Found devices:
[09:58:08][C][gpio.one_wire:082]:     0x0280000003678628 (DS18B20)
[09:58:08][C][gpio.one_wire:082]:     0x9c800000036f0b28 (DS18B20)
[09:58:08][C][gpio.one_wire:082]:     0x85800000039a9f28 (DS18B20)
```

Pulpet90, 2024-06-21:
[10:07:16][C][gpio.one_wire:082]:     0x0280000003678628 (DS18B20)
[10:07:16][C][gpio.one_wire:082]:     0x9c800000036f0b28 (DS18B20)
[10:07:16][C][gpio.one_wire:082]:     0x85800000039a9f28 (DS18B20)
[10:07:16][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[10:07:16][C][dallas.temp.sensor:034]:   Address: 0x0280000003678628 (DS18B20)
[10:07:16][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[10:07:16][C][dallas.temp.sensor:036]:   Update Interval: 3.0s
[10:07:16][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[10:07:16][C][dallas.temp.sensor:034]:   Address: 0x9c800000036f0b28 (DS18B20)
[10:07:16][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[10:07:16][C][dallas.temp.sensor:036]:   Update Interval: 3.0s
[10:07:16][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[10:07:16][C][dallas.temp.sensor:034]:   Address: 0x85800000039a9f28 (DS18B20)
[10:07:16][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[10:07:16][C][dallas.temp.sensor:036]:   Update Interval: 3.0s
[10:07:16][C][captive_portal:088]: Captive Portal:
[10:07:16][C][mdns:115]: mDNS:
[10:07:16][C][mdns:116]:   Hostname: temp-sensor-x3
[10:07:16][C][esphome.ota:073]: Over-The-Air updates:
[10:07:16][C][esphome.ota:074]:   Address: temp-sensor-x3.local:8266
[10:07:16][C][esphome.ota:075]:   Version: 2
[10:07:16][C][esphome.ota:078]:   Password configured
[10:07:16][C][safe_mode:018]: Safe Mode:
[10:07:16][C][safe_mode:019]:   Boot considered successful after 60 seconds
[10:07:16][C][safe_mode:021]:   Invoke after 10 boot attempts
[10:07:16][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[10:07:16][C][api:139]: API Server:
[10:07:16][C][api:140]:   Address: temp-sensor-x3.local:6053
[10:07:16][C][api:142]:   Using noise encryption: YES
[10:07:17][D][dallas.temp.sensor:054]: 'temperature2': Got Temperature=25.1°C
[10:07:17][D][sensor:093]: 'temperature2': Sending state 25.12500 °C with 1 decimals of accuracy
[10:07:17][D][dallas.temp.sensor:054]: 'temperature1': Got Temperature=25.8°C
[10:07:17][D][sensor:093]: 'temperature1': Sending state 25.81250 °C with 1 decimals of accuracy
[10:07:18][D][dallas.temp.sensor:054]: 'temperature3': Got Temperature=25.4°C
[10:07:18][D][sensor:093]: 'temperature3': Sending state 25.37500 °C with 1 decimals of accuracy
[10:07:20][D][dallas.temp.sensor:054]: 'temperature2': Got Temperature=25.1°C
[10:07:20][D][sensor:093]: 'temperature2': Sending state 25.12500 °C with 1 decimals of accuracy
[10:07:20][D][dallas.temp.sensor:054]: 'temperature1': Got Temperature=25.8°C
[10:07:20][D][sensor:093]: 'temperature1': Sending state 25.81250 °C with 1 decimals of accuracy
[10:07:21][D][dallas.temp.sensor:054]: 'temperature3': Got Temperature=25.4°C
[10:07:21][D][sensor:093]: 'temperature3': Sending state 25.37500 °C with 1 decimals of accuracy
[10:07:23][D][dallas.temp.sensor:054]: 'temperature2': Got Temperature=25.1°C
[10:07:23][D][sensor:093]: 'temperature2': Sending state 25.12500 °C with 1 decimals of accuracy
[10:07:23][D][dallas.temp.sensor:054]: 'temperature1': Got Temperature=25.8°C
[10:07:23][D][sensor:093]: 'temperature1': Sending state 25.75000 °C with 1 decimals of accuracy
[10:07:24][D][dallas.temp.sensor:054]: 'temperature3': Got Temperature=25.4°C
[10:07:24][D][sensor:093]: 'temperature3': Sending state 25.43750 °C with 1 decimals of accuracy
[10:07:26][D][dallas.temp.sensor:054]: 'temperature2': Got Temperature=25.1°C
[10:07:26][D][sensor:093]: 'temperature2': Sending state 25.12500 °C with 1 decimals of accuracy
[10:07:26][D][dallas.temp.sensor:054]: 'temperature1': Got Temperature=25.8°C

Makin-Things, 2024-06-21:
> Yes, I missed the setup call and it seems that it only matters in some cases. This change was tested by several people on various platforms with no issues found (including people that couldn't use [...]

This has worked for me on an ESP32 GPIO15. I will politely point out that this issue was opened against the BETA version a number of days before the release.

ssieb, 2024-06-21:
> I will politely point out that this issue was opened against the BETA version a number of days before the release.

Yes, but **this issue** was only discovery not working for one person and even then, the sensors were still reading correctly.  And there was a workaround to retry the discovery.

gabest11, 2024-06-21:
@ssieb this->retries_++ < 5 is always evaluated when this->devices_.empty() is true. In other words, the incrementing does not stop at 5, it just returns false (until it rolls over. then it searches again).

crossbreed3, 2024-06-21:
Hi can anyone please share a snippet of the yaml file, im not coming right in the configs. only got my first sensor yesterday, with my luck it was the day the internet broke. i dont know how to implement the external component and if any changes are needed from this to the rest of the code. 

Using ESP32 Wroom
Dallas Sensor on GPIO14

1) where to place "external component" line
2) one_wire configs
3)sensor configs. 

Thanks

gabest11, 2024-06-21:
external_components:
  - source: github://ssieb/esphome@onewire
#  - source: github://gabest11/esphome@one_wire # with fixed retry counting
    components: [ gpio ]
    refresh: 1min

one_wire:
  - platform: gpio
    pin: GPIO14
    id: one_wire_on_gpio14

sensor:
  - platform: dallas_temp
    one_wire_id: one_wire_on_gpio14
    name: "Temperature"
#    address: 0x... # only needed if you have more than one

brostudiodev, 2024-06-21:
I hope this works great... 
The question is... Should we wait until official fix or do a workaround? My experience tells me to wait and keep v2024.5.5 but it is just my preference...  

> ```
> external_components:
>   - source: github://ssieb/esphome@onewire
> #  - source: github://gabest11/esphome@one_wire # with fixed retry counting
>     components: [ gpio ]
>     refresh: 1min [...]

I hope this works great... 
The question is... Should we wait until official fix or do a workaround? My experience tells me to wait and keep v2024.5.5 but it is just my preference...

thargy, 2024-06-21:
> ```yaml
>   - source:
>       type: git
>       url: https://github.com/ssieb/esphome
>       ref: onewire
>     components: [ gpio ]
>     refresh: 1min
> ```

I have reverted to `2024.6.1` with the above patch, and the device has found the 9 sensors - however, it did that before but failed after 3-4 hours, so I will have to monitor. Also, I am concerned about the implementation of the retry logic, and have commented on the [git commit](https://github.com/ssieb/esphome/commit/742165dc2a2e0f5d090f95a6726bd29a394adda9) (see also [this comment above](https://github.com/esphome/issues/issues/5908#issuecomment-2182421051) by @gabest11).

> Can I respectfully encourage @ssieb to focus on creating a PR that allows the legacy dallas component to coexist with the new one_wire & dallas_temp components so that it can be released ASAP and [...]

Any view on this?

crossbreed3, 2024-06-21:
> ```
> external_components:
>   - source: github://ssieb/esphome@onewire
> #  - source: github://gabest11/esphome@one_wire # with fixed retry counting
>     components: [ gpio ]
>     refresh: 1min [...]

thanks you are "GA best" i do see its changed in the log, it picks up the sensor now as dallas but now says unable to select an address. any idea? i normally would just surf the internet until i come right, but seeing this is a new problem i cant find any info on this. 

Thanks

[13:28:58][C][logger:185]: Logger:
[13:28:58][C][logger:186]:   Level: DEBUG
[13:28:58][C][logger:188]:   Log Baud Rate: 115200
[13:28:58][C][logger:189]:   Hardware UART: UART0
[13:28:58][C][gpio.one_wire:027]: GPIO 1-wire bus:
[13:28:58][C][gpio.one_wire:028]:   298378 retries required
[13:28:58][C][gpio.one_wire:029]:   Pin: GPIO14
[13:28:58][W][gpio.one_wire:078]:   Found no devices!
[13:28:58][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[13:28:58][W][dallas.temp.sensor:031]:   Unable to select an address
[13:28:58][C][captive_portal:088]: Captive Portal:
[13:28:58][C][mdns:115]: mDNS:
[13:28:58][C][mdns:116]:   Hostname: test
[13:28:58][C][esphome.ota:073]: Over-The-Air updates:
[13:28:58][C][esphome.ota:074]:   Address: test.local:3232
[13:28:58][C][esphome.ota:075]:   Version: 2
[13:28:58][C][esphome.ota:078]:   Password configured
[13:28:58][C][safe_mode:018]: Safe Mode:
[13:28:58][C][safe_mode:020]:   Boot considered successful after 60 seconds
[13:28:58][C][safe_mode:021]:   Invoke after 10 boot attempts
[13:28:58][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[13:28:58][C][api:139]: API Server:
[13:28:58][C][api:140]:   Address: test.local:6053
[13:28:58][C][api:142]:   Using noise encryption: YES
[13:29:45][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[13:29:45][D][esp32.preferences:114]: Saving 1 preferences to flash...
[13:29:45][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed

gabest11, 2024-06-21:
Which board do you have? Also, try other pins and my branch. "298378 retries required" is from ssiebs, it runs away quickly and might rescan, I don't really know how fast.

roblamoreaux, 2024-06-21:
@ssieb 
The latest patch seems to do a search in all cases I've tried. 

I ran various configs last night to determine when it did not search and found that if I had correct addresses in my sensors it would search and if I did not it would not search. If I had no sensors it would search. The latest seems to search in all cases.

Having worked with Dallas 1-wire close to 30 years, I understand it can be a challenge. The only interfaces I have found to be truly reliable are the dedicate 1-wire interfaces, especially with devices that need more current for writing to them.
I wish I had had more time to debug during the Beta, but this is why I keep the non-beta esphome install a version behind the beta, and I don't install on critical ESPs until I have installed on non-critical ones. Heck even the one I consider critical is just the hot tub, and that isn't too important.

Since I am very interested in adding additional 1-wire devices partly to get my original Dallas Semi Weather Station off the Raspberry Pi and onto an ESP32, please let me know if you need any testing before things to go beta and I will try to test them. I have all sorts of 1-wire devices a number of which are no longer sold.

Rob

roblamoreaux, 2024-06-21:
@crossbreed3 
Your log shows the search is not finding any 1-wire devices ( I think it is saying it tried searching 298378 times). 
Do you have a pull up (1k seems best) on the IO pin? 

The IO pins do not always drive enough current to provide power to the devices so often that is a problem. I also usually connect power to the supply pin of the devices so they aren't relying on the parasitic power alone.

I found with the latest code it is best to add the one-wire component with no sensor the run and look at the logs to get the addresses on the wire. Once the addresses are in the sensor config it seems more reliable.

mathieucarbou, 2024-06-21:
> I found with the latest code it is best to add the one-wire component with no sensor the run and look at the logs to get the addresses on the wire. Once the addresses are in the sensor config it seems more reliable.

Which is exactly what the previous ESPHome doc was advising to doc, besides not using indexes.

gabest11, 2024-06-21:
> I have switched to your branch and getting the 5 retries. ive tried pin 4, 8, 15, 16, 21 all with and without the 4.7k resistor, still says unable to select an address.

8 is connected to the flash memory, but the rest works no problem. Funny thing, I put all pins into the yaml and just hotplugged the jumper cable between the pins. Even if nothing gets detected during setup, it will read the temperature if you give the exact address.

Links2004, 2024-06-21:
Have the changes running now for multiple hours so and its working as expected. 
0 retries at startup and no read errors or wired values, looks as good as with 2024.5.5.
So I assume that in my cases it where the missing reset that where the problem.

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: onewire
    components: [ gpio ]
    refresh: 1min
```

Tested Pins:

| ESP32 | ESP32-S2 | ESP32-C3 |
| ------- | ---------- | ----------- |
| 13, 14  | 9             | 5               |

all test with power (no parasitic)

crossbreed3, 2024-06-21:
thanks for the replies

but how do i get the address?

no i dont have one with screw terminals.

ConvergintMarkMartens, 2024-06-21:
Just added the repo

```
# ESPHome 2024.06.1

  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: onewire
    components: [ gpio ]
    refresh: 1min
	
one_wire:
  - platform: gpio
    pin: 13

# Individual sensors
sensor:
  - platform: dallas_temp
    address: 0x4c000004ce160028
    name: "Cold Water Supply"
    accuracy_decimals: 1
#    update_interval: 60s
~ Truncated ~
```

Looks great so far!

I will monitor for a few hours and let you know, but that looks like the fix.  (At least for me)

Thank you

gabest11, 2024-06-21:
@crossbreed3 The address is logged during the setup. When it fails you see that "Found no devices!" message.

[136 more replies]

closed 2024-06-22 by jesserockz