esphome/issues #5910: Waveshare 7.5" display: Wifi stops working when display is enabled

eikowagenknecht, 2024-06-16:
### The problem

I bought a Waveshare 7.5" display set (https://www.amazon.de/dp/B0CN96WHRC?psc=1&ref=ppx_yo2ov_dt_b_product_details):
- 7.5" display
    - Label on the back: "075BN-T7" "2024031802"
    - Label on the cable: "FPC-C001" "21.08.30" "HB"
- Waveshare e-Paper Adapter
- Waveshare [e-Paper ESP32 Driver Board

I tried setting it up with ESPHome for quite some time now. Unfortunately, I can't get it to work properly.

What works:
- The display refreshes every 60s (as set) and shows the text set in the config just fine. 

What doesn't work:
- As soon as I add the spi/display blocks in the YAML, I can not reach the device via wifi any more. Constantly pinging it shows that the connection is sometimes available, sometimes not (see logs below), but I can never reach the web interface (http://IP). As soon as I refresh with spi/display blocks commented out, it's working fine again.

I tried:
- Setting busy_pin without inverted: way worse, text broken in half sometimes.
- Setting update_interval to never makes it not update at all (as it should), but the wifi is still broken
- I tried models "7.50in, 7.50inV2, 7.50inV2alt, 7.50in-bV2". Out of these, only 7.50inV2 and 7.50inV2alt display the text on the screen, but all produce the wifi problems.

How can I debug this further or even fix it?

### Which version of ESPHome has the issue?

2024.5.5

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

Waveshare e-Paper ESP32 Driver Board

### Component causing the issue

display

### Example YAML snippet

```yaml
substitutions:
  name: "epaper-display-buero"
  friendly_name: KG.Büro ePaper Display

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: ""

  manual_ip:
    static_ip: 192.168.40.95
    gateway: 192.168.40.1
    subnet: 255.255.255.0
    dns1: 192.168.40.1

  # Fallback wifi access point in case wifi connection fails
  ap:
    ssid: "ePaper Fallback Hotspot"
    password: "password"

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

# Google Fonts
font:
 - file: "gfonts://Inter@900"
   id: medium
   size: 60

# Pins for Waveshare ePaper ESP Board
spi:
 # SPI Clock Pin
 clk_pin: GPIO13
 # SPI Master Out Slave In Pin
 mosi_pin: GPIO14

# Pins for Waveshare ePaper ESP Board
display:
 - platform: waveshare_epaper
   id: screen
   # Chip Select Pin
   cs_pin: GPIO15
   # Data/Command Pin
   dc_pin: GPIO27
   # Busy Pin
   busy_pin:
     number: GPIO25
     inverted: true
   # Reset Pin
   reset_pin: GPIO26
   reset_duration: 2ms
   model: 7.50inV2
   update_interval: 60s
   rotation: 90°
   lambda: |-
     it.print(40, 10, id(medium), "Hello World!");
```

### Anything in the logs that might be useful for us?

```txt
ESP Logs (attached with USB to PC because wifi doesn't work)

[10:06:08][I][waveshare_epaper:2464]: Power on the display and hat
[10:06:14][W][component:237]: Component display took a long time for an operation (6721 ms).
[10:06:14][W][component:238]: Components should block for at most 30 ms.

(this repeats on every display refresh)

Ping example:

Reply from 192.168.40.95: bytes=32 time=39ms TTL=255
Reply from 192.168.40.95: bytes=32 time=52ms TTL=255
Reply from 192.168.40.95: bytes=32 time=64ms TTL=255
Reply from 192.168.40.95: bytes=32 time=76ms TTL=255
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.40.110: Destination host unreachable.
Request timed out.
Request timed out.
Reply from 192.168.40.110: Destination host unreachable.
Request timed out.
Reply from 192.168.40.110: Destination host unreachable.
Reply from 192.168.40.95: bytes=32 time=1820ms TTL=255
Reply from 192.168.40.95: bytes=32 time=5ms TTL=255
Reply from 192.168.40.95: bytes=32 time=77ms TTL=255
Request timed out.
Reply from 192.168.40.95: bytes=32 time=42ms TTL=255
Reply from 192.168.40.95: bytes=32 time=356ms TTL=255
Reply from 192.168.40.95: bytes=32 time=369ms TTL=255
Request timed out.
Request timed out.
Reply from 192.168.40.95: bytes=32 time=47ms TTL=255
Reply from 192.168.40.95: bytes=32 time=62ms TTL=255
Reply from 192.168.40.95: bytes=32 time=78ms TTL=255
Reply from 192.168.40.95: bytes=32 time=85ms TTL=255
```

guevara777, 2024-07-01:
i have exactly the same problem. Did you make any progress on your problem?

eikowagenknecht, 2024-07-02:
Unfortunately not. The display project has been set aside for that reason (for now). Please let me know if you manage to find a solution.

miT-nib-hcI, 2024-08-03:
Unfortunatly the bug seamingly still presists as my project is at the same point as yours, as i get the same error with the only solution i found being to not use the display. Which defeats the Purpose of my Projekt.

sylvainzabe, 2024-08-09:
Same problem here :-/

rybus, 2024-09-23:
Same issue here, with 7.5 + Waveshare ESP32 Board. I used 7.50inV2alt and now the connection is never lost.

eikowagenknecht, 2024-09-24:
>  I used 7.50inV2alt and now the connection is never lost.

Can you maybe post your config file? Which physical switch position do you use? A or B?

Schuchie, 2024-10-03:
Got the same problem with the same setup.

Schuchie, 2024-10-03:
@eikowagenknecht I think I found the problem or at least the fix. The following changes worked for me. 
I got the following wifi config:
```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: True
  power_save_mode: NONE
  manual_ip: 
    static_ip: "192.168.10.98"
    subnet: "255.255.255.0"
    gateway: "192.168.10.1"
    dns1: "192.168.10.1"
```
And I think my problem was solved by disabling/comment out the following section (which came from the default config):
```
# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
#esp32_improv:
#  authorizer: none
```
This is also required to comment out, so that we can use "power_save_mode: NONE". All in all I think deactivating the bluetooth LE was the trick.

Cheers

eikowagenknecht, 2024-10-22:
Thank you so much! Commenting out the Bluetooth stuff and changing the WiFi settings made it way more reliable:

```
Reply from 192.168.40.95: bytes=32 time=3ms TTL=255
Reply from 192.168.40.95: bytes=32 time=3ms TTL=255
Reply from 192.168.40.95: bytes=32 time=3ms TTL=255
Reply from 192.168.40.95: bytes=32 time=3ms TTL=255
Reply from 192.168.40.95: bytes=32 time=3ms TTL=255
Reply from 192.168.40.95: bytes=32 time=22ms TTL=255
Reply from 192.168.40.95: bytes=32 time=3ms TTL=255
Reply from 192.168.40.95: bytes=32 time=3ms TTL=255
Reply from 192.168.40.95: bytes=32 time=3ms TTL=255
Reply from 192.168.40.95: bytes=32 time=3ms TTL=255
Reply from 192.168.40.95: bytes=32 time=3ms TTL=255
Reply from 192.168.40.95: bytes=32 time=10ms TTL=255
Reply from 192.168.40.95: bytes=32 time=3ms TTL=255
Reply from 192.168.40.95: bytes=32 time=3ms TTL=255
Reply from 192.168.40.95: bytes=32 time=3ms TTL=255
Reply from 192.168.40.95: bytes=32 time=3ms TTL=255
Reply from 192.168.40.95: bytes=32 time=3ms TTL=255
Reply from 192.168.40.95: bytes=32 time=3ms TTL=255
```

I still get the "long time" warnings, though:

```
[12:17:23][W][component:237]: Component display took a long time for an operation (7619 ms).
[12:17:23][W][component:238]: Components should block for at most 30 ms.
```

But besides logging these, it works fine as far as I can tell. Here's the complete config for reference:

The physical switches are set to "A" and "ON".

```yaml
substitutions:
  name: "epaper-display-buero"
  friendly_name: KG.Büro ePaper Display

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
  platform: esphome

# Allow provisioning Wi-Fi via serial
# improv_serial:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: True
  power_save_mode: NONE
  domain: ""

  manual_ip:
    static_ip: 192.168.1.95
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1

#  # Fallback wifi access point in case wifi connection fails
#  ap:
#    ssid: "ePaper Fallback Hotspot"
#    password: "password"
#  # In combination with the `ap` this allows the user
#  # to provision wifi credentials to the device via WiFi AP.
#  captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
# This needs to be disabled for reliable WiFi when used with the Waveshare display!
# esp32_improv:
#  authorizer: none

# Enable the web server, so it is reachable at http://<ip>/
web_server:

# Google Fonts
font:
# - file: "gfonts://Inter@900"
#   id: big
#   size: 120
 - file: "gfonts://Inter@900"
   id: medium
   size: 60
# - file: "gfonts://Inter@700"
#   id: small
#   size: 30
# - file: "gfonts://Inter@700"
#   id: tiny
#   size: 24

# Pins for Waveshare ePaper ESP Board
spi:
 # SPI Clock Pin
 clk_pin: GPIO13
 # SPI Master Out Slave In Pin
 mosi_pin: GPIO14

# Pins for Waveshare ePaper ESP Board
display:
 - platform: waveshare_epaper
   id: screen
   # Chip Select Pin
   cs_pin: GPIO15
   # Data/Command Pin
   dc_pin: GPIO27
   # Busy Pin
   busy_pin:
     number: GPIO25
     inverted: true
   # Reset Pin
   reset_pin: GPIO26
   reset_duration: 2ms
   model: 7.50inV2alt
   update_interval: 60s
   rotation: 90°
   lambda: |-
     it.print(40, 10, id(medium), "7.50inV2alt");
```

closed 2024-10-22 by eikowagenknecht

sylvainzabe, 2024-10-22:
> @eikowagenknecht I think I found the problem or at least the fix. The following changes worked for me. I got the following wifi config:
> 
> ...

Thank you very much, it's ok for me too ! \o/