esphome/issues #5913: MPR121 not working in 2024.6.0b

SmarterHomeLife, 2024-06-17:
### The problem

The MPR121 causes a panic and boot loop ending in safe mode being activated when the chip is connected.

### Which version of ESPHome has the issue?

2024.6.0b2 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

21024.6.3

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

mpr121

### Example YAML snippet

```yaml
substitutions:
  devicename: homeiq-d1-remote-5d2a6a
  name: HomeIQ D1 Remote
  static_ip: 10.0.4.16

esphome:
  name: $devicename
  platform: ESP8266
  board: d1_mini
  name_add_mac_suffix: no

wifi:
  ssid: !secret ssid
  password: !secret psk
  fast_connect: on
  manual_ip:
    static_ip: $static_ip
    gateway: 10.0.4.1
    subnet: 255.255.255.0
#  use_address: devicename.local
  ap:
    ssid: $name WiFi
    password: !secret hotspot-psk

captive_portal:

web_server:
  port: 80
  version: 3

logger:

api:
  encryption:
    key: !secret esphome-key
  reboot_timeout: 0s

ota:
  platform: esphome
[... 243 lines cut ...]
[20:22:51]3fffffb0:  3fffdad0 00000000 3fff048c 40231f0c  
WARNING Decoded 0x40231f0c: loop_wrapper() at core_esp8266_main.cpp
[20:22:52]3fffffc0:  feefeffe feefeffe 3fffdab0 4010048d  
WARNING Decoded 0x4010048d: cont_wrapper
[20:22:52]<<<stack<<<
[20:22:52]
[20:22:52]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[20:22:52]
[20:22:52] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[20:22:52]
[20:22:52]load 0x4010f000, len 3424, room 16 
[20:22:52]tail 0
[20:22:52]chksum 0x2e
[20:22:52]load 0x3fff20b8, len 40, room 8 
[20:22:52]tail 0
[20:22:52]chksum 0x2b
[20:22:52]csum 0x2b
[20:22:52]v00083b60
[20:22:52]~ld
[20:22:52]\x8c\x82n\xc7r\x90\xf2n|\xecl\xc4l`bbrl\xe2nb\x82nl`\x8crl\xc4l\xe4\xfe[I][logger:034]: Log initialized
[20:22:52][C][safe_mode:079]: There have been 2 suspected unsuccessful boot attempts
[20:22:52][I][app:029]: Running through setup()...
[20:22:52][I][i2c.arduino:218]: Performing I2C bus recovery
[20:22:52][C][mpr121:014]: Setting up MPR121...
```

SmarterHomeLife, 2024-06-17:
I rolled back to esphomne 2024.5.5 and it's working perfectly.

To be clear, the firmware compiles and installs fine without the mpr121 connected it just fails the i2c scan but when I boot with the mpr121 connected I get the above.

coolinger, 2024-06-21:
Just commenting my observations:

Only binary_sensor with platform mpr121 triggers the error. 

Just using it as GPIO for input

```
- platform: gpio
    pin: 
      mpr121: mpr121c
      number: 4
      mode: INPUT
      inverted: True
    id: Num4
```
or output

```
output:
  - id: ext_5v_power
    platform: gpio
    pin:
      mpr121: mpr121c
      number: 10
      mode: OUTPUT
      inverted: true
```
or not using it at all does not trigger the error.

I have this problem on ESP32-WROVER and a ESP32-WROOM-32U devkits with 2024.6.1

morris2009, 2024-06-22:
Same issue here.
in 2024.6.0-dev it was still working
in 2024.7.0-dev it is broken

> [23:10:48][C][mpr121:014]: Setting up MPR121...
> 
> [23:10:49]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> [23:10:49]
> [23:10:49]Core  1 register dump:
> [23:10:49]PC      : 0x400e0908  PS      : 0x00060030  A0      : 0x800e0969  A1      : 0x3ffb25e0  
> [23:10:49]A2      : 0x3ffb2b70  A3      : 0x0000005e  A4      : 0x3ffb260c  A5      : 0x00000001  
> [23:10:49]A6      : 0x00000001  A7      : 0x00000001  A8      : 0x801913ae  A9      : 0x00000000  
> [23:10:49]A10     : 0x00000000  A11     : 0x00000002  A12     : 0x3ffb25e0  A13     : 0x00000002  
> [23:10:49]A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
> [23:10:49]EXCVADDR: 0x00000025  LBEG    : 0x4008a394  LEND    : 0x4008a3aa  LCOUNT  : 0xffffffff  
> [23:10:49]
> [23:10:49]
> [23:10:49]Backtrace:0x400e0905:0x3ffb25e00x400e0966:0x3ffb2610 0x400e0b0d:0x3ffb2630 0x40191db9:0x3ffb2650 0x40191e45:0x3ffb2670 0x400e6380:0x3ffb2690 0x400e90ca:0x3ffb26c0 0x400ef666:0x3ffb2820 
> [23:10:49]
> [23:10:49]
> [23:10:49]
> [23:10:49]
> [23:10:49]ELF file SHA256: 0000000000000000
> [23:10:49]
> [23:10:49]Rebooting...
> [23:10:49]ets Jun  8 2016 00:22:57
> [23:10:49]
> [23:10:49]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> [23:10:49]configsip: 0, SPIWP:0xee
> [23:10:49]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> [23:10:49]mode:DIO, clock div:2
> [23:10:49]load:0x3fff0030,len:1184
> [23:10:49]load:0x40078000,len:13132
> [23:10:49]load:0x40080400,len:3036
> [23:10:49]entry 0x400805e4
> [23:10:49][I][logger:156]: Log initialized
> [23:10:49][C][safe_mode:079]: There have been 10 suspected unsuccessful boot attempts

closed 2024-06-29 by kbx81