esphome/issues #5919: Bluetooh Proxy doesnt detect Devices

mokshmridul, 2024-06-19:
### The problem

I am unable to see / discover bluetooth devices using the bluetooth proxy component. The esp32 was configured using the webservice on the ESPHome page.

### Which version of ESPHome has the issue?

2024.5.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.3

### What platform are you using?

ESP32

### Board

ESP32-WROOM

### Component causing the issue

Bluetooth Proxy

### Example YAML snippet

```yaml
substitutions:
  name: esp32-ble-proxy
  friendly_name: ESP32 Bluetooth Proxy
packages:
  esphome.bluetooth-proxy: github://esphome/firmware/bluetooth-proxy/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: "secret_api_key"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.5.5
INFO Reading configuration /config/esphome/esp32-bluetooth-proxy-19b8ac.yaml...
INFO Starting log output from 192.168.1.19 using esphome API
INFO Successfully connected to esp32-ble-proxy @ 192.168.1.19 in 0.006s
INFO Successful handshake with esp32-ble-proxy @ 192.168.1.19 in 0.096s
[01:20:21][I][app:100]: ESPHome version 2024.5.5 compiled on Jun 20 2024, 00:40:44
[01:20:21][I][app:102]: Project esphome.bluetooth-proxy version 1.0
[01:20:21][C][wifi:580]: WiFi:
[01:20:21][C][wifi:408]:   Local MAC: FC:F5:C4:19:B8:AC
[01:20:21][C][wifi:413]:   SSID: [redacted]
[01:20:21][C][wifi:416]:   IP Address: 192.168.1.19
[01:20:21][C][wifi:420]:   BSSID: [redacted]
[01:20:21][C][wifi:421]:   Hostname: 'esp32-ble-proxy'
[01:20:21][C][wifi:423]:   Signal strength: -42 dB ▂▄▆█
[01:20:21][C][wifi:427]:   Channel: 5
[01:20:21][C][wifi:428]:   Subnet: 255.255.255.0
[01:20:21][C][wifi:429]:   Gateway: 192.168.1.1
[01:20:21][C][wifi:430]:   DNS1: 192.168.1.1
[01:20:21][C][wifi:431]:   DNS2: 0.0.0.0
[01:20:21][C][logger:185]: Logger:
[01:20:21][C][logger:186]:   Level: DEBUG
[01:20:21][C][logger:188]:   Log Baud Rate: 115200
[01:20:21][C][logger:189]:   Hardware UART: UART0
[01:20:21][C][bluetooth_proxy:088]: Bluetooth Proxy:
[01:20:21][C][bluetooth_proxy:089]:   Active: YES
[01:20:21][C][safe_mode.button:022]: Safe Mode Button 'Safe Mode Boot'
[01:20:21][C][safe_mode.button:022]:   Icon: 'mdi:restart-alert'
[01:20:21][C][esp32_ble:374]: ESP32 BLE:
[01:20:21][C][esp32_ble:376]:   MAC address: FC:F5:C4:19:B8:AE
[01:20:21][C][esp32_ble:377]:   IO Capability: none
[01:20:21][C][esp32_ble_tracker:649]: BLE Tracker:
[01:20:21][C][esp32_ble_tracker:650]:   Scan Duration: 300 s
[01:20:21][C][esp32_ble_tracker:651]:   Scan Interval: 320.0 ms
[01:20:21][C][esp32_ble_tracker:652]:   Scan Window: 30.0 ms
[01:20:21][C][esp32_ble_tracker:653]:   Scan Type: ACTIVE
[01:20:21][C][esp32_ble_tracker:654]:   Continuous Scanning: True
[01:20:21][C][mdns:115]: mDNS:
[01:20:21][C][mdns:116]:   Hostname: esp32-ble-proxy
[01:20:21][C][ota:096]: Over-The-Air Updates:
[01:20:21][C][ota:097]:   Address: esp32-ble-proxy.local:3232
[01:20:21][C][ota:103]:   OTA version: 2.
[01:20:21][C][api:139]: API Server:
[01:20:21][C][api:140]:   Address: esp32-ble-proxy.local:6053
[01:20:21][C][api:142]:   Using noise encryption: YES
[01:20:21][C][improv_serial:032]: Improv Serial:
[01:23:13][D][esp32_ble_tracker:266]: Starting scan...
```

randybb, 2024-06-19:
BLE Proxy is forwarding everything to HA and if something is near and supported, it will just pop up in HA. You will not see it in the esphome log.

mokshmridul, 2024-06-20:
Ahh okay. The ikea Idasen desk is supported and right next to the esp32, but is not showing up. Would this issue be better suited for the HA github?

jesserockz, 2024-06-20:
@mokshmridul Leave this issue open for now, but your best bet here is to join [Discord](https://discord.gg/KhAMKrd) where we can help with faster responses.

jesserockz, 2024-06-20:
Just to check, you did add the ESPHome device to Home Assistant right? 
[image]

mokshmridul, 2024-06-20:
Yes, the esp device is added to Home Assistant. Will reach out to you on Discord for help. Thank you.

mokshmridul, 2024-06-20:
Closing this as it seems to be an issue with HA and not with the ESP Proxy, as the device MAC is visible in the ESP Diagnostic logs.

closed 2024-06-20 by mokshmridul