esphome/issues #5922: Spurious one-wire warning "Found no devices!" after 6.0 update

tescher, 2024-06-20:
### The problem

Might be related to https://github.com/esphome/esphome/pull/6860

In the log a warning is issued that no one_wire devices are found, but in fact they later log data just fine:

[18:41:07][C][logger:185]: Logger:
[18:41:07][C][logger:186]:   Level: DEBUG
[18:41:07][C][logger:188]:   Log Baud Rate: 115200
[18:41:07][C][logger:189]:   Hardware UART: UART0
[18:41:07][C][gpio.one_wire:016]: GPIO 1-wire bus:
[18:41:07][C][gpio.one_wire:017]:   Pin: GPIO4
**[18:41:07][W][gpio.one_wire:078]:   Found no devices!**
[18:41:07][C][template.text_sensor:020]: Template Sensor 'AK ESP CABIN BASEMENT ALIVE'
[18:41:07][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[18:41:07][C][dallas.temp.sensor:034]:   Address: 0xd1000004a7beaa28 (DS18B20)
[18:41:07][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[18:41:07][C][dallas.temp.sensor:036]:   Update Interval: 60.0s
[18:41:07][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[18:41:07][C][dallas.temp.sensor:034]:   Address: 0x830000038969eb28 (DS18B20)
[18:41:07][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[18:41:07][C][dallas.temp.sensor:036]:   Update Interval: 60.0s
[18:41:07][C][captive_portal:088]: Captive Portal:
[18:41:07][C][web_server:173]: Web Server:
[18:41:07][C][web_server:174]:   Address: 192.168.2.142:80
[18:41:07][C][mdns:115]: mDNS:
[18:41:07][C][mdns:116]:   Hostname: ak_cabin_basement
[18:41:07][C][esphome.ota:073]: Over-The-Air updates:
[18:41:07][C][esphome.ota:074]:   Address: 192.168.2.142:8266
[18:41:07][C][esphome.ota:075]:   Version: 2
[18:41:07][C][safe_mode:018]: Safe Mode:
[18:41:07][C][safe_mode:019]:   Boot considered successful after 60 seconds
[18:41:07][C][safe_mode:021]:   Invoke after 10 boot attempts
[18:41:07][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[18:41:07][C][api:139]: API Server:
[18:41:07][C][api:140]:   Address: 192.168.2.142:6053
[18:41:07][C][api:144]:   Using noise encryption: NO
[18:41:10][D][api:102]: Accepted 192.168.2.138
[18:41:10][D][api.connection:1375]: Home Assistant 2024.6.3 (192.168.2.138): Connected successfully
[18:41:36][D][text_sensor:064]: 'AK ESP CABIN BASEMENT ALIVE': Sending state 'I'm alive'
**[18:41:37][D][dallas.temp.sensor:054]: 'AK Cabin Outside': Got Temperature=17.5°C
[18:41:37][D][sensor:093]: 'AK Cabin Outside': Sending state 17.50000 °C with 1 decimals of accuracy
[18:41:38][D][dallas.temp.sensor:054]: 'AK Cabin Basement': Got Temperature=16.8°C
[18:41:38][D][sensor:093]: 'AK Cabin Basement': Sending state 16.75000 °C with 1 decimals of accuracy**

### Which version of ESPHome has the issue?

2024.6.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
This is my config:

one_wire:
  - platform: gpio
    pin: 4
  
sensor:
  - platform: dallas_temp
    address: 0xD1000004A7BEAA28
    name: "AK Cabin Outside"
  - platform: dallas_temp
    address: 0x830000038969EB28
    name: "AK Cabin Basement"
```

grssll, 2024-06-20:
Same issue here with latest update.
No sensor found

ssieb, 2024-06-20:
@grssll is yours also an 8266?

mattfro, 2024-06-20:
Same here and I have an esp32.

supergvg, 2024-06-20:
Same issue after update on ESP32 :(
In log i see next rows:

```
[12:23:01][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[12:23:02][W][dallas.temp.sensor:074]: 'Temperature #1 (Green)' - reading scratch pad failed bus reset
[12:23:02][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed
```

7Van737, 2024-06-20:
I'm having trouble updating my esp8266 after the recent update.

Get the following error - 

INFO ESPHome 2024.6.1
INFO Reading configuration /config/esphome/cylinder2.yaml...
Failed config

one_wire.unknown: [source /config/esphome/cylinder2.yaml:14]
  
  'one_wire' requires a 'platform' key but it was not specified.
  pin: GPIO05
  update_interval: 30s

My config code - which does define a platform as per the example

esp8266:
  board: esp01_1m

#dallas:
#  - pin: GPIO05
#    update_interval: 30s

one_wire:
  - platform: gpio
  - pin: GPIO05
    update_interval: 30s

sensor:
  - platform: dallas_temp
    address: "0xe63ce10457a12428" 
    name: "Cylinder Thermostat"
  - platform: dallas_temp
    address: "0xf93ce10457fb2528" 
    name: "Cylinder Top Temperature"
  - platform: dallas_temp
    address: "0x72041771ff8aff28"
    name: "Loft Temperature"
    ``

Different error to this issue but I'm guessing perhaps related??   Grateful for any advice.

randybb, 2024-06-20:
@7Van737 yes, this is different, your formatting is wrong. you have - before platform and then - before gpio which is wrong. Change this:
```yaml
one_wire:
  - platform: gpio
  - pin: GPIO05
    update_interval: 30s
```
to this:
```yaml
one_wire:
  - platform: gpio
    pin: GPIO05
    update_interval: 30s
```

7Van737, 2024-06-20:
@randybb Thank you for your help - this sorted my problem!    Sorry for spamming this thread!

schinddo, 2024-06-20:
@tescher Have you also checked if the temperatures of your sensors are correct? In addition to this issue I had also the problem with some sensors having wrong temperatures.

tescher, 2024-06-20:
@schinddo yes the temps appear to be correct. The only issue I've noticed is what I referenced at the top, the spurious warning

sanchosk, 2024-06-20:
> - source:
>       type: git
>       url: https://github.com/ssieb/esphome
>       ref: onewire
>     components: [ gpio ]
>     refresh: 1min```

Would you be so kind and point me out where in the YAML should I include this, please?

horinka-gabor, 2024-06-20:
I got the same warning on esp8266 pin D5 with Dallas.

grssll, 2024-06-20:
> ```
>        type: git
>        url: https://github.com/ssieb/esphome
>        ref: onewire
>      components: [ gpio ]
>      refresh: 1min```
> ```
> 
> Would you be so kind and point me out where in the YAML should I include this, please?

external_components:
  - source:

grssll, 2024-06-20:
```
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: onewire
    components: [ gpio ]
    refresh: 1min
```

pauljatho, 2024-06-21:
Also having the same issue....

[11:37:04][C][gpio.one_wire:016]: GPIO 1-wire bus:
[11:37:04][C][gpio.one_wire:017]:   Pin: GPIO13
[11:37:04][W][gpio.one_wire:078]:   **Found no devices!**

Config:

```
#-------Temperature------------------------------------
#dallas:
one_wire:
  - platform: gpio
    pin: GPIO13

sensor:
  - platform: dallas_temp
    address: 0xe00300a279bb0828
    name: "ESP Temperature"
#-----------------------------------------------------

ssieb, 2024-06-21:
Try adding that external component chunk to your config and see if it fixes it.

pauljatho, 2024-06-21:
> Try adding that external component chunk to your config and see if it fixes it.

Ah missed that, will give it a go...

u20p17, 2024-06-24:
> ```
> external_components:
>   - source:
>       type: git
>       url: https://github.com/ssieb/esphome
>       ref: onewire
>     components: [ gpio ]
>     refresh: 1min
> ```

thanks, this fixed the issue for now... (Y)

sebsebseb1982, 2024-07-16:
> Fixed in 2024.6.2

Hi @ssieb, I [still have the problem with 2024.6.6](https://github.com/esphome/issues/issues/6045). Do you have an idea ?

LaurentMicrotec, 2024-07-25:
Hi, have the same problem about my DS18S20.
I don't understand what the solution.
Some one can help please?
Thank

ssieb, 2024-07-25:
@LaurentMicrotec come ask on discord.  There are many possible reasons.

zimbo86, 2024-08-19:
same here, 2024.6.6
Sensor works for weeks and stops in the middle of the day. 

[image]