esphome/issues #5924: firebeetle32 doesn't connect after update to 2024.6.0

emefff, 2024-06-20:
### The problem

I have one single firebeetle32 that does not reconnect after the recent update to 2024.6.0. It also has 3 different 'lanes' of dallas sensors. When I built the device a few years ago, I had a lot of issues when I put all the dallas sensors on one-wire (seems quite funny now... :-) ).

### Which version of ESPHome has the issue?

2024.6.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.06

### What platform are you using?

ESP32

### Board

firebeetle32

### Component causing the issue

unknown, I don't suspect dallas (one_wire now)

### Example YAML snippet

```yaml
esphome:
  name: esphome--12
  platformio_options:
    board_build.f_cpu: 81000000L #board temp lower, cannot use 80000000L because of issue 3909

esp32:
  board: firebeetle32
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ***REDACTED***

ota:
  platform: esphome
  password: ***REDACTED***

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  #output_power: 20dB #8.5-20db
  manual_ip:
    static_ip: 192.168.178.163
    gateway: 192.168.178.10
    subnet: 255.255.255.0
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome--12 Fallback Hotspot"
    password: ***REDACTED***

captive_portal:

time:
  - platform: homeassistant
    id: homeassistant_time
    
one_wire: #we need more than one pin this time....
  - platform: gpio
    pin: 5  
    id: dallas5
  - platform: gpio
    pin: 14
    id: dallas14
  - platform: gpio
    pin: 25
    id: dallas25
  
# Individual sensors
sensor:
  - platform: dallas_temp
    one_wire_id: dallas25
    address: 0x2c09213026549128
    name: "Wohnzimmer 2 ESP Dallas Onboard Temperature"
    accuracy_decimals: 1
    update_interval: 61s
    filters:
      - filter_out: nan
      - filter_out: 85.0
      - offset: 0

  - platform: dallas_temp
    one_wire_id: dallas14
    address: 0xcc06225043066e28
    name: "Wohnzimmer 2 Dallas HZK Zulauf"
    accuracy_decimals: 1
    update_interval: 59s
    filters:
      - filter_out: nan
      - filter_out: 85.0
      - offset: +0.1 #nachträglich +0.04 zu den 0.06

*** only two sensors are shown, there are a lot more. With 2024.6.0 I had to put the update_interval below the sensor, before it was below the dallas component declaration.
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.6.0
INFO Reading configuration /config/esphome/esphome--12.yaml...
INFO Detected timezone 'Europe/Vienna'
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 192.168.178.163 using esphome API
WARNING Can't connect to ESPHome API for esphome--12 @ 192.168.178.163: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.178.163', port=6053))]: [Errno 111] Connect call failed ('192.168.178.163', 6053) (SocketAPIError)
INFO Trying to connect to esphome--12 @ 192.168.178.163 in the background

**** The shown GPIO2 and GPIO5 warnings were never a problem, other working devices show them too. Don't ask about them please.
```

### Additional information

This is my only 'board: firebeetle32' I updated until now. I have one battery powered device left, that is still on 2024.5.something.
Could this be the issue?

ssieb, 2024-06-20:
You need to get serial logs to find out why it's not connecting to wifi.

emefff, 2024-06-20:
I have no idea how to do this, I guess I have to connect it via USB to my computer running HA? Also, how can I make the USB accessible to HA that runs in virtualbox? That never worked out, I always used ESPHome-Flasher-1.4.0-Ubuntu-x64 (which by the way, is sort of crippled now also) for flashing via USB. It's not easily accessible at the moment, I have to uninstall some cables. At the moment, I can't do this. Although, I probably have to reflash anyway. But at the moment the given info is all I have.

ssieb, 2024-06-20:
You shouldn't be using esphome flasher.  Plug the ESP with USB to your computer.  Then use either the web installer or a program like putty to watch the serial port.

emefff, 2024-06-20:
As I said before. That NEVER worked for reasons unknown to me. The web installer never sees the USB device. I have to use the options that work.

emefff, 2024-06-20:
When I plug the firebeetle into a USB of my computer running Ubuntu 22.04 I get the following in my dmesg:
`[ 5964.653586] usb 1-6.1: new full-speed USB device number 6 using xhci_hcd
[ 5964.755412] usb 1-6.1: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice=81.32
[ 5964.755425] usb 1-6.1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 5964.755430] usb 1-6.1: Product: USB Serial
[ 5964.763558] ch341 1-6.1:1.0: ch341-uart converter detected
[ 5964.764217] usb 1-6.1: ch341-uart converter now attached to ttyUSB0
[ 5965.386431] input: BRLTTY 6.4 Linux Screen Driver Keyboard as /devices/virtual/input/input32
[ 5965.387740] usb 1-6.1: usbfs: interface 0 claimed by ch341 while 'brltty' sets config #1
[ 5965.388675] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[ 5965.388694] ch341 1-6.1:1.0: device disconnected
`
Does that help, should I do anything else?

emefff, 2024-06-20:
....and now it's a brick. No connection to USB whatsoever.

emefff, 2024-06-20:
Ok, i removed brttly according to this:
https://stackoverflow.com/questions/70123431/why-would-ch341-uart-be-disconnected-from-ttyusb

Then I flashed the firebeetle32 with the esphome-FW it should receive initially. It entered a boot-loop and then proceeded to safe-mode. In the log I found this:

[12:54:02][     7][E][esp32-hal-cpu.c:182] setCpuFrequencyMhz(): Bad frequency: 81 MHz! Options are: 240, 160, 80, 40, 20 and 10 MHz

It seems 81MHz is not allowed anymore, this will probably problematic with the battery powered devices. With 80MHz (as can be read in issue 3909) I cannot use the dallas sensors. But maybe that is not necessary anymore. With the battery powered devices I have to set the lowest CPU freq possible. Let's see what is possible.

emefff, 2024-06-20:
I'll mark this issue solved, but it's just a workaround. https://github.com/esphome/issues/issues/5929 persists.

closed 2024-06-20 by emefff